cncjs/cncjs #11: Zeroing axes - offsets vs. work coordinates

meikrantz, 2015-12-07:
Can you change the behavior of the "zero axis" drop down to include G10 in addition to G92 (setting work coordinates instead of offsets)?  Perhaps also add workspace awareness, i.e. G54.

See discussion in grbl repository - https://github.com/grbl/grbl/issues/155

cheton, 2015-12-08:
Taking Notes 
* G10 L2 Set Coordinate System<br>
  http://linuxcnc.org/docs/2.5/html/gcode/gcode.html#sec:G10-L2_
* G10 L20 Set Coordinate System<br>
  http://linuxcnc.org/docs/2.5/html/gcode/gcode.html#sec:G10-L20
* G54-G59.3 Select Coordinate System<br>
  http://linuxcnc.org/docs/2.5/html/gcode/gcode.html#sec:G54-G59_3

meikrantz, 2015-12-08:
Good explanation on the Shapeoko Wiki entry on gcode - under using the work coordinate systems.
http://www.shapeoko.com/wiki/index.php/G-Code

Also take a look at how it's been implemented in chilipeppr GUI for tinyg.
http://chilipeppr.com/tinyg

cheton, 2015-12-10:
I'm currently working on this. It should allow user to select a work coordinate system (G54-G59) and setting the X/Y/Z axis offsets.

In addition to work coordinates, is it necessary to add  support for G28.1 and G28 commands to save and restore the absolute position?

It seems that Grbl does not support G28.2 and G28.3 commands to set absolute coordinates just like TinyG does (ref. https://github.com/synthetos/TinyG/wiki/Coordinate-Systems).

meikrantz, 2015-12-10:
I will sometimes use G28 and G28.1, but not frequently enough to be a dedicated "button".  Easily set as a macro or just direct command input.  Grbl doesn't support 28.2 and 28.3 as you have noted.

The other missing features I'd like to see as a "button" or something are Home, Unlock, and Reset.  I see those are available from the drop down, but those are frequently used, so a more direct method would be nice.

cheton, 2015-12-10:
This enhancement has been added to [v0.8.0](https://github.com/cheton/cnc.js/releases/tag/v0.8.0).

cheton, 2015-12-10:
I agreed with you on that. I will find a proper place to put Home, Unlock, and Reset buttons.

meikrantz, 2015-12-10:
Looks better.  One suggestion - since there is no easy way to change between workspaces in the current GUI, I would just have the action dropdown only select zeroing in G54.  Most Grbl users are probably not using multiple work coordinate systems, so G54 is probably sufficient.  I'll have a chance to play with it a bit more over the next few days.

cheton, 2015-12-11:
A "Grbl Quick Access Toolbar" has been added to the top navigation bar. In the future, the toolbar buttons will be customizable under the "Settings > Configure Quick Access Toolbar" dropdown.

The G10 command for zeroing out offsets in G55-G59 were removed in v0.9.x to make UI more simple.
I will consider adding it back once the "Select Coordinate System" feature was added to the UI.

meikrantz, 2015-12-11:
Looks great.  One other minor suggestion - under the XYZ menu could you add zero out workspace coordinates as well as zero out temporary offsets?

cheton, 2015-12-11:
Fixed in v0.9.2

closed 2015-12-21 by cheton