cncjs/cncjs #12: 3D visualizer in check gcode mode ($C)

meikrantz, 2015-12-11:
Was just running some offline stuff with cnc.js, and noticed that the visualizer doesn't work when I load up some code and run it in gcode check mode.

Not sure if that's design intent.  Or it could be a problem running on Mac with the 3D visualizer not being supported.

cheton, 2015-12-12:
You cannot see axes movement while the Check G-code Mode ($C) is enabled, but the 3D visualizer should be able to simulate both online and offline tasks.

May you check out some G-code examples under the (examples)[/examples] directory, and perform the simulation again in your Mac OS X? 

It will be helpful if you can provide me the screenshot for troubleshooting. 

Thanks!

meikrantz, 2015-12-12:
Some of the gcode examples in the /examples directory do show up in the visualizer.  Others that I have generated with Vectric don't display, although the gcode seems to load fine.  Example attached.

[isometric_gcode.txt](https://github.com/cheton/cnc.js/files/60042/isometric_gcode.txt)

cheton, 2015-12-12:
Looks like I didn't consider the use case of having no spaces between commands, and these commands will be ignored by the parser.

Examples:
G1Z10 - not working
G1 Z10 - works as expected

Recently I'm working on new G-code [parser](https://github.com/cheton/gcode-parser) and [interpreter](https://github.com/cheton/gcode-interpreter), and I will get it fixed up later.

cheton, 2015-12-12:
This issue is fixed in [v0.9.3](https://github.com/cheton/cnc.js/releases/tag/v0.9.3).
Now the 3D visualizer should be able to render the G-code data that have no spaces between commands.

meikrantz, 2015-12-12:
Works great, thanks for the quick fix!

cheton, 2015-12-15:
I also made an important bug fix in v0.10.3 which will resolve an issue where the periodic status check for `?` and `$G` commands may stop working after sending the Ctrl-X command to reset Grbl.

meikrantz, 2015-12-15:
I was going to post a new issue that this may have solved.  I noticed that the coordinates stopped updating periodically, but I never made a correlation as to why.  Could have been the Ctrl-X.  I'll test v0.10.3 soon

closed 2015-12-21 by cheton