cncjs/cncjs #14: First impressions

jarretluft, 2015-12-18:
Hi,
I just installed and ran cnc.js, and it looks fantastic.  However, i'm on a mac, and either I am missing something obvious or there is a problem, because I cannot load a gcode file.  I assumed since there is no upload option that I should just drag/drop the .nc or .gcode file onto the browser window, but that does not work for me (on either chrome or safari), this may be a mac-specific problem...

Also, is there a way to suppress all the console output printing the ? and $G responses by default and only showing that if a verbose/debug flag is sent on startup? The app is using 51% of my cpu while sitting idle, i suspect some of that overhead is due to constantly printing those console output lines.

cnc.js definitely has potential though!
Cheers,
-J.

meikrantz, 2015-12-19:
Jarret,

I am also running the web interface on a Mac, with no issue.  To upload a file, click on the little cloud icon in the 3d viewer area.  When you want to load a different file, click on the little trash can to close the existing gcode file.

Haven't noticed high CPU or console messages though.

cheton, 2015-12-19:
Hi Jarret,

Thanks for the note. The drag-and-drop feature is indeed currently missing from the web interface, but I had added it to the to-do list and will  add it soon.

As for the high CPU usage when sitting idle, I will look into it for further investigation.

cheton, 2015-12-19:
I also noticed the CPU usage is high caused by the web browser even for being idle, it seems that I ran into the same issue as https://github.com/mrdoob/three.js/issues/7670.
```js
function render() {
    requestAnimationFrame(render);
    renderer.render(scene, camera);
}
render();
```

By default, the `requestAnimationFrame` loop will render the scene with a 60HZ refresh rate. The CPU will keep working high even if nothing needs to update in the scene.

I will try to turn into using the static scene rendering only when an event occurred, like so:
```js
// controls
let controls = new THREE.OrbitControls(camera);
controls.addEventListener('change', render);

let renderAnimationFrame = false; // a boolean flag that will be updated from mouse and keyboard event

function render() {
    if (renderAnimationFrame) {
        requestAnimationFrame(render);
    }
    renderer.render(scene, camera);
}
render();
```

cheton, 2015-12-20:
HI Jarret,

You can check out [v0.11.6](https://github.com/cheton/cnc.js/releases/tag/v0.11.6) for the fixes of high CPU usage when sitting idle:
- Resolving an issue of the web intrerface that caused high CPU usage in idle state - #14

cheton, 2015-12-20:
The fixes will stop the render animation loop if current status is not in running state. Any state changes will trigger `componentDidUpdate` when the component has been updated. It seems `componentDidUpdate` is a good place to call  `requestAnimationFrame` to restart the loop.

For example:
```js
renderAnimationLoop() {
    let isAgitated = (this.state.activeState === ACTIVE_STATE_RUN) &&
                     (this.state.workflowState === WORKFLOW_STATE_RUNNING);
    if (isAgitated) {
        requestAnimationFrame(::this.renderAnimationLoop); // 60 fps refresh rate
        // start animations
     } else {
        // stop animations
    }

    this.renderer.render(this.scene, this.camera);
}
componentDidUpdate(prevProps, prevState) {
     // The renderAnimationLoop will check the state of activeState and workflowState
    requestAnimationFrame(::this.renderAnimationLoop);
}
```

cheton, 2015-12-21:
Drag and drop support for loading G-code file is added in [v0.12.0](https://github.com/cheton/cnc.js/releases/tag/v0.12.0)

jarretluft, 2015-12-22:
Just tested 0.12.0, this looks fantastic.

Both the high CPU usage on idle, and the drag and drop are fixed.  I like what you did with the gcode drag/drop box, as well I like that the 3d viewer no longer scrolls with the other boxes, so there is no whitespace underneath and you never lose the play/pause buttons off screen at the top.

A couple other observations:
- It seems that when hitting pause, it doesn't immediately stop movement, but instead finishes cutting all gcode that has been sent into the grbl queue already.  As well, during a pause, a person can still jog.  If a ! was used to pause, and ~ to unpause the stop would be immediate, and grbl would be locked so no jogging could occur that might put the endmill in an unexpected location for the next cut in the queue to execute incorrectly.

- For Jogging, most other software seems to ignore feedrate and use G0 commands (rapid speeds set in GRBL), is there a way to give users this option?  I would rather never have to set a feed rate and always move at rapids speed which I have pre-defined than have to adjust the default feedrate (250) to my typical rapid rate (5000) each time I load the page.  An alternative to this could be to store a settings cookie that allowed my feedrate setting of 5000 to persist between page loads.

- It would be great if any changes to the ordering/placement of widgets in the side bars, or the hiding of side bars was stored in a cookie to persist between page refreshes as well.  Right now I customize the placement of widgets only to have them all reset the next time the page is loaded

- The duration counter does not stop when the gcode finishes sending... It seems to keep counting indefinitely :)

- Any thoughts on coloring G0, G1, G2/3 lines different colors in the 3d viewer? I think i'm just used to this in chilipeppr but find myself missing it when using cnc.js, as it was a quick way to see where the rapid movements would be vs the actual milling.  Even a faded out green for the G0 and the dark green for G1/2/3 would give some indication of which movements the spindle will be under load vs rapid movements between cuts.

Sorry if these should all be separate issues, I can move them all out individually if that would be better.

Keep up the great work; I am hoping to dive more into your code in the next few weeks and see if there is anything I could contribute back to your project.  I developed the chilipeppr grbl workspace so I am familiar with much of the code there for the touch plate widget, autolevel widget, grbl buffer handling, etc.   

Cheers,
-J.

cheton, 2015-12-23:
Thank you so much for your advise. I really appreciate it.

Please see below my response to the points that you have raised:

- [v0.12.2](https://github.com/cheton/cnc.js/releases/tag/v0.12.2): Now clicking on the `[PAUSE]` button will also send the `!` feed hold command to Grbl to hold current operation immediately. The `~` cycle start command will be issued when clicking `[PLAY]` again to resume current operation.
Within the `Hold` state (i.e. paused), you'll have a chance to stop and clear all motions queued in the buffer by clicking on the `[STOP]` button. It will issue a `Ctrl-X` command as well to reset Grbl from `Hold` state to `Idle` state.<br> However if one is doing so, Grbl may enter the `Alarm` state in certain conditions, it is necessary to let user know an `Unlock` command is required to allow G-code function to work again. I'm sill working on this.

- [v0.12.4](https://github.com/cheton/cnc.js/releases/tag/v0.12.4): The feedrate control has been removed from the Axes widget since it is not necessary for jogging.

-  Persistent settings is a missing feature since the very beginning, I will try to add it soon.

- The non-stopping duration counter is a bug, and it will be fixed in later version.

- With respect to rendering different colors for G0~G3 commands, I made a quick fix in [v0.12.3](https://github.com/cheton/cnc.js/releases/tag/v0.12.3), but it would be better if the user is able to select custom colors according to his preferences.

cheton, 2015-12-26:
Now the widget position can be remembered since [v0.13.4](https://github.com/cheton/cnc.js/releases/tag/v0.13.4)

cheton, 2015-12-26:
[v0.13.4](https://github.com/cheton/cnc.js/releases/tag/v0.13.4): Remember widget position
[v0.13.5](https://github.com/cheton/cnc.js/releases/tag/v0.13.5): Bug fixes for the non-stopping duration counter

cheton, 2016-01-04:
TODO: Restore UI after browser refresh

cheton, 2016-01-10:
[v0.14.7](https://github.com/cheton/cnc.js/releases/tag/v0.14.7) Restoring G-code after browser refresh

cheton, 2016-01-10:
I will close the issue since most issues mentioned above have been fixed. Thanks for your feedbacks here.

closed 2016-01-10 by cheton