cncjs/cncjs #16: Implement G28/G30

meikrantz, 2015-12-23:
It would be nice to have a dedicated button for G28 and G30 predefined positions.  Perhaps with a drop down with G28.1 and G30.1 to set the value.

cheton, 2015-12-23:
According to the article, "[G28 AND G30 - RETURN TO HOME](http://www.tormach.com/g28_g30.html)",  it will move from current point to the intermediate point if axis words are provided, and finally to the home position defined by G28.1 and G30.1.

Do you think is it necessary to provide the intermediate point option for G28 and G30 commands?

meikrantz, 2015-12-23:
I haven't seen any other grbl controllers implement the intermediate point.  I suppose it could be useful for raising Z first, but I will generally just issue a G28 or G30 from console without an intermediate point, or create a macro button that issues a G28 or G30.  If you do decide to give option for intermediate point, probably makes sense to support Z axis more than X or Y.

meikrantz, 2016-01-02:
Looks good, works as expected!

closed 2016-01-04 by meikrantz