cncjs/cncjs #17: Probe support

meikrantz, 2015-12-23:
Would be nice to have a widget that supports probe functionality.  G38.x as well as G43.1 tool offset.  Jarret who has also posted here put together the probe function for the Chilipeppr grbl workspace.

cheton, 2015-12-25:
Taking notes: http://www.instructables.com/id/Probing-and-Milling-a-PCB-using-an-Arduino-Based-C/?ALLSTEPS

meikrantz, 2015-12-27:
For me, autolevel is a bit of overkill.  I just need a single probe to be able to set bit zero height.  More like "touch plate" functionality that does simple G38.2 or G38.3 probe with a G43.1/G49 tool length offset.  

Maybe an option to choose probe type (G38.2 G38.3, G38.4), probe speed, depth to probe, and tool length (for the G43.1 offset).  Button for G49 to clear tool length offset as well.

cheton, 2015-12-31:
Could you please confirm if the example below is a correct usage for Z probing?

--
Assume current unit mode is metric system (mm), the depth to probe is `-10mm`, the probe feedrate is `25mm/min`, the height of touch plate (i.e. tool length offset) is `8mm`, and the retract distance is `2mm`. 
The full commands might look like so:
```bash
G90  # absolute distance mode
G49  # TLO cancel
G92 Z0  # zero out temporary Z axis
G38.x Z-10 F25  # G38.x=G38.2~G38.5, depth-to-probe=-10mm, feedrate=25mm/min
G43.1 Z8  # TLO=8mm
G91  # relative distance mode
G0 Z2  # retract distance is 2mm
G90  # absolute distance mode
```

* For `G38.2` and `G38.3` commands which will probe toward workpiece, the depth to probe should be a nagative value.
* For `G38.4` and `G38.5` comamnds which will probe away from workpiece, the depth to probe should be a positive value.

--

meikrantz, 2015-12-31:
That looks good.  I might add a G10 L20 P1 Z0 after the G43 command, assuming that I am going to be setting the bit depth zero to the CNC bed.  I'm not a fan of the G92 offsets, since they get reset at the end of a job generally.  Once a particular bit is set via probe, Z0 should probably stay constant until next probe or manual change of Z.

cheton, 2015-12-31:
Thanks for your suggestions. I made some changes below as you mentioned. Could you check it again?
```bash
G91  # Set relative distance mode
G49  # Cancel TLO
G38.2 Z-10 F25  # Probe toward workpieces (maximum travel distance is 10mm)
G43.1 Z8  # Set TLO to 8mm (= the height of touch plate)
G10 L20 P1 Z0  # Zero Out Work Offsets for G54
G0 Z2  # Retract slightly from the touch plate
G90  # Set absolute distance mode
```

meikrantz, 2015-12-31:
Yes, that looks good.  I tend to use G38.3 more than G38.2, and have never used the .4/.5 options.

cheton, 2015-12-31:
Great! It's available in the latest release [v0.14.0](https://github.com/cheton/cnc.js/releases/tag/v0.14.0).

meikrantz, 2015-12-31:
Looks great!  I will have a chance to test later today.

meikrantz, 2015-12-31:
Had a chance to test, and it's not working properly.  You changed to G91 at the start in the production code, which gives incorrect results.  Also the TLO setting should be negative and not positive.

So this should be correct:

G90  # Set absolute distance mode
G49  # Cancel TLO
G38.2 Z-10 F25  # Probe toward workpieces (maximum travel distance is 10mm)
G43.1 Z-8  # Set TLO to -8mm (= the height of touch plate)
G10 L20 P1 Z0  # Zero Out Work Z axis for G54
G91 # Set relative distance mode
G0 Z2  # Retract slightly from the touch plate
G90  # Set absolute distance mode

cheton, 2016-01-01:
Thanks for correcting me. There is one thing just got confused.
In above example of absolute distance mode, `G38.2 Z-10` will move down to the negative Z-axis until the `Z-10` position has reached. If the starting position is `Z15`, the travel distance will be `25mm` (15 - (-10) = 25) at maximum.
Is that a expected behavior for the definition of "Depth to Probe"? I used to think the "Depth to Probe" should be a relative value, not the absolute one.

cheton, 2016-01-01:
Just tried the Touch Plate widget on [ChiliPeppr](http://chilipeppr.com/grbl). A `G92 Z0` command is sent prior to `G38.2`. Should I zero out temporary offset on Z axis before running the straight probe?
```
 G21G90
 G49
 G92Z0
 G38.2Z-10F25

meikrantz, 2016-01-01:
I think the axis parameter for G38.x is always relative, regardless of the distance mode (G90/G91).
As far as the touch plate widget goes, I'm not a big fan of using G92 offsets and I think @chamnit (grbl developer) tries to discourage their use.  That being said, I think a lot of folks use them.

cheton, 2016-01-01:
I just found an article of your past discussion here :)
https://groups.google.com/forum/#!msg/chilipeppr/7SoMcKPNHpY/Npl2AmOAE4MJ

Maybe I can turn it into this?
```bash
G49 # Cancel TLO
G91 # Set relative distance mode
G38.2 Z-10 F25 # Probe toward workpieces (maximum travel distance is 10mm)
G90 # Set back to absolute distance mode
G10 L20 P1 Z0 # Zero Out Work Z axis for G54
G43.1 Z-8 # Set TLO to -8mm (= the height of touch plate)
G91 # Set relative distance mode
G0 Z2 # Retract slightly from the touch plate
G90 # Set back to absolute distance mode
```

meikrantz, 2016-01-01:
Ha!  Never did figure out why sometimes that release of grbl would move the z axis in the wrong direction periodically!  Your workflow above should work, but I don't think you need the G91 before the probe.  Won't hurt anything, I guess.  I did some testing earlier today, and it worked for me as expected without the G91, ie the G38.x parameter always seemed to be relative, even if grbl was in absolute mode.

cheton, 2016-01-01:
May I know which Grbl version you're using? I tested it with `Grbl 0.9j` and got the following result:
```
Grbl 0.9j ['$' for help]
> ?
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
> G91
> G0 Z5
> G90
ok
ok
ok
> ?
<Idle,MPos:0.000,0.000,5.000,WPos:0.000,0.000,5.000>
> G38.2 Z-10 F20
ALARM: Probe fail
[PRB:0.000,0.000,0.000:0]
ok
> ?
<Alarm,MPos:0.000,0.000,-10.000,WPos:0.000,0.000,-10.000>
```

In absolute mode, if I set the starting position to Z5, after probing with `G38.2 F-10 F20`, the final Z position is `Z-10`, not the `Z-5` as expected to be relative.

meikrantz, 2016-01-01:
I'm using 0.9i.  If you put it in relative mode with G91, does it work properly?  Not sure how G38 was implemented in grbl.  I thought in linuxcnc, you only had to specify the axis and speed without distance.  Since G38 specifies either toward or away from workpiece, I would have thought it always to be relative.

cheton, 2016-01-01:
Hmmm. I flahsed my Arduino UNO to Grbl 0.9i with factory default settings, and still got the same result.

I just published [v0.14.1](https://github.com/cheton/cnc.js/releases/tag/v0.14.1) with the fixes mentioned above, the `G91` is added before the probe command `G38.x` to ensure relative movement.

meikrantz, 2016-01-01:
Awesome, I'll give it a try tomorrow.  Happy New Year!

cheton, 2016-01-01:
Happy New Year! :yum:

meikrantz, 2016-01-01:
Works perfectly!

closed 2016-01-02 by cheton