cncjs/cncjs #19: Raspberry Pi

Protoneer, 2015-12-31:
Hi. Will this work on a Raspberry Pi B+?

I installed it with NPM but it failed...

meikrantz, 2015-12-31:
Protoneer, I have it running on RPi B+ and RPi 2, both with Wheezy and Jessie.  I had some weirdness with packages at first, but can't remember the details.

What are the errors you are seeing?

Protoneer, 2016-01-01:
I tried installing it from npm but there were issues with the Serial lib.

Will try and get the logs for you...

cheton, 2016-01-01:
Which version of Node.js you are running? If you are running Node v4 or v5, you need to use gcc/g++ to v4.8 or later to compile serialport module correctly.

meikrantz, 2016-01-01:
I had to install serialport directly at one point during the cnc.js version upgrades on one of my systems.  Everything is working fine on two different Pi's, one running node v0.12.1 and one running node v5.2.0.

On the node v5.2.0 version, I couldn't upgrade to cnc.js v0.14 until I uninstalled the previous version, then it worked fine.

Protoneer, 2016-01-01:
Hi @cheton , @meikrantz . Lets start fresh. I want to create an img file for Raspberries to go with my RPI CNC board. Currently I have bCNC is working but I would like to add CNC.js as an alternative.(Scripts I have created to setup the RPI environment before I create an image : https://github.com/Protoneer/RPI-CNC-Config-Scripts/tree/master/scripts)

Looking at the Wiki it's not clear what version on NodeJS I need for CNC.js. RPI Jesse currently ships with 0.10.29... What version is best to target?

Protoneer, 2016-01-02:
Below is my install logs...

I manged to upgrade Node to 0.12.6 from Adafruits tutorial.
https://learn.adafruit.com/node-embedded-development/installing-node-dot-js

Then did "sudo npm install -g cncjs"  with the following errors.(Takes a very long time to run)

Am I missing a step?

```
pi@raspberrypi:~ $ sudo npm install -g cncjs
npm WARN peerDependencies The peer dependency history@^1.17.0 included from reac                                                                                        t-router will no
npm WARN peerDependencies longer be automatically installed to fulfill the peerD                                                                                        ependency
npm WARN peerDependencies in npm 3+. Your application will need to depend on it                                                                                         explicitly.
npm WARN engine hawk@0.10.2: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6"                                                                                        ,"npm":"2.11.2"})

> react-router@1.0.3 postinstall /usr/local/lib/node_modules/cncjs/node_modules/                                                                                        react-router
> node ./npm-scripts/postinstall.js

\
> bufferutil@1.2.1 install /usr/local/lib/node_modules/cncjs/node_modules/socket                                                                                        .io/node_modules/engine.io/node_modules/ws/node_modules/bufferutil
> node-gyp rebuild

 user "root" does not have permission to access the dev dir "/root/.node-gyp/0.1                                                                                        2.6"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/                                                                                        node_modules/cncjs/node_modules/socket.io/node_modules/engine.io/node_modules/ws                                                                                        /node_modules/bufferutil/.node-gyp"
make: Entering directory '/usr/local/lib/node_modules/cncjs/node_modules/socket.                                                                                        io/node_modules/engine.io/node_modules/ws/node_modules/bufferutil/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
make: Leaving directory '/usr/local/lib/node_modules/cncjs/node_modules/socket.i                                                                                        o/node_modules/engine.io/node_modules/ws/node_modules/bufferutil/build'
npm WARN deprecated i18next-client@1.10.3: you can use npm install i18next from                                                                                         version 2.0.0
npm WARN engine hoek@0.7.6: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6",                                                                                        "npm":"2.11.2"})
npm WARN engine boom@0.3.8: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6",                                                                                        "npm":"2.11.2"})
npm WARN engine cryptiles@0.1.3: wanted: {"node":"0.8.x"} (current: {"node":"0.1                                                                                        2.6","npm":"2.11.2"})
npm WARN engine sntp@0.1.4: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6",                                                                                        "npm":"2.11.2"})

> utf-8-validate@1.2.1 install /usr/local/lib/node_modules/cncjs/node_modules/so                                                                                        cket.io/node_modules/engine.io/node_modules/ws/node_modules/utf-8-validate
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/roo                                                                                        t/.node-gyp/0.12.6"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/                                                                                        node_modules/cncjs/node_modules/socket.io/node_modules/engine.io/node_modules/ws                                                                                        /node_modules/utf-8-validate/.node-gyp"
make: Entering directory '/usr/local/lib/node_modules/cncjs/node_modules/socket.                                                                                        io/node_modules/engine.io/node_modules/ws/node_modules/utf-8-validate/build'
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/usr/local/lib/node_modules/cncjs/node_modules/socket.i                                                                                        o/node_modules/engine.io/node_modules/ws/node_modules/utf-8-validate/build'
-
> utf-8-validate@1.2.1 install /usr/local/lib/node_modules/cncjs/node_modules/so                                                                                        cket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules                                                                                        /ws/node_modules/utf-8-validate
> node-gyp rebuild
[... cut ...]
npm ERR! Failed at the serialport@2.0.5 install script 'node-pre-gyp install --f                                                                                        allback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log

```

cheton, 2016-01-02:
Hi Protoneer,

You can try adding `prefix=~/.npm` to your `~/.npmrc`. For example:
```bash
[cheton@piduino:~]
$ echo "prefix=~/.npm" >> ~/.npmrc

[cheton@piduino:~]
$ cat ~/.npmrc 
prefix=~/.npm
```

After that, you can install `cncjs` without using `sudo`:
```bash
[cheton@piduino:~]
$ npm install -g cncjs
```

This should resolve your installation problem on Raspberry Pi.

cheton, 2016-01-02:
I updated the installation guide in README.md.
https://github.com/cheton/cnc.js#installation

Protoneer, 2016-01-02:
@cheton ,  now we are cooking with gas!!!

Your suggestions solved the errors. Only issue I am facing now is that running "cnc" returns "Command not found"... :)  I have restarted the RPI as well.

Any clue what that could be?

cheton, 2016-01-02:
It might be installed inside the ~/.npm/node_modules/ directory. If so, append ~/.npm/bin to your PATH environment in ~/.bash_profile or ~/.bashrc, like below:

export PATH=$PATH:~/.npm/bin/

Protoneer, 2016-01-03:
Thanks @cheton. I managed to get it going. for some reason I can't go "cnc" I have to go 
"./cnc" or "node cnc".

Otherwise I have connected to GRBL and moved things around.

cheton, 2016-01-03:
Good. Fell free to let me know if you have further questions or any issue encountered.

Protoneer, 2016-01-03:
Thanks @cheton ... I am surprised that I have only seen CNC.JS now. It seems very polished and well thought out. Great job!!!

Protoneer, 2016-01-04:
I am back again.... :(

I did a fresh install and I now have another problem installing it. Looks like Socket.IO is having issues installing....
```
npm WARN peerDependencies The peer dependency history@^1.17.0 included from react-router will no
npm WARN peerDependencies longer be automatically installed to fulfill the peerDependency
npm WARN peerDependencies in npm 3+. Your application will need to depend on it explicitly.
npm WARN engine hawk@0.10.2: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6","npm":"2.11.2"})

> react-router@1.0.3 postinstall /home/pi/.npm/lib/node_modules/cncjs/node_modules/react-router
> node ./npm-scripts/postinstall.js

npm WARN deprecated i18next-client@1.10.3: you can use npm install i18next from version 2.0.0
npm WARN engine cryptiles@0.1.3: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6","npm":"2.11.2"})
npm WARN engine sntp@0.1.4: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6","npm":"2.11.2"})
npm WARN engine boom@0.3.8: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6","npm":"2.11.2"})
npm WARN engine hoek@0.7.6: wanted: {"node":"0.8.x"} (current: {"node":"0.12.6","npm":"2.11.2"})
|
> serialport@2.0.5 install /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
  COPY /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build/Release/node-v14-linux-arm/serialport.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build'
npm ERR! Linux 4.1.13+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "cncjs"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2

npm ERR! shasum check failed for /tmp/npm-23332-3c063b35/registry.npmjs.org/socket.io-parser/-/socket.io-parser-2.2.4.tgz
npm ERR! Expected: f9ce19bf1909608ceb15d97721e23bfdd1e7cf65
npm ERR! Actual:   e188bb3c7f0e4175c0a2268fb168868874061efe
npm ERR! From:     https://registry.npmjs.org/socket.io-parser/-/socket.io-parser-2.2.4.tgz
npm ERR!
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/RPI-CNC-Config-Scripts/npm-debug.log

```

cheton, 2016-01-04:
It looks weird. Could you check if you have enough disk space to download, or try uninstalling cncjs first using `npm -g uninstall cncjs`.

cheton, 2016-01-04:
It shouldn't have checksum failures unless the tarball file was not saved properly. Try to delete all files under the /tmp directory if it is almost full.

Protoneer, 2016-01-04:
Finally resolved it....

All of my tests were with a clean version of Jesse and fully expanded 8gb SD cards.

Seems like the issue was with Socket.IO . Below is the process I followed to install it.
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get -y remove nodejs
curl -sLS https://apt.adafruit.com/add | sudo bash
sudo apt-get install node

echo "prefix=~/.npm" >> ~/.npmrc
cat ~/.npmrc
npm install -g socket.io
npm install -g socket.io-parser
npm install -g cncjs
export PATH=$PATH:~/.npm/bin/
```

closed 2016-01-04 by Protoneer

cheton, 2016-01-04:
Both `socket.io` and `socket.io-parser` are not necessary to install globally, they will be installed as the package dependency of `cncjs`.

It's recommended that you uninstall previous version of `cncjs` if you want to upgrade to a newer version: `npm uninstall -g cncjs && npm install -g cncjs`.

To make updates much easier, I'd consider to use [electron](https://github.com/atom/electron) to build cross platform app in the future.

Protoneer, 2016-01-04:
Yes, for some reason the socket stuff that is part of CNC.js's dependencies did not want to install(As per the log above).

I have been playing with Electron and it seems capable. I also like the idea of the AutoUpdated (Unfortunately there is no Linux support for it yet).

Thanks for the help.