cncjs/cncjs #20: Shuttle Express jog dial support

meikrantz, 2016-01-01:
One of the other features that I really liked in Chilipeppr was the ability to use a Shuttle Express USB jog dial for machine shuttle.  @lordmundi wrote the code specifically to communicate via the Serial Port JSON Server, which is used by Chilipeppr.  Take a look at the project here:  https://github.com/lordmundi/shuttleCP

Not sure how easy it would be to change it to use web calls instead of JSON.

All this being said, I think there are plans for a jogging mode/interface in upcoming grbl releases.

lordmundi, 2016-01-01:
It should be pretty easy to convert my codebase over to something that works with cnc.js, as my code is a standalone program that is just reading the jog dial and then sending G-code over a websocket to command the machine.  If your code has a websocket server that can accept commands, then it should be pretty trivial to port.

cheton, 2016-01-04:
This may take a while because I don't have the Shuttle Express Jog Dial. It will be great if someone can provide support for the porting. :smiley:

lordmundi, 2016-01-04:
> It will be great if someone can provide support for the porting. [image:

I can provide support. And, we could probably get some folks to chip in
some money to get you a jog dial.

FG

cheton, 2016-01-04:
Just did a quick search and found the jog dial sold by Amazon. Is this the one that's supported by ShuttleCP? 
http://www.amazon.com/gp/product/B0032Y0OH0?psc=1&redirect=true&ref_=ox_sc_act_title_1&smid=A1YG1VZVF48PM9

lordmundi, 2016-01-04:
yeah, that's it.  i'd be happy to donate $20 toward you getting one.

I should also mention that most of my work was to get it to read the jog
dial under linux and that was running on a different machine than my
chilipeppr instance.  If you have the jog dial running on the same machine
as your cnc.js instance, its possible it might be way easier.  For example,
i think there is some way to get the jog dial to simply reproduce key
strokes when you hit the various buttons and things (I think this is most
certainly how people use it in windows). If you had keyboard shortcuts for
all your jog motions, it's possible all you would need to do is setup the
jog dial to use those key-strokes just like folks do with video editing and
things.

That would be a more fool-proof way to integrate things... i just didn't do
that with shuttleCP because my jog dial wasn't on the same machine as
Chilipeppr and chilipeppr didn't have keyboard shortcuts for everything.

FG

meikrantz, 2016-01-04:
I would also be happy to chip in some money to get you a unit!

cheton, 2016-01-05:
Thanks for your kindly help. The jog dial looks interesting and the price is affordable for me, so I'd like to buy one myself. I will keep you updated on progress once I received the package.

lordmundi, 2016-01-05:
Send me your PayPal address and I'll send over $20. We need more CNC apps
to support how dials and get away from keyboards! :)

FG

cheton, 2016-01-05:
I created a donation button at the bottom of the README (https://github.com/cheton/cnc.js#donate). Thanks for your support on the hardware investment.

lordmundi, 2016-01-05:
Ok, I just sent a donation. Let me know if you didn't get it.

FG

cheton, 2016-01-05:
I just received your payment. Thank you for making a donation.

cheton, 2016-01-06:
I placed the order on Amazon on Jan 4. It will be delivered to Shipito's warehouse, and then forwarded to Taiwan. The expected arrival date might be in late January. Also thank @meikrantz for your donation.

lordmundi, 2016-01-06:
Cool. Definitely look into using it as keyboard shortcuts first as I think
that is better and more in line with how other programs use that hardware.
I'm not sure, but I think the manufacturer has utilities to map the
keystrokes on their website.

FG

cheton, 2016-01-07:
So, before I started to integrate your shuttleCP, the first thing I can do is to provide a configuration UI to configure keyboard shortcuts that map to certain buttons (e.g. jog motions) on cnc.js. This would be the most easiest way to control the UI by ShuttleExpress jog dial.

lordmundi, 2016-01-07:
yep, i think so... keys to switch the active axis, to switch the jog
increment, and to move positive or negative on the selected axis.  then,
for the jog wheel, you'll need to configure it somehow to just send the key
over and over at different speeds or maybe something else... not sure.  but
for all the other buttons and the main rotary wheel it should be axis
select, speed select, and move forward and move backward.

FG

cheton, 2016-01-09:
Found an interesting CNC Simulator at http://cncsimulator.info/screenshots.
It provides support for ShuttleExpress as well: http://cncsimulator.info/blog/?p=413

lordmundi, 2016-01-09:
Very cool. So are there any utilities to configure the keys for Linux?

FG

cheton, 2016-01-24:
Added some keyboard shortcuts to [v0.15.2](https://github.com/cheton/cnc.js/releases/tag/v0.15.2):
`!` - Feed Hold
`~` - Resume
`ctrl + alt + command + h` - Homing
`ctrl + alt + command + u` - Unlock
`ctrl + alt + command + r` - Reset
`ctrl + alt + command + =` - Switch Jog Distance: 1, 0.1, 0.01, 0.001, or custom
`ctrl + alt + command + [` - Jog Backward
`ctrl + alt + command + ]` - Jog Forward
`ctrl + alt + command + x` - Select/Deselect X Axis
`ctrl + alt + command + y` - Select/Deselect Y Axis
`ctrl + alt + command + z` - Select/Deselect Z Axis

My Shuttle Express Settings just look like the following:
* Button 2: `ctrl + alt + command + x`
* Button 3: `ctrl + alt + command + y`
* Button 4: `ctrl + alt + command + z`
* Button 5: `ctrl + alt + command + =`
* Turn Jog Left: `ctrl + alt + command + [`
* Turn Jog Right: `ctrl + alt + command + ]`

For the Shuttle Wheel support, it might need a mechanism to count the number of keystrokes per second to determine Shuttle Zone (-7 to +7) and G1 parameters like feedrate and distance. I will try to find a better way to add Shuttle Wheel support to the web interface.

lordmundi, 2016-01-24:
Awesome.

For the wheel, you could just fake it. That's what I'm doing in that
shuttleCP. Basically setting some parameters for how far to move for each
update and then continually sending them. This could be a configurable
parameter in the jog dial settings. IIRC, I have one at the top of my code.

This is great! Can't wait to see a video of it working.

FG

meikrantz, 2016-01-26:
Okay, had a chance to test this out this evening.  It works, but performance is not that great.  Buttons select the appropriate axis, and highlight it in the GUI, which is nice.  The problem is with buffering.  You have to turn the jog wheel very slowly, or it skips quite a bit.  That would be okay with fine movements, but without shuttle wheel support for large movements it's somewhat limiting.

Good start though!  Nice to actually have the ShuttleXpress make the machine move again!

cheton, 2016-01-26:
I haven't implemented the buffering mechanism, but only use a simple flag `canJog` to check if it needs to send commands to Grbl.
```js
'JOG_FORWARD': () => {
    let { port, activeState } = this.props;
    let canJog = (!!port && (activeState === ACTIVE_STATE_IDLE));

    if (canJog) {
        let distance = this.getJogDistance();
        let jog = {
            'x': () => this.jog({ X: distance }),
            'y': () => this.jog({ Y: distance }),
            'z': () => this.jog({ Z: distance })
        }[this.state.selectedAxis];

        jog && jog();
    }
}
```

Indeed, current implementation will lose steps if current state is not `Idle`, that is originally developed to prevent accidentally ShuttleXpress touches while running a G-code file. That will be solved in a later version to make smooth movement, and the shuttle wheel support will be provided as well.

cheton, 2016-02-01:
The latest update [v0.15.4](https://github.com/cheton/cnc.js/releases/tag/v0.15.4) should has resolved the jog wheel performance issue, and it also includes shuttle wheel support.

https://github.com/cheton/cnc.js/wiki/User-Guide#shuttlexpress-settings

### ShuttleXpress Settings

#### Buttons
* Button 2 - Select/Deselect X Axis<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>x</kbd>
* Button 3 - Select/Deselect Y Axis<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>y</kbd> 
* Button 4 - Select/Deselect Z Axis<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>z</kbd> 
* Button 5 - Switch Jog Distance (1, 0.1, 0.01, 0.001, or a custom value)<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>=</kbd> 

#### Jog Wheel
* Turn Jog Left<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>[</kbd> 
* Turn Jog Right<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>]</kbd> 

#### Shuttle Wheel

Adjust the keystroke repeat rate to **10 times per second** for all Shuttle Zones except the Shuttle Zone 0.

* Shuttle Zone -7<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>7</kbd>
* Shuttle Zone -6<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>6</kbd>
* Shuttle Zone -5<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>5</kbd>
* Shuttle Zone -4<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>4</kbd>
* Shuttle Zone -3<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>3</kbd>
* Shuttle Zone -2<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>2</kbd>
* Shuttle Zone -1<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>shift</kbd> + <kbd>1</kbd>
* Shuttle Zone 0<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>0</kbd>
* Shuttle Zone 1<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>1</kbd>
* Shuttle Zone 2<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>2</kbd>
* Shuttle Zone 3<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>3</kbd>
* Shuttle Zone 4<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>4</kbd>
* Shuttle Zone 5<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>5</kbd>
* Shuttle Zone 6<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>6</kbd>
* Shuttle Zone 7<br>
  <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>7</kbd>

cheton, 2016-02-01:
Hi @lordmundi,

According to your `shuttlecp.c` at [#L201](https://github.com/lordmundi/shuttleCP/blob/master/shuttlecp.c#L201), there is an OVERSHOOT factor for distance calculation. May I know if it is necessary while using the shuttle wheel?

Here is my code:
https://github.com/cheton/cnc.js/blob/master/src/web/components/widgets/axes/JogPad.jsx#L93

lordmundi, 2016-02-01:
this is fantastic!!! got any videos of it working?  it's not real till we
see a video.  :)

for the overshoot, the reason is because of the acceleration / deceleration
profiles.  If you time it such that the next command begins as the previous
one ends, then the controller will ramp up, get to top speed, then start
ramping down and then eventually hit the desired mark and stop.  Then the
next command would come and it would do that again.  So you would get this
"rarr...rarr...rarr...rar...." sound (sorry to use such technical terms
here).  The idea is to get the next command in before it started
decelerating so that it would continue on at top speed and not decelerate
until the commands stopped coming in when you let go of the wheel.  I
experimented and found that an overshoot of about 6% was giving me good
results.

That's all it was.  If that doesn't make sense, let me know and I'll try to
explain it differently.

These great keystrokes and additions should make code like mine not
needed... this is really the better way to do things.  Plus, this should
work for any OS i believe.

FG

cheton, 2016-02-01:
I just connected it to my homemade CNC machine, and it sounds so horribl just like what you mentioned.
 (rarr...rarr...rarr... :scream_cat: :scream: )  XDDD

[image]
 
However, giving a small amount of overshoot factor (e.g. 1.06~1.2) did not help much for my test. I may need to figure out where I can improve to eliminate deceleration between subsequent commands.

lordmundi, 2016-02-01:
> (rarr...rarr...rarr... [image: :scream_cat:] [image: :scream:] ) XDDD

haha... yeah, i think it took me a bit to calculate just how far to command
for each "cycle" of the software, which for you will depend on the
keystroke frequency I suppose.  Then, you just need to tune that amount to
have a certain amount of overshoot which is dependent on the speed.

Feel free to steal any of my code that helps.

FG

cheton, 2016-02-02:
I guess a buffering delay solution plus the overshoot factor might work by accumulating each small steps and use a large flush interval, may be a 0.5s time interval or longer. I will make more tests to find a way that can significantly reduce frequent deceleration.

cheton, 2016-02-12:
I have an interesting finding. I eventually discovered that the sharp sound during acceleration and deceleration can be significantly reduced with a lower feed rate (mm/min) and a higher acceleration speed (mm/sec^2).

My NEMA 23 (57mm) 3V stepper motor can run in full speed without missing steps with following max rate and acceleration settings: 
```
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=50.000 (x accel, mm/sec^2) // Grbl's default value is 10.000 (mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
```

After several tests, I found my optimal values by setting the maximum feed rate from 1500 to 800 (mm/min) and increasing the acceleration speed from 10 to 50 (mm/sec^2), and never heard the humming noise. For example:
```
> G91 G1 F200 X1.33  # Shuttle Zone 1
> G90
   :   :   :
> G91 G1 F800 X4  # Shuttle Zone 7
> G90
ok
```

Since the settings may show different results between different hardwares, I may provide a popup menu for configuring min/max feed rate value (e.g. 200~800) for Shuttle Wheel to meet different needs.

cheton, 2016-02-12:
The sample test code: ShuttleZone.jsx
https://github.com/cheton/cnc.js/blob/master/src/web/components/widgets/axes/ShuttleZone.jsx#L30

lordmundi, 2016-02-12:
so does that mean you are adjusting your grbl config parameters?  or are you adjusting the max feed rate being commanded in the GCODE?

if you have an amount of overshoot that you can configure, then that would give the user one single "knob" to adjust to get the motion of their machine smooth without too much extra travel.

cheton, 2016-02-13:
May I know your Grbl system settings? The overshoot factor didn't help much with my CNC machine,
so I adjusted both the x/y/z acceleration speed (in Grbl) and the max feed rate (e.g. 800 mm/min) sent with the G-code, and finally get the expected result. For my test, the timing control is handled by web browser, which is different from ShuttleCP, I guessed it might be related to some timing accuracy issues between web browser and the server.
I'd also like to add overshoot factor as a configurable option, it should help with different scenarios.

meikrantz, 2016-02-13:
I haven't had a chance to do more testing, but wanted to share my grbl settings.  In order to get smooth movement on my machine, I greatly increased the max rate and acceleration.  Looking forward to getting back to the shop to further test the shuttle!

$110=10000.000 (x max rate, mm/min)
$111=10000.000 (y max rate, mm/min)
$112=1500.000 (z max rate, mm/min)
$120=300.000 (x accel, mm/sec^2)
$121=300.000 (y accel, mm/sec^2)
$122=150.000 (z accel, mm/sec^2)

cheton, 2016-02-14:
The latest update [v0.15.6](https://github.com/cheton/cnc.js/releases/tag/v0.15.6) should have fixed the issue of humming noise when scrolling the Shuttle Wheel. You can configure feed rate range, repeat rate, and distance overshoot by clicking the <kbd>Edit</kbd> button at the top of the Axes widget. 

I also made an update to the wiki: [Reduce humming sound when accelerating and decelerating](https://github.com/cheton/cnc.js/wiki/User-Guide#reduce-humming-sound-when-accelerating-and-decelerating)

Let me know if you have any suggestions for further improvements.

meikrantz, 2016-03-13:
Finally had a chance to get in the shop and do some testing with 0.15.8 and ShuttleXpress.  It works great!  Haven't tried tweaking the overshoot and acceleration parameters, but movement is very smooth.

cheton, 2016-03-14:
It's great to hear that. :smile:

closed 2016-03-20 by cheton

slimline33, 2020-06-19:
Hello, everyone,

**I'm really sorry to bring up this old issue again**. But I have some general questions about the Shuttle Express...

I have CNCJS running on a Raspberry 3. 

Am I right in assuming that I simply plug the ShuttleExpress into the server USB port and don't have to open the browser window from the client PC? Do I need any drivers to control the Shuttle?

Or does it only work under Windows in the browser with the ShuttleExpress software? Like this:

[image]

I already tried to check it on Facebook but I don't think anyone understands me :)

I hope someone has an answer.

Regards Christoph

MitchBradley, 2020-06-20:
According to my reading of the code, the Shuttle Express support is in the client-side - browser - code, not in the server code.  The Shuttle Express feeds simulated keystrokes to the browser-based UI, which then interprets them and sends control commands to the server over the network connection. Plugging it into the computer that runs the CNCjs server - unless the browser is running on the same computer - would not be effective. 

I expect that the reason why you are getting little response on the Facebook page is because very few CNCjs users have a Shuttle Express.  I certainly don't have one.

karoria, 2020-06-20:
@slimline33 
I agree to @MitchBradley . I have a ShuttleExpress. One of the most important thing is, it doesn't send its keystrokes unless you configure it to do so. As far as my knowledge, it supports windows and mac only (I mean its configuration software). That means, that software always runs in background while you use shuttleexpress from windows or mac client and maps your inputs to relevant keystrokes. For Linux (client), I had no success with my little efforts but there might be some short of workaround to map it with keystrokes or some efforts are required to do so.
In short, you can use shuttleexpress with windows and mac running browsers (as clients) while your server part is in RPi3. I hope this is helpful.

slimline33, 2020-06-20:
Hello @MitchBradley and @karoria . Thank you so much for your answers.

That's what I suspected. But I wonder why in CNCJS settings there is a possibility to set the feedrate and the repeat.

What surprised me is that @Lordmundi described (https://github.com/cncjs/cncjs/issues/20#issuecomment-168703235) that he runs the SHUTTLE EXPRESS on the same machine as CNCJS. I think he means that he also controls the machine via the servers browser.

Unfortunately I have my problems with the Windows keys under Windows 10 when working with the Shuttle Express via the client. For example he wants to open Microsoft OneDrive for me and so on. I would not like to disable the key shortcuts under Windows 10. Who knows if this will remain until the next update.

lordmundi, 2020-06-20:
just to clear up any confusion, I'm not running CNCJS, but I did write some code that read the shuttle express on linux and output commands to some other cnc utilities.  I think you could adapt it if you want.  That project is here: https://github.com/lordmundi/shuttleCP

slimline33, 2020-06-21:
Hi @lordmundi 
Thank you for your answer.

I already know your code. But I did not understand the context of this issue. I assumed that @cheton had implemented parts from your code in CNCJS. I must have misunderstood that!