cncjs/cncjs #22: Probe settings persistence

meikrantz, 2016-01-04:
Is it possible to make the probe settings persistent?  The offset value always seems to default to 10mm instead of the last entered value.  For instance, I am using a touch plate that is 0.02mm thick, so I have to reset that every time I relaunch cnc.js.

cheton, 2016-01-04:
I also found it quite disturbing. Will fix it later today.

cheton, 2016-01-04:
Fixed in [v0.14.3](https://github.com/cheton/cnc.js/releases/tag/v0.14.3) using HTM5 Local Storage.

It can remember both imperial and metric unit settings. You can try to change length units by clicking on the `[in / mm]` button on the Axes widget or typing G20 or G21 commands from the Console widget to verify if it is working.

meikrantz, 2016-01-05:
Works for me.
  
Quick semi related question.  When I went to update cncjs via npm from 14.2 to 14.3, the 'npm update -g cncjs' command didn't do any updating.  Just exited with no messages after a period of time.  I had to do an uninstall first, then reinstall to get the new version.

cheton, 2016-01-05:
Not sure what happened. It might be related to NPM registry cache or some other stuffs.

closed 2016-01-05 by meikrantz