cncjs/cncjs #23: Question: Beyond GRBL

jarretluft, 2016-01-04:
Hey @cheton, just wanted to throw an idea at you and see what you thought.

I'm wondering if you have any suggestions on how a user could more easily configure cnc.js to work with other cnc controllers.  Do you think it is possible to override all of the grbl specific code through a config file so that controllers that don't understand things like $H, ~, ?, etc could be replaced with their own version of these commands?

I ask because along with my grbl controller I have an arduino DUE running TinyG2, and a smoothieboard controller.  These controllers could theoretically be controlled by cnc.js if the code was updated to handle their specific commands instead of GRBL's, but right now it looks like the only way to do this would be to search through your code for all instances of grbl specific commands and update with their equivalent on the other two boards.  As well, the protocol for parsing the responses and handling their gcode buffer/queue would need to be implemented.

Do you have any ideas on an easier way to expand cnc.js beyond just GRBL?  I think their may be an opportunity to develop the client side (web interface) independently of any one controller's code, and then create different server.js files to handle those controller specific commands, queues, status updates, etc for additional controller boards, but you would know better than me if this is feasible or if it would create more problems than it is worth. :)

I would appreciate any ideas you have, or if that is not a direction you want to take cnc.js I would understand as you may not want to add complexity to handle controllers that you can't test first-hand!

Thanks,
-J.

Protoneer, 2016-01-04:
@jarretluft ... https://github.com/openhardwarecoza/LaserWeb is also heading that way as is bCNC.

cheton, 2016-01-05:
The future direction will go on that way. Right now most client widgets doesn't highly depend on Grbl, I'd say it is doable by making an abstraction interface between the client side and the server side, that will allow to work with different controllers by sending vendor-specific commands (e.g. ~, !, ?, $G) directly from server instead of web interface.

Otherwise, I do have a plan to completely rewrite the server (i.e. `server.js`), and will consider making support for both Grbl and TinyG2 using same client-to-server communication interface but providing with different server implementations. However, adding support for TinyG2 might have to wait for the release of v2, there are several tasks that have to complete in the first stable release of v1.

A rough roadmap for the upcoming releases:
* v1
  * Widget administration UI
  * Client/Server abstraction
  * Refactoring server code
  * Ready for i18n
  * Documentation
* v2
  * TinyG2 support
  * Shuttle Express jog dial support (Note. may move to v1 if I can made some progress on that :sweat_smile: )

cheton, 2016-01-20:
Just finished a complete rewrite of CNC server in ES6, and the controller interface is ready as well. Now it will be much easier to add new controller support at [here](https://github.com/cheton/cnc.js/tree/master/src/app/controllers).

In addition, I also made a purchase of [gSheild v5](http://synthetos.myshopify.com/products/gshield-v5) few days ago. I will have my Arduino Due running TinyG2 once I received newly purchased packages, and try to figure out how to make it work with `cnc.js`.

hareedy, 2016-06-14:
Any news on the TinyG2 support?

cheton, 2016-06-15:
Hi @hareedy,

I'm currently busy with another project but I might get to it in a month or so. Currently I can run TinyG on an Arduino Due, but it will be helpful if there were code samples that can parse TinyG responses. That will save me a lot of time. 😄

hareedy, 2016-06-15:
Thank you @cheton , the chilipeppr project by John Lauer does support TinyG2 and it handles it well so far,  I guess the TinyG widget repo would be a nice kickstart.
https://github.com/chilipeppr/widget-tinyg

cheton, 2016-06-16:
@hareedy 

Thanks a bunch, I really appreciate it!

This [onRecvCmd function](https://github.com/chilipeppr/widget-tinyg/blob/master/widget.js#L743) in [widget.js](https://github.com/chilipeppr/widget-tinyg/blob/master/widget.js) contains useful information for parsing TinyG responses. :+1:

hareedy, 2016-06-16:
@cheton glad to contribute, I have to say chilipeppr is awesome but we need an offline solution as the desktop version of cncjs, i hope it will support the G2 as the amazing support of GRBL.
Thank you.

cheton, 2016-07-06:
I'm currently working on this. Stay tuned for more updates. :)

cheton, 2016-07-22:
Some notes of TinyG/TinyG2:
https://github.com/synthetos/TinyG/wiki/Tinyg-Communications-Programming
https://github.com/synthetos/TinyG/wiki/TinyG-Status-Codes
https://github.com/synthetos/TinyG/wiki/TinyG-Status-Reports
https://github.com/synthetos/g2/wiki/Job-Exception-Handling

cheton, 2016-07-29:
The support for TinyG2 is now available in the `master` branch. I will run some tests with G-code samples in the following days to make sure current implementation will work as expected.

cheton, 2016-08-03:
It's almost done. The v1.5.0 release will be published in the next few days.

cheton, 2016-08-05:
I just published the cnc v1.5.0 release to NPM. It's been tested with TinyG2 v0.97 firmware running on the Arduino Due, but I'm sure if it will work with other firmware builds or hardware platforms. Please post issues here if you see any problems.

### Connection widget

Select <b>TinyG2</b> before establishing a serial port connection

[image]

### TinyG2 widget

You can view the real-time status in the TinyG2 widget

[image]

closed 2016-08-19 by cheton