cncjs/cncjs #25: Small versioning request

meikrantz, 2016-01-08:
Can you identify the version number in the web interface somewhere?  I've gotten confused a couple of times with multiple systems running :-)

cheton, 2016-01-08:
How about adding a version string to HTML title? You can quickly check the version on browser tab.

meikrantz, 2016-01-08:
That would work, or even something like adding it to the Settings dropdown as the first item in the list.

cheton, 2016-01-09:
Added version number in [v0.14.6](https://github.com/cheton/cnc.js/releases/tag/v0.14.6).

You can check the version number on the browser tab, or hovering the mouse over the string of "CNC.js" on the top-left corner.

I will do several changes to the Settings dropdown in the future, so I just kept it unchanged in current version.

closed 2016-01-09 by meikrantz