cncjs/cncjs #27: windows binary

lordmundi, 2016-01-25:
Not sure if you have any users that have used this on windows before.  I'm assuming it is possible.

If so, I think I've seen other folks take a node program and package it up into a standalone windows app with an installer and everything which includes node.js.  

Just wanted to request that someone do that for cnc.js.

cheton, 2016-01-25:
As far as I know, both NW.js and Electron can create native desktop application. Do you know which projects on GitHub have done similar things that packaged with native modules like [node-serialport](https://github.com/voodootikigod/node-serialport)? It will help me a lot by using their build workflows.

lordmundi, 2016-01-25:
the NW.js stuff sure sounds cool...

don't know if that is the right way to go or not... but perhaps something
like this might help?  I'm pretty sure this is how the Easel folks are
packaging up their Easel Local Sender node app for controlling the serial
port.  (Just guessing from seeing all the node files getting installed)

https://www.youtube.com/watch?v=M6JaWjZKyW4

I'm not sure if you would want it as a service, since it might mean that
your app would have taken hold of the serial port all the time.  Probably
just want an icon on the desktop to launch it.

FG

cheton, 2016-04-21:
Notes:
* awesome-electron<br>
  https://github.com/sindresorhus/awesome-electron
* electron-builder<br>
  https://github.com/electron-userland/electron-builder
* Use AppVeyor to build Electron app for Windows<br>
  https://github.com/develar/onshape-desktop-shell/blob/master/appveyor.yml
* Use Travis to build Electron app for Linux and OS X<br>
  https://github.com/develar/onshape-desktop-shell/blob/master/.travis.yml
* Publishing Artifacts<br>
  https://github.com/electron-userland/electron-builder/wiki/Publishing-Artifacts

cheton, 2016-04-21:
Just created a [electron](https://github.com/cheton/cnc/tree/electron) branch for ongoing development. Now it's okay to create native packages using electron, but I encountered [extremely high CPU usage of serialport module inside electron](https://github.com/electron/electron/issues/2928). I will try to use [node-serialport-worker](https://github.com/weiway/node-serialport-worker) to mitigate this issue for electron builds.

lordmundi, 2016-04-21:
will this make it possible to publish a windows standalone executable? or a
windows installer?

cheton, 2016-04-21:
Yes, it is possible to package Node.js app into Linux, Mac OS X, and Windows executables (or installers) using [electron-packager](https://github.com/electron-userland/electron-packager) and [electron-builder](https://github.com/electron-userland/electron-builder). For native modules like [node-serialport](https://github.com/voodootikigod/node-serialport), it requires an extra step by using [electron-rebuild](https://github.com/electron/electron-rebuild) to rebuild packages for each platform (e.g. linux, darwin, and win32) and arch (e.g. x64, ia32, etc.). I just made it work on Mac OS X, building Windows executable and installer should be just around the corner.

By now the only issue is the high CPU usage of node-serialport inside electron, I will dig into [node-serialport-worker](https://github.com/weiway/node-serialport-worker) for much more testing.

cheton, 2016-04-21:
Serial Proxy
https://github.com/mlwmlw/esp-editor/blob/master/serial-proxy.js

cheton, 2016-04-29:
A good progress has been made, here is a screenshot that I'm running <b>cnc</b> as a native app on my Windows 10 (x64).

[image]

lordmundi, 2016-04-29:
ooh... looks nice.  can't wait to try this out on my pipo x9

FG

cheton, 2016-04-29:
I have some preview versions [here](https://github.com/cheton/cnc-builds/releases) available for download, that were automatically built by Travis CI and AppVeyor.
 
The <b>[cnc-win32-x64.zip](https://github.com/cheton/cnc-builds/releases/download/cnc-dev-latest/cnc-win32-x64-latest.zip)</b> file contains node_modules paths longer than 260 characters, that could cause problems to some legacy unzip programs, so I'd recommend using [7-zip](http://www.7-zip.org/download.html) to decompress a zip file which contains long file names.

cheton, 2016-04-29:
There are some running CI builds in this repo, I will remove the temporary cnc-builds repo once finished.
* dev branch: https://github.com/cheton/cnc/releases/tag/ci-dev-latest
* master branch: https://github.com/cheton/cnc/releases/tag/ci-master-latest

cheton, 2016-06-06:
Added in v1.2.0

closed 2016-06-06 by cheton