cncjs/cncjs #28: Feature request

morrowsend, 2016-02-02:
Instead of using Node.js, and having to install dependancies, what prevents you from using the same thing this chrome plugin uses for serial: https://chrome.google.com/webstore/detail/gcode-sender/ngncibnakmabjlfpadjagnbdjbhoelom?hl=en  If I'm not mistaken, you can run that plugin on a chromebook with no additional installations (I know I can run it on my windows machine without installing node.js)  you can view the code for the chrome plugin by turning on the developer options in chrome, and tracking down the folder the plugin lives in.

cheton, 2016-02-02:
Thank you for your suggestions.

I'm afraid I might not able to do this by current implementation, a chrome plugin requires only client side scripts, which means I need to do a complete rewrite for client/server communication, and remove the dependencies with Node.js.

Furthermore, a chrome plugin also has a limitation if the browser window was refreshed (F5) or accidentally closed due to some reasons, it will stop current execution of G-code commands and it's not recoverable. Please correct me if I'm wrong.

I'd say it should be another new project for chrome plugin integration, but not belong to the current scope of cnc.js.

cheton, 2016-02-02:
To make installation easier for Windows or Mac, I'd consider using NW.js or Electron to pack a native application running with Node.js. See also #27 for windows binary.

cheton, 2016-06-06:
Windows installer is added in v1.2.0

closed 2016-06-06 by cheton