cncjs/cncjs #29: Feature Request (Webcam Support)

adam2070, 2016-02-02:
Hey @cheton

Congratulations on cnc.js I have been testing it out for a while,  it is fast becoming one of my favourite CNC tools. Some of the things that I would like to see in the future are 

- A Webcam Widget similar to GRBLs webcam support.
- Improve synchronization when using multiple connections when using the local server browser and a remote one things to seem not to sync up

Do you have any plans to set up a patron page for to allow us to support the project?

Keep up the good work

Adam

cheton, 2016-02-03:
Hi Adam,

* The webcam support will be added to the current series of v0.15.x. The next version is going to be v1.0-stable.
* Indeed, in current version, not all operations from one browser will be automatically synchronized to all other browsers, but only the server status and the execution status of G-code files will be reported to all connected browsers.

  For example, assume that you have started running a G-code file in a local server browser, and you want to monitor the G-code running status from remote, the first thing you have to do is re-reload(or refresh) the remote browser window, and connect to the same serial port by clicking on [Open] button from the Connection widget (<b><i>Note. Do not click on [Close] button unless you want to disconnect from Grbl</i></b>).
  After doing that, it will automatically download the G-code file from server, render the G-code tool paths in the 3D visualizer, and show current status in Axes, Grbl, and G-code widgets.

  May I know what synchronization problems you've encountered? Any screenshots or videos including your reproduce steps would help for further improvements.

* There is a [donation link](https://github.com/cheton/cnc.js#donate) at the bottom of the README. I will setup a patron page if desired. :smile:

adam2070, 2016-02-03:
I will record some videos over the weekend if I am still haveing issues. The main problem was that if you stop the gcode running in one browser, it will continue to run on the other.

cheton, 2016-02-04:
Just did a test using one desktop computer and one laptop, I also found the data synchronization issue when using local and remote browsers. I will try to figure out where it went wrong. 

I have just received your donation. Thank you for your support of the project.

cheton, 2016-02-05:
Hi @adam2070
 
May I know how did you setup and access your WebCam, for example, accessing the webcam from an IP address or with WebRTC?

adam2070, 2016-02-05:
Hi.

At the moment, i am using a USB webcam on the raspberry pi. and streaming MJPG-Streamer to stream to VLC. it can stream to a web page but have not tried it out yet. 

There are probably much better software around.

cheton, 2016-02-14:
The latest update [v0.15.6](https://github.com/cheton/cnc.js/releases/tag/v0.15.6) provides Webcam support. For the synchronization issue when using multiple connections, I will create another issue for further tracking.

Let me know if you have any suggestions for any improvements.

closed 2016-03-20 by cheton