cncjs/cncjs #30: CNC.js gets confused by .nc file (Chilipeppr is fine)

sigxcpu76, 2016-02-05:
It looks that the attached file confuses CNC.js 3D display. There are lots of horizontal lines which shouldn't be there. I didn't actually try to send this to the machine, just noticed the display.

Chilipeppr.com and displays it fine, Universal G-Code sender too.

[BitFan.nc.zip](https://github.com/cheton/cnc.js/files/119067/BitFan.nc.zip)

cheton, 2016-02-05:
Thanks for reporting this issue. I will look into what went wrong with the 3D display.

cheton, 2016-02-06:
The [gcode-toolpath](https://github.com/cheton/gcode-toolpath) module has a bug while parsing a G-code file with the following format:
```
G0 X1.1185 Y0.5562
Z0.0394
Z-0.0231
```
The X/Y position will be reset to zero at the 2nd and 3rd line.

I will fix the bug in [gcode-toolpath](https://github.com/cheton/gcode-toolpath) at first, and update the bug fixes to CNC.js as well.

cheton, 2016-02-06:
The bug is fixed, you can check it out with [v0.15.5](https://github.com/cheton/cnc.js/releases/tag/v0.15.5).

sigxcpu76, 2016-02-07:
I can confirm that is displayed correctly now. Thank you.

closed 2016-02-07 by sigxcpu76