cncjs/cncjs #31: Extra webpage with only the jog buttons

mano1979, 2016-02-06:
I have cnc.js running on a raspberry pi with the pitft screen (320x240p). I would like to have only the jog function on that screen. The rest i will do on my pc.

Could you make an extra page on the server that will show only that? For example at http://localhost:8000/pendant ?

cheton, 2016-02-11:
Thank you for your suggestion. May I know which buttons are necessary in your case?
For example: X+ / X- / Y+ / Y- / Z+ / Z- / Step (1|0.1|0.01|0.001), ..., etc.

mano1979, 2016-02-11:
Indeed! Exactly the buttons you mentioned. XYZ + and - and a dropdown menu for selecting  the travel distance per klick.

adam2070, 2016-02-11:
Just an idea but how about having it so you can select just the widgets grouped together without the 3d view and can customize which ones you wish to show.

mano1979, 2016-02-11:
Adam Woodall; this won't work because the widget that has the jog function has a lot more than just the jog buttons. And it is just to big. On my display i have 320x240p. Unless your using an hdmi 7" display, the pi compatible displays don't get much bigger than this.

cheton, 2016-03-03:
Hi @mano1979,

I added a new mount option in [v0.15.8](https://github.com/cheton/cnc.js/releases/tag/v0.15.8), you can use this option to set a mount point to serve static files. For example:
```
$ cnc -h
  Usage: cnc [options]
  Options:
    -m, --mount [<url>:]<path>  set the mount point for serving static files (default: /static:static)
$ cnc -m /static:static
```

There is also a <b>TinyWeb</b> example under the [examples/tinyweb](https://github.com/cheton/cnc.js/tree/v0.15.8/examples/tinyweb) directory. You have to copy the directory to your Raspberry Pi to serve as static files. Let's assume you put it under the directory `/home/tinyweb`.

After that, run cnc with the mount option, like below:
```
$ cnc -m /tinyweb:/home/tinyweb
```

Then, you should be able to see the tinyweb console as shown below at `http://localhost:8000/tinyweb`.

[image]

closed 2016-03-20 by cheton

mano1979, 2016-03-20:
That is amazing! I missed your post from 17 days ago. Sorry for that. Thank you so much for your help!

cheton, 2016-03-20:
The [source code](https://github.com/cheton/cnc/blob/master/examples/tinyweb/app.js) should be easy to read and modify if you already know jQuery. It's freely to change to fit your needs.