cncjs/cncjs #32: Autolevel function like in bCNC.

mano1979, 2016-02-06:
I use autolevelling for every job. And i would love to see it implemented in cnc.js.

cheton, 2016-02-11:
I will take some time to study how Auto Level works from the sample in ChiliPeppr at http://fiddle.jshell.net/jarret/377aat1g/.

mano1979, 2016-02-11:
I understand. If you like, i can bring you into contact with the maker of chilipeppr.

Just let me know...

cheton, 2016-06-16:
ChiliPeppr Auto-Level Widget
https://github.com/chilipeppr/widget-autolevel

nekromant, 2016-11-05:
+1, the only thing missing fature that still makes still stick to bCNC

Smuk3D, 2017-01-01:
+1, i also stick only for this feature on bCNC .. your GUI is much better and want to migrate .. pls make it happend :-)

greetings smuk

Carsten3P, 2017-01-01:
Adding this would be really great for milling PCBs, I would really appreciate it if you can add this feature.

SpudInNZ, 2017-01-07:
First up, thanks for your efforts on this software, it's excellent.

I would also really like to see auto-level implemented, it seems like the most obvious missing feature to me.

I'd also like to see support for x and y probing too, but I guess that should be in a different issue.

Thanks again.

GeerGuy, 2017-01-08:
Agree with others, having auto-level feature would make this great program the ultimate solution for my needs.  Integrating something like [https://github.com/martin2250/OpenCNCPilot](url) would be sweet.

Built in X and Y Probing would be nice as well, but you can use macros for that.

cheton, 2017-01-09:
Thank you for the information provided. I did a test with the auto-level feature in OpenCNCPilot, and it looks pretty straightforward to me.

The auto-level feature is originally planned to be implemented in v2.0 milestone, but now I will try to give the task at a higher priority, and make progressive enhancements in a separate <b>auto-level</b> branch.

cheton, 2017-02-18:
May I know if there's any one who can help contribute a UI prototype based on Chilipeppr's autolevel widget: https://github.com/chilipeppr/widget-autolevel/blob/master/widget.js

I would very much appreciate it!

elandesign, 2017-02-22:
@cheton I've started hacking on an autoleveller widget over on my fork at https://github.com/elandesign/cncjs/tree/autoleveller - it still needs a lot of work, and I'm spending much of the time reverse engineering what events I need to listen to to get it working. 

My plan is to generate an array of `{x: y: z:}` coordinates, then to look at actually applying them to the loaded GCode (probably learning from the Chilipeppr/bCNC code for this). 

Happy to take any advice people can offer in the process :)

cheton, 2017-02-25:
@elandesign 
Glad to see it's happening 👍 
https://github.com/elandesign/cncjs/commits/autoleveller

GeerGuy, 2017-02-27:
Some general ideas/thoughts on autoleveling...

I see 2 areas where it would be used; waste board or table and on the workpiece itself.  In the end these may be treated very similarly or the same.

Waste Board or Table
- when you want to improve the general flatness of your table, as opposed to tramming it
- covers the entire table, or at least within the limits of the "cuttable" area
- once homed the surface is probed based on the limits of the X and Y
- selectable number of points taken
- alternatively a file could be provided with the Machine coordinates to probe, say you wanted to avoid T-Slots or holes
- correction matrix is saved and optionally loaded on start
- autoleveling correction is automatically applied to all G Code

Workpiece
- when you need very accurate Z control, say with milling PCBs or engraving
- covers only the work surface
- need to supply a matrix of points to probe or automatically defined based on G Code loaded
- once homed the surface is probed based on the defined matrix or area
- selectable number of points taken
- correction matrix is saved and can be loaded
- autoleveling correction is automatically applied to loaded G Code
 
I have both an electrical continuity and micro switch style touch probe, so if anyone wants help testing, I'm ready to go.

shjackal, 2017-04-17:
Great to see there is some progress. 
How is it going with auto leveling? I too would really appreciate such a feature for PCB engraving.

dirtdiver9, 2017-04-25:
Very happy to see this going as well.  Thank you!

erpalma, 2017-06-26:
I just wrote a simple Python script that I use to probe the surface of PCBs. It is very raw at the moment but I tried to optimize it for maximum Z precision. I already wrote the code for applying the probed matrix to the input gcode and some code to optimize the tool path (about 20/40% decrease in time), but I need some time to polish and integrate this code into my script.

kikolobo, 2017-08-23:
This would be a very cool feature!

Smuk3D, 2017-09-14:
Hi all. Any news on this cool FeatureRequest. I still have to use bCNC :-/ ;-)

again 1+ vote for this here. 

best wisches from germany
smuk3d

kikolobo, 2017-10-06:
I really think this is a very needed feature 😊 Hope to see it soon!

tkurbad, 2018-01-01:
@elandesign, in https://github.com/tkurbad/cncjs/tree/autolevel, I adapted your code to the latest master. Nice interface!
Do you happen to have familiarized yourself with the math that has to be done to get this actually running? If not, I'd take a look at @erpalma's python code and see if I can adapt it...

kikolobo, 2018-01-04:
Question:
Is this integrated into the latest masteR?

tkurbad, 2018-01-04:
@kikolobo No. @elandesign has built an interface (aka widget) for autolevelling, but so far it's only a GUI element without an actual implementation...
Nothing of this is integrated in the official CNCjs code so far.

kikolobo, 2018-01-04:
Does it work ?

Saludos
FLG

tkurbad, 2018-01-04:
Again: It shows a GUI, but there is _nothing_ behind that GUI. Someone still has to implement the actual autolevelling routines.

kikolobo, 2018-01-05:
Got it! Thanks 🙏🏻 

Saludos
FLG

erpalma, 2018-01-10:
@tkurbad not the best code I have written, but it does the job. It is quite tuned for my needs and my cnc although. If you need help in understanding/porting the code, please tell me.

I have also a draft (super hackish) code for optimizing the gcode toolpaths to minimize distance. Using the gcode produced by pcb-gcode I've experimented from 10% to 50% reduction in time to complete the PCB etching. I hope to merge it next month...

tkurbad, 2018-01-10:
@erpalma Thank you for offering your help! I still didn't have time to look at your code, but I'll do so for sure next week.

sobreviventee, 2018-01-15:
@erpalma Could I somehow use your autolevel implementation? I really wanted to this function in the program.

erpalma, 2018-01-15:
@sobreviventee of course you can, it's open source for this reason :D Watch out for default clearance values since are quite tight!

cheton, 2018-01-16:
Yet another auto level code written in Java, that might be worth reading.
https://github.com/Daedelus1982/AutoLeveller/blob/master/src/autoleveller/probe/Probe.java

MikeColes, 2018-02-05:
I fell in love with Octoprint in the FDM world. This project looks equally awesome. 

To complete this task, what are the requirements?

> UI
> Data structure to store the height maps for reuse
--> Which coordinate system would be the best? Machine and then offsets?
> Trigonometry to break the gcode line segments down so the Z adjustments can be inserted
> Trig to know what part of the height map should be applied to the line segment

Does anyone else know of other gotchas that may be encountered for an implementation?

sobreviventee, 2018-02-16:
@erpalma @cheton  Where do I copy the autoprobe archive? I'm using the desktop version.

makarenya, 2018-02-23:
Hi. Look at this code:
It's generated heightmap pretty well.
I read code very quick, but i think that this is:
- [generator for probing g-code](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3066)
- [height map visualizer](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3146) 
- [gcode modification algorighm](https://github.com/Denvi/Candle/blob/master/src/frmmain.cpp#L3461)

Looks like very simple, one week work )))

soler2000, 2018-03-25:
This would be a great addition if someone can get this working.

kikolobo, 2018-06-12:
Another code I found that is open source. I used it and it works fine, hope to see this soon on CNCJS. 😊I wish I had the skills to help, but I think I will make another donation to help out.

https://bitbucket.org/daedelus1982/autolevellerae

easonke, 2018-09-11:
I start to work on this and hope that with it I can mill nice PCBs. Currently height map works (https://youtu.be/FWR0WONiACw). I plan to get GCode modification work for G0/G1 as next step.
Because I am new to almost everything (cncjs/CNC/grbl/github/js/threejs/react/...), the implementation is dirty and I did not push my code yet. Not sure what I should do to contribute.

easonke, 2018-09-15:
Prototype for G0/G1 works with hard coded probe parameters.
https://youtu.be/jA2wEEyMomk
I will clean code and push to my fork these several days.

kikolobo, 2018-09-15:
> Prototype for G0/G1 works with hard coded probe parameters.
> https://youtu.be/jA2wEEyMomk
> I will clean code and push to my fork these several days.

This is great news‼️ can’t wait to try it!

easonke, 2018-09-18:
Prototype code pushed to my fork. With it, I am able to mill a 13mm x 13mm PCB and get a LED blink with ATTINY13A :)
Currently it is not ready for everyone to use... I will continue to work on UI for configuration. But I just have grbl board and have no idea about tinyG or others. Hope there are some others can help on those.

jamhot1, 2018-11-18:
please implement this! Would be a great feature to add. How are we getting along with this feature?

f0m3, 2018-12-20:
Unfortunately this only seems to work with the payed version 👎 Cant get it running... There is no "Probe File Generator" Button in version 0.7

atmelino, 2018-12-20:
The Autoleveller AE author has announced that the advanced version will be free, starting February 1st.
[http://www.cncsoftwaretools.com/forum/viewtopic.php?f=4&t=848](url) 
I have replaced the TinyG controller with a GRBL Arduino Mega and external TB6600 drivers, probing many points works great now.
One note: The probing G-code generated by Autoleveller AE does not seem to work right away with cnc.js. Just remove the initial comment section and it works.

f0m3, 2018-12-21:
How do you "save probing results as Raw Probe File"? I am on osx and no file is created...

kreso-t, 2019-04-21:
If you are interested in a simpler solution (not requiring so many manual steps) until this feature is fully supported within cncjs, I uploaded to github [https://github.com/kreso-t/cncjs-kt-ext](cncjs-kt-ext)  a small cncjs extension (integration to master branch cncjs similar to i.e. cncjs-pendant-keyboard, no need to use forked cncjs) that executes probing and transforms the gcode in-place. It does not have any GUI, you just send gcode like  i.e
``` 
(#autolevel D15.0 H5.0 F20) 
```
and it will execute probing and transform the currently loaded gcode. 

Currently tested only with grbl and primarily targeted at PCB isolation milling - so not full scope of gcode is supported i.e. transforms only the G90/abs. coordinate mode, G91/rel moves are left intact, and circular interpolation G02/G03 will work fine if diameter is not larger than distance between points in the probing grid, however circular interpolation and relative moves are usually not present in the gcode produced by PCB isolation milling generators, so it should not matter.

cheton, 2019-04-21:
@kreso-t This is the simplest solution and easy to integrate. Thank you for providing a valuable solution.

9H1LO, 2019-05-06:
@kreso-t this is priceless for pcb milling, i will try it out ASAP ! thanks !

i think this should be in mainstream release or as a plugin 

using arduino based (grbl 1.1) which pin should be used to "sense" it has touched the board ?

f0m3, 2019-05-07:
> 
> 
> @kreso-t This is the simplest solution and easy to integrate. Thank you for providing a valuable solution.

Could it be integrated in the release version?

9H1LO, 2019-05-08:
@kreso-t 

i have tested this out with CNCJS, using pin A5 connected to mill bit and gnd to pcb 

i get output for example:
STEP: 5 mm HEIGHT:2 mm FEED:40
WCO: { x: 63.908, y: -2.28, z: -16.483 }
probed 1/81> -1.627 -1.626 0.282
probed 2/81> 3.895 -1.626 0.269
probed 3/81> 9.416 -1.626 -1.432
probed 4/81> 14.936 -1.626 0.078
probed 5/81> 20.457 -1.626 0.143

but when starting at actual job, the Z axis is going all over the place, sometimes deep into the board, then some places barely touching it, I am probably doing something wrong as I'm new to all this. I am setting up Z-Zero using ZProbe in CNCJS and flatcam Cut-Z to -0.001

any help or recommendations would be appreciated

kreso-t, 2019-05-08:
> @kreso-t
>
> i have tested this out with CNCJS, using pin A5 connected to mill bit and gnd to pcb
>
> i get output for example:
> STEP: 5 mm HEIGHT:2 mm FEED:40
> WCO: { x: 63.908, y: -2.28, z: -16.483 } [...]

Hi,

the third reading looks way off (the third number: -1.432 is probed Z coordinate and it should not deviate that much from point to point, especially here where they are distanced only by 5 mm):
probed 3/81> 9.416 -1.626 -1.432

one possible reason for such deviations may be that probe cable is picking EM noise. On my machine I have the probe connected to arduino similar way as limit switches i.e. [ https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches) I am using the option with opto-couplers and single power supply (the 'OK' version). 
Adding a 0.1uF capacitor between A5 and ground may work also OK.

Also instead of connecting the A5 to mill bit, a better option may be to connect the A5 to PCB (that must fixed to the table with some non-conductive clamps or similarly isolated from the ground) and gnd to the mill bit (cause there may already be some current leaking to ground through the spindle/mount/frame of the machine, depending on what kind of machine do you have)

hope this helps
br

9H1LO, 2019-05-09:
Dear @kreso-t 

Your recommendations with that 100nF cap made a BIG difference 

look at my values now:
probed 1/20> -2.286 -2.287 0.073
probed 2/20> 8.293 -2.287 0.029
probed 3/20> 18.873 -2.287 0.028
probed 4/20> 29.451 -2.287 0.054
probed 5/20> 40.030 -2.287 0.021
probed 6/20> -2.286 8.432 0.114
probed 7/20> 8.293 8.432 0.081
probed 8/20> 18.873 8.432 0.094
probed 9/20> 29.451 8.432 0.112
probed 10/20> 40.030 8.432 0.074

and here is what i went from and what I went to !

still lots to improve upon but already great progress

thanks a million !

[image]

[image]

ulab, 2019-10-04:
Can this module be used with the packaged CNCjs versions too?

(If not, I second the wish to get this into a release. Pretty please. :-))

imwhocodes, 2019-11-18:
I'm currently using @kreso-t autolever plugin (https://github.com/kreso-t/cncjs-kt-ext) for PCB milling and the core aligner functionality work flawlessly, the setup is a Carbide 883 with Grbl 1.1f

There are some quirk mostly caused by being an external plugin (no way to pause or reset the procedure, load / save status, not recognising the cnc reboot...) but as I said the core function is working good

IMHO it will be very nice to have it integrated by default / as a widget on CNCjs

greg-wood, 2019-11-27:
I would very much like to see this integrated. bCNC is very unstable on my macs. Trying to get  @kreso-t autolever plugin to work. I am using desktop app. It does not do anything. I suspect it's not connecting to the app. Is there any debug logs or anything for the plugin? when I start it it just stays there.  

this is all i get
> cncjs-kt-ext@1.0.0 start /Users/greg/cncjs-kt-ext
> node . "--config" "/Users/greg/.cncrc" "--socket-address" "localhost" "--socket-port" "49311"

rbenji23, 2020-01-05:
To get @kreso-t autolever plugin to work using the prebuilt electron app on mac. I had to to the following things:
1. Use the fork from @leoheck. [https://github.com/leoheck/cncjs-kt-ext](url)
2. Add a user to cncjs. Couldn't figure out how to make it work without a user. I used 
_id:"user", password:"user"._ 
3. Launch _View->Toggle Developer Tools_ pane on CNCjs, look at the sources tab, find 127.0.0.1:XXXXX where XXXXX is the port number we're looking for. The electron app seems to randomize this every time CNCjs is launched so you'll need to get this every time the program is launched.
4. Find the .cncrc file at your user root directory _/Users/usernamehere/.cncrc_, it will be hidden, so you'll need to show hidden files. Open it in a text editor.
5. If you haven't already checkout the _cncjs-kt-ext_ repo using git, do that, 
and then run `npm install`.
6. Run the following command replacing values with those from your steps above
_.cncrc file (--id --name --secret) 
cncjs window (--port --baudrate --socket-port)_
`node . --id '.cncrc_user_id_here' --name 'user' --secret '.cncrc_secret_here' 
--port '/dev/tty.usbserial-1420' --baudrate 115200 --socket-address 'localhost' 
--socket-port 50760
`
7. Add the macro for autoleveling `(#autolevel D10 H2.0 F50)` per the addon readme and run it!

Hopefully this functionality get added into the base program soon, maybe stick it in the probe widget!?

leoheck, 2020-01-06:
Thanks for the reference @rbenji23 

The URL of for my fork is this one [leoheck/cncjs-kt-ext](https://github.com/leoheck/cncjs-kt-ext/blob/master/index.js) @kreso-t  made a good work with that plugin.

I totally agree with @imwhocodes 
> IMHO it will be very nice to have it integrated by default / as a widget on CNCjs

The initial setup is not simple by default. I am also starting it with a service on my RPI. So both CNCjs and auto leveler is initialized automatically after restarts.

leoheck, 2020-01-06:
@rbenji23 The you can also use the default macro
This `(#autolevel)` instead of `(#autolevel D10 H2.0 F50)`

Did your setup work? Do you have any pictures that you can share here with us?

I was playing with my CNC 3018 yesterday but I was not able to produce something good/usable yet. I feel my machine is not very stable to keep the drill bit in its place. Also, my drill bit is not sharp anymore, so this can be the culprit too.

imwhocodes, 2020-01-19:
@cheton I’m thinking of implementing @kreso-t  code as a widget by forking CNCJS Master and porting the alignment core code into it, would it be this the correct way to do it? (My main concern is how to properly credit @kreso-t for the source) 

Also I’m not promising any timeline being mostly a Cpp and Python programmer with some web develop experience, but I’m watching other PR of widget here to get the hang of it

kreso-t, 2020-01-20:
@imwhocodes: you may use the source in any way you like, don't worry about the credit

leoheck, 2020-01-20:
@imwhocodes you can use the master from @kreso-t. He merged my changes some days ago.

Also, you can fork @kreso-t repo and work in your own fork, to make @kreso-t work as a CNCjs plugin. 

Then later, you can make a PR for @kreso-t (to merge your changes into the main repo).

Then later you can make a fork in CNCjs to merge the @kreso-t repo (or even your fork). I think this way you will keep the history, and the authorship of everything, your work will be also documented.

I never did that, but I feel it is doable.

https://saintgimp.org/2013/01/22/merging-two-git-repositories-into-one-repository-without-losing-file-history/

benishor, 2020-10-05:
Any news on master integration of this one?

[6 more replies]