cncjs/cncjs #34: When unknown command, milling stops

SanZamoyski, 2016-02-23:
Hi!

We are generating gcode with some program that generates unknown commands for GRBL. While using UniversalGcodeSender, we get response that this command is not recognisible and work is being continued. Haw can i prevent cnc.js from stopping after issuing not-known gcode command?

Best regards!

cheton, 2016-02-24:
I will fix it soon. May I have your G-code sample for verification? Thanks.

SanZamoyski, 2016-02-24:
Thank You for Your responce.
Here You are: [G-KOD.txt](https://github.com/cheton/cnc.js/files/145187/G-KOD.txt)

cheton, 2016-02-25:
It's been fixed in [v0.15.7](https://github.com/cheton/cnc.js/releases/tag/v0.15.7). Thank you for reporting this issue.

SanZamoyski, 2016-02-26:
Thank You!
Work like a charm.

closed 2016-02-26 by SanZamoyski