cncjs/cncjs #35: Simple linear motion

gokhandilek, 2016-05-06:
Hi,
Thank you for creating this. It is really useful.
In the demo that you have, do you mind sharing with us which stepper driver/controller you have used?
https://www.youtube.com/watch?v=fJyq4fyiGSc
I was planning to use the web cnc to control each stepper manually and move them to a specific length.
Would this be possible?

cheton, 2016-05-06:
The link below contains a collection of photos for the CNC machine that I self-made in this video. You can take a look into this:
https://www.facebook.com/cheton.wu/media_set?set=a.10207901184746263.1073741852.1195704289&type=3

To build this machine, I purchased three high performance TB6600 stepper motor drivers from China (taobao.com) with 3A max current and 1/128 microstepping, which worked well with NEMA 23 stepper motor (Max current: 3A) and Arduino Nano with Grbl 0.9i. Just let me know if you would like to know more details of it.

If you would like to control movement of each axis, use the Axes widget to move to desired position, it allows a precision of 0.001 mm:

https://github.com/cheton/cnc/wiki/User-Guide#axes-widget
[image]

In addition to the Axes widget, you can also type G-code commands (e.g. G0 X10 Y10) in the Console widget to make the movement manually.
[image]

Besides using a mouse, [keyboard shortcuts](https://github.com/cheton/cnc/wiki/User-Guide#keyboard-shortcuts) are also supported, you can use a keyboard or [Contour ShuttleXpress jog dial](https://github.com/cheton/cnc/wiki/User-Guide#contour-shuttlexpress) to work with the cnc web interface.

[image]

gokhandilek, 2016-05-06:
Thanks for your reply.
Can i connect to arduino YUN via ethernet?
How can I detect if stepper motor is forced and I can stop it automatically?

gokhandilek, 2016-05-06:
BTW I couldn't access the facebook link.

cheton, 2016-05-06:
By now the original [Grbl](https://github.com/grbl/grbl) source tree is mainly written for ATmega328P boards (e.g. Nano, Uno, or [Mega2560 by altering CPU type](https://github.com/grbl/grbl/issues/881)), I never tried it on Arduino YUN, but you can raise your question [here](https://github.com/grbl/grbl/issues) to see if there is any chance to run Grbl on it.

To pause or resume operations while sending G-codes, you can click on <kbd>Feed Hold</kbd> or <kbd>Cycle Start</kbd> buttons at top of the UI. Sorry for that I didn't get your point of your second question - <b>"How can I detect if stepper motor is forced and I can stop it automatically?"</b>. Can you explain it in more details?

cheton, 2016-05-06:
Photos have been uploaded at https://github.com/cheton/cnc/wiki/Others

gokhandilek, 2016-05-06:
Thanks for that.
You know how garage door stops if it hist an object?
That's what I am looking for just like safety feature really.

cheton, 2016-05-06:
Besides connecting a physical Emergency button to A1/A2 pins for manually stop, a common solution is to use 6 physical limit switches to restrict X-max, X-min, Y-max, Y-min, Z-max, and Z-min. You can check below for how to configure Grbl settings for limit switches:
https://github.com/grbl/grbl/wiki/Connecting-Grbl
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#20---soft-limits-bool
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#21---hard-limits-bool

gokhandilek, 2016-05-06:
Thanks for your reply.
I understand you regarding the limit switches.
This will not help if put your hand inside the machine or if an object gets in contact with moving mechanisms -  I need the machine to stop when such unexpected object is detected.

cheton, 2016-05-06:
What you're talking about can be achieved using [Capacitive Sensing](http://playground.arduino.cc/Main/CapacitiveSensor?from=Main.CapSense), which can sense the electrical capacitance of the human body. However, this topic is far beyond the scope of Grbl and CNC, you might need to find solutions from forums or websites . Thanks.

gokhandilek, 2016-05-06:
Thanks for that.
Capacitive sensing is a good solution however, it still doesnt resolve if an object(other than human) is obstructing the CNC machine's moving mechanism.

cheton, 2016-05-07:
I think there is no simple way of doing this since it's hard to tell what is the so-called obstruction for machine. 
Using limit switches is a common way to prevent machinery malfunction. Some people may use webcams on their CNC machine to monitor it from a remote location, if there is something going wrong, they can trigger a stop from a remote web interface to hold the movement.

closed 2016-05-07 by cheton