cncjs/cncjs #37: Jog controls greyed out

MrSurly, 2016-05-20:
Been playing with cnc.js on a Raspi with the Protoneer GRBL hat.  Working fine until I attached it to my laser cutter.  Now, the jog controls are greyed out.  Not sure why; I don't have limits enabled.

Under what circumstances will the jog controls be inoperative?  

GRBL seems to be in an OK state (idle).

cheton, 2016-05-21:
The jog controls will be grayed out if one of the following conditions is met:
1. Serialport connection is not established or closed
2. Grbl active state is not <b>Idle</b>
3. Server crashed

May you provide me information about your environment:

1. A full screenshot of your Firefox 44 browser when jog controls are grayed out

2. Which Grbl version you're using?

3. What version of cnc you're running?
  ```bash
$ cnc -V
1.1.0
```

4. Run `cnc -vv` on your Raspi to turn on verbose mode and check if there any unexpected errors occurred from the server

    ```bash
    $ cnc -vv
```

5. Is it always reproducible by launching Firefox browser on another PC or Laptop other than Eee PC?
  If yes, open Developer Tools with your Firefox and check if there are any Console errors.
![image](https://cloud.githubusercontent.com/assets/447801/15447946/8c57bd96-1f84-11e6-8f72-7829e0a2804e.png)

6. Can you respond with Grbl from the Console widget when the jog controls are grayed out?
  You can try sending G-code commands to Grbl to make sure the connection is alive:
  ```
G91 G0 X10
G90
```
![image](https://cloud.githubusercontent.com/assets/447801/15447990/8f0433ac-1f85-11e6-8c8e-23bec51010b9.png)

MrSurly, 2016-05-22:
CNC version:
> pi@laser:~ $ ./.npm/bin/cnc -V
> 1.0.0

GRBL Version: 
> Grbl 0.9i ['$' for help]

Screenshot (mouse cursor is a barred circle, the screen is small, you can't see everything at once):
[image]

GRBL manual control:
```
ok
> g91 g0x20
ok
> ?
<Idle,MPos:20.000,0.000,0.000,WPos:20.000,0.000,0.000>
> g91 g0x20
ok
> ?
<Idle,MPos:40.000,0.000,0.000,WPos:40.000,0.000,0.000>
```

Output of CNC -vv: 
```
pi@laser:~ $ ./.npm/bin/cnc -vv
Server is listening on 0.0.0.0:8000
-uMisVQt-eb55Zqy6s7LwsPVOvTc0Ofb GET / 304 244.289 ms
-uMisVQt-eb55Zqy6s7LwsPVOvTc0Ofb GET /api/controllers 304 32.486 ms
vTNnlB_LDTx0SSoNzIWdJ0ynhykvvxaI GET / 304 41.614 ms
vTNnlB_LDTx0SSoNzIWdJ0ynhykvvxaI GET /api/controllers 304 20.075 ms
pi@laser:~ $ 
```

Firefox 46.0 on Ubuntu Linux (trusty) i686 seems to work fine.

It was just handy for me to grab my old eeePC, as it's the only laptop available.

cheton, 2016-05-22:
There is an update in [v1.0.1](https://github.com/cheton/cnc/releases/tag/v1.0.1) that resolves a critical issue of outputing thousands of error to browser console:
`Google Chrome reported thousands of errors while zooming in/out a loaded G-code object.
[.CommandBufferContext]RENDER WARNING: Render count or primcount is 0.`

I'm not sure if it might affect Firefox (i386), but I'd recommend that you update to [v1.1.0](https://github.com/cheton/cnc/releases/tag/v1.1.0) which will be good for small screen display.
```bash
$ npm update -g cncjs@1.1.0  # or `npm install -g cncjs@1.1.0`
$ cnc -V
1.1.0
```

By the way, I'm preparing to build native <b>cnc</b> app (w/ Electron) to provide alternative installation methods, you can check it out at [https://github.com/cheton/cnc/releases/tag/master-latest](https://github.com/cheton/cnc/releases/tag/master-latest).
I can confirm it work for Windows x64 and Mac OS X, but haven't had a chance to test it with Linux. It will be great if you could try it out with Linux i386 platform, that will help me a lot! Thanks. 😃

cheton, 2016-05-22:
/cc @Protoneer
May you help update the link in the <b>Links and Extra reading</b> section of your [wiki](http://wiki.protoneer.co.nz/Raspberry_Pi_CNC_-_Quick_Start_Guide)? The old project link has been renamed to https://github.com/cheton/cnc. 😸

MrSurly, 2016-05-22:
1.1.0 Has the same issue with Firefox 44.

Additionally when using 1.1.0 (and I can write this up as a separate issue if desired), when you maximize the console widget in modern Chrome on i686, the top of the widget is hidden under the top black bar, so you can't restore the widget to it's original size.

I actually started building CNC with Electron myself, but realized I don't really have a need for it for my use case.

I downloaded the CNC Electron build on the Eee PC; looks like there's a missing dependency for libnss3, and after that was resolved, looks like there's [this issue](https://github.com/mongodb-js/electron-installer-dmg/issues/3), but I'm no Node expert ...
``` 
cnc: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory
epoulsen@ubuntu:~$ ```epoulsen@ubuntu:~$ cnc
cnc: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory
epoulsen@ubuntu:~$ sudo apt-get install libnss3
--------- 8<------------ (installed)
epoulsen@ubuntu:~$ cnc
Gtk-Message: Failed to load module "canberra-gtk-module"
A JavaScript error occurred in the main process
Uncaught Exception:
TypeError: Path must be a string. Received undefined
    at assertPath (path.js:7:11)
    at basename (path.js:1355:5)
    at Command.parse (/opt/cnc/resources/app/node_modules/commander/index.js:446:30)
    at Object.<anonymous> (/opt/cnc/resources/app/cnc.js:49:632)
    at Module._compile (module.js:541:32)
    at Object.Module._extensions..js (module.js:550:10)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)

```

Protoneer, 2016-05-22:
> /cc @Protoneer <https://github.com/Protoneer>
> May you help update the link in the *Links and Extra reading* section of
> your wiki
> <http://wiki.protoneer.co.nz/Raspberry_Pi_CNC_-_Quick_Start_Guide>? The
> old project link has been renamed to https://github.com/cheton/cnc. 😸

Thanks @cheton , I Fixed the link in the Wiki...

cheton, 2016-05-22:
@MrSurly 
I'm using electron-builder to bundle Electron app, electron-packager and package installers are used by electron-builder under the hood.

Since you're still facing the same problem with v1.1.0, I will try to get a Linux i386 VM or Docker image and check if I can reproduce the issue with Firefox 44.

In addition, the issue of maximizing a widget was fixed in this commit https://github.com/cheton/cnc/commit/836265dd261cb0295542679e1d2815deb8de4573. I will make a hotfix update (i.e. v1.1.1) or just publish a new release (v1.2.0) containing Electron builds in the next few days.

The initial purpose of bundling Electron app is mainly for Windows and Mac users, I just realized that you're running cnc server on a Raspberry Pi, so it is not necessary to install it on your ancient Eee PC, but I pretty thank you for reporting the installation problem with Linux platform.

cheton, 2016-05-22:
@Protoneer,
I purchased your Raspberry Pi CNC Board last month. It is a complete solution by combining Raspberry Pi, Arduino Nano, stepper drivers, and breakout board into one. Thanks for making this great product. 👍

cheton, 2016-05-22:
https://github.com/cheton/cnc/releases/tag/v1.1.1
* Fix an overlapping issue when a widget is in full screen mode
* Display server address at startup

cheton, 2016-05-23:
I installed two Linux VMs today and tried using _Firefox 44 on Debian 8.4 (i386)_ and _Firefox 46 on Cent OS 7 (x64)_ with [v1.0.0](https://github.com/cheton/cnc/releases/tag/v1.0.0) and [v1.1.2](https://github.com/cheton/cnc/releases/tag/v1.1.2), but still cannot reproduce the issue that you have encountered.

Have you ever tried Firefox 46 or Chrome on your ancient Eee PC? If they can work, it might be a corner case for certain environment.

MrSurly, 2016-05-23:
> I installed Linux VMs today and have tried using *Firefox 44 on Debian
> 8.4 (i386)* and *Firefox 46 on Cent OS 7 (x64)* with v1.0.0
> <https://github.com/cheton/cnc/releases/tag/v1.0.0> and v1.1.2
> <https://github.com/cheton/cnc/releases/tag/v1.1.2>, but still cannot
> reproduce the issue that you've encountered.
>
> Have you ever tried Firefox 46 or Chrome on your ancient Eee PC? If they
> can work, it might be a corner case for certain environment.

I can't even *find *chrome for i386 at all. I'll see about updating the Eee
PC.

Additionally, the Epiphany browser for the Raspi itself never gets past the
loading spinner.

I'm looking for a way to use cnc.js in production; I like it better than
the alternatives (laserweb, bCNC, etc).  But short of a desktop machine or
a full-size laptop, it's not really feasible. =(

cheton, 2016-05-24:
Can you press <kbd>Ctrl</kbd> + <kbd>Shift</kbd> + <kbd>I</kbd> with your Firefox to open JavaScript debugger and check <b>Console</b> messages, I suppose there should be errors when your jog controls were grayed out.

MrSurly, 2016-05-24:
> Can you press Ctrl + Shift + I with your Firefox to open JavaScript
> debugger and check *Console* messages, I suppose there should be errors
> when your jog controls were grayed out.

I'll check it out.

MrSurly, 2016-05-24:
The connection to ws://carambola.local:8000/
socket.io/?EIO=3&transport=websocket&sid=R3RYbyL7iRSbfxVjAAAC was
interrupted while the page was loading. socket.io.js:1971:31
2016-05-24T09:41:22-07:00 D cnc: Object {  } [23]@
http://carambola.local:8000/e3429934/app.js:1:6099 app.js:1:16750
mutating the [[Prototype]] of an object will cause your code to run very
slowly; instead create the object with the correct initial [[Prototype]]
value using Object.create vendor.js:39:11550
2016-05-24T09:41:23-07:00 D socket.io: connected [511]/<@
http://carambola.local:8000/e3429934/app.js:14:7043 app.js:1:16750
THREE.WebGLRenderer 75 vendor.js:26:31786
THREE.WebGLRenderer: Error creating WebGL context. vendor.js:27:2000

Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
vendor.js:27:1563
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. vendor.js:27:1563
Error: WebGL: WebGL creation failed. vendor.js:27:1563
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
vendor.js:27:1603
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. vendor.js:27:1603
Error: WebGL: WebGL creation failed. vendor.js:27:1603
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
vendor.js:27:1679
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. vendor.js:27:1679
Error: WebGL: WebGL creation failed. vendor.js:27:1679
TypeError: gl is null

cheton, 2016-05-25:
I found this in mozilla.dev mailing list. You can type <b>about:config</b>  in Firefox location bar and check if webgl.disabled is set to true.

[image]

---
https://groups.google.com/forum/#!topic/mozilla.dev.developer-tools/wRKAXB77FSE

WebGL can be disabled for graphics cards if everything isn't working properly and sometimes an issue may be fixed but the graphics card not removed from the blacklist. 

You can force WebGL to be turned on: 

1. In the Location bar, type about:config and press Return. 
2. The about:config "This might void your warranty!" warning page may appear. Click I'll be careful, I promise! to continue to the about:config page. 
3. In the Filter box, type webgl.disabled. 
4. Doubleclick on the webgl.disabled item to switch it to false. 
5. Click the menu button New Fx Menu and then click Quit Close 29. 
6. Start Firefox normally. 

---

cheton, 2016-05-25:
Before that, could you try to drag the Axes widget from the right pane to the left pane? I personally guess widgets in the right pane just stopped execution due to JavaScript errors in the 3D visualizer.

cheton, 2016-05-25:
http://ubuntuforums.org/showthread.php?t=1688042

MrSurly, 2016-05-25:
> I found this in mozilla.dev mailing list. You can type *about:config* in

webgl.disabled is false, but there are several others that are non-default,
e.g.:

gfx.blacklist.webgl.angle  integer 4
gfx.blacklist.webgl.msaa   integer 4
gfx.blacklist.webgl.opengl integer 4

I've tried setting them back to default (unset), but they return upon
restart.

WRT to the enumerated steps, you lost me at "New Fx Menu" ...

MrSurly, 2016-05-25:
The drag/drop of widgets only works on the left ...

cheton, 2016-05-25:
@MrSurly 
You can try this workaround to enable Axes widget:
1. Click <kbd>Manage Widgets</kbd> on the left side, turn off <b>Axes Widget</b> and <kbd>Save</kbd>.
2. After that, click <kbd>Manage Widgets</kbd> button again and turn it on, then you will see the widget on the left side.

cheton, 2016-05-25:
Finally, I can reproduce your case by setting <b>webgl.disabled</b> to true in Firefox.

However, I'm afraid that the graphics card (or chip) in your Eee PC cannot fully support modern WebGL. I can try to resolve JavaScript errors and provide an option to turn off WebGL animation for G-code simulation, but I cannot guarantee this will work on your environment.

MrSurly, 2016-05-25:
> Finally, I can reproduce your case by setting *webgl.disabled* to true in
> cannot guarantee this will work properly on your environment.

I don't think its' really necessary to support this.  It's probably OK to
have a "must have working webgl" as a requirement.  I was only using the
Eee PC because it's the only laptop I could find to bring near the laser
cutter.

cheton, 2016-05-25:
It's fine with me to provide a fallback option when WebGL is disabled, and it's not hard to handle JavaScript errors like that. I can work on it in the next few days and let you know when a new build is ready for test.

MrSurly, 2016-05-26:
I'd be happy to test it. =)

cheton, 2016-05-27:
Hi @MrSurly,

You can try install `1.2.0-rc1` that will use Canvas as a fallback when WebGL is unavailable.
```bash
$ npm uninstall -g cncjs  # uninstall previous version
$ npm install -g cncjs@1.2.0-rc1
```

You may see performance drop while doing G-code 3D animation, I will provide an option that can turn off 3D animation for better performance.

cheton, 2016-05-29:
I added an dropdown menu item in [v1.2.0-rc2](https://github.com/cheton/cnc/releases/tag/v1.2.0-rc2) that can toggle  toolhead animation in 3D visualizer. You can run `npm install -g cncjs@1.2.0-rc2` to install the latest version to your Raspberry Pi and test it with your Eee PC.

MrSurly, 2016-05-31:
Hi Cheton,

I cannot find the dropdown menu.  Looks like it's supposed to be near the
run/pause/stop/close buttons in the rendering area, but I cannot see it. I
can confirm it says 1.2.0-rc1 in the titlebar.

I don't see any dropdown using Chrome on Mac, either.

However, I am NOT connected to the Raspi (it's at home Eee PC is at work);
I can try this later tonight, probably.

Errors from the console are:

```mutating the [[Prototype]] of an object will cause your code to run very
slowly; instead create the object with the correct initial [[Prototype]]
value using Object.create vendor.js:8:18691
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
app.js:14:31417
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. app.js:14:31417
Error: WebGL: WebGL creation failed. app.js:14:31417
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
app.js:14:31446
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. app.js:14:31446
Error: WebGL: WebGL creation failed. app.js:14:31446
2016-05-31T08:44:44-07:00 D cnc: Object { version: "1.2.0-rc1", state:
Object } [23]@http://carambola.local:8000/a29f5979/app.js:1:6099
app.js:1:16750
2016-05-31T08:44:44-07:00 D socket.io: connected [516]/<@
http://carambola.local:8000/a29f5979/app.js:14:13074 app.js:1:16750
THREE.CanvasRenderer 76 app.js:14:24317
An iframe which has both allow-scripts and allow-same-origin for its
sandbox attribute can remove its sandboxing.```

On Sun, May 29, 2016 at 7:46 AM, Cheton Wu <notifications@github.com> wrote:

> I added an dropdown menu item in v1.2.0-rc2
> <https://github.com/cheton/cnc/releases/tag/v1.2.0-rc2> that can toggle
> toolhead animation in 3D visualizer. You can run npm install -g
> cncjs@1.2.0-rc2 to install the latest version to your Raspberry Pi and
> test it with your Eee PC.
>
> —
> You are receiving this because you were mentioned.
> Reply to this email directly, view it on GitHub
> <https://github.com/cheton/cnc/issues/37#issuecomment-222364429>, or mute
> the thread
> <https://github.com/notifications/unsubscribe/AAYHCEMWYA-kjw02YXdMKwWU5uqW0ctyks5qGaa5gaJpZM4Ijjp9>
> .
>

cheton, 2016-06-01:
Sorry I didn't make it clear about your environment. I used to run cnc server on one computer (e.g. Raspi), and use web interface on another one (e.g. laptop or desktop computer), it allows me to separate CPU loads (because the web interface may causes high CPU usage for 3D display) and re-open browser at anytime during sending G-code data, so I'd assume you may do it like this way.

[1.2.0-rc1](https://github.com/cheton/cnc/releases/tag/v1.2.0-rc1) provides a fallback to Canvas renderer if WebGL is unavailable, [1.2.0-rc2](https://github.com/cheton/cnc/releases/tag/v1.2.0-rc2) adds a new dropdown menu that can turn off toolhead animation to gain a bit performance. You have to update to 1.2.0-rc2 if you need to turn off animation.

According to your console logs, it should have been using Canvas renderer instead of WebGL renderer:
```
Error: WebGL: WebGL creation failed. app.js:14:31417
Error: WebGL: WebGL creation failed. app.js:14:31446
THREE.CanvasRenderer 76 app.js:14:24317
```
The <i>WebGL creation failed</i> message is no harm since it's trying to detect if WebGL is available, but I found several WebGL blacklisting errors in your console logs as below:
```
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting. app.js:14:31417
Error: WebGL: Refused to create native OpenGL context because of blacklisting. app.js:14:31417
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting. app.js:14:31446
Error: WebGL: Refused to create native OpenGL context because of blacklisting. app.js:14:31446
```

May I know if you're still experiencing problems with Axes widget in either 1.2.0-rc1 or 1.2.0-rc2 build?

MrSurly, 2016-06-07:
> The *WebGL creation failed* message is no harm since it's trying to

Hello @cheton,

My apologies for the delay. I became ill after my trip.  In any case, I
tried 1.2.0-rc2.  I can see the gear icon in the rendering area, but it is
also greyed out.

Logs as follows:

socket.io.js:1971:31
2016-06-07T15:49:23-07:00 D cnc: Object { version: "1.2.0-rc2", state:
Object } [20]@http://carambola.local:8000/7edb4d74/app.js:1:2961
app.js:1:16762
mutating the [[Prototype]] of an object will cause your code to run very
slowly; instead create the object with the correct initial [[Prototype]]
value using Object.create vendor.js:8:18691
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
app.js:15:1188
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. app.js:15:1188
Error: WebGL: WebGL creation failed. app.js:15:1188
Error: WebGL: Disallowing antialiased backbuffers due to blacklisting.
app.js:15:1217
Error: WebGL: Refused to create native OpenGL context because of
blacklisting. app.js:15:1217
Error: WebGL: WebGL creation failed. app.js:15:1217
2016-06-07T15:49:23-07:00 D socket.io: connected [516]/<@
http://carambola.local:8000/7edb4d74/app.js:14:14863 app.js:1:16762
THREE.CanvasRenderer 77 app.js:14:26106
An iframe which has both allow-scripts and allow-same-origin for its
sandbox attribute can remove its sandboxing. carambola.local:8000

cheton, 2016-06-09:
Take good care of yourself.

Thanks for giving it a try and sorry it didn't work as expected. Since I don't have a blacklisted graphics card at hand to reproduce your case, it's not easy to trace which JavaScript errors may stop execution according to your console logs. I assumed these exceptions should have been handled within this try-catch block (https://github.com/cheton/cnc/blob/master/src/web/lib/three/Detector.js#L14).

Sorry that I may postpone resolving this issue until I got a blacklisted graphics card or found a workable solution.

cheton, 2016-10-24:
Recently I had a similar problem with my 2012 MacBook Air that Google Chrome reported WebGL is unavailable. I tried a solution by browsing to `chrome://flags/` and enable the option <b>Override software renderning list</b>, restart Chrome and WebGL will work fine. 

[image]

cheton, 2016-10-25:
I paste notes for Firefox here:

http://superuser.com/questions/836832/how-can-i-enable-webgl-in-my-browser

## Firefox
First, enable WebGL:

* Go to `about:config`
* Search for `webgl.disabled`
* Ensure that its value is `false` (any changes take effect immediately without relaunching Firefox)

Then inspect the status of WebGL:

Go to `about:support`
* Inspect the <b>WebGL Renderer</b> row in the <b>Graphics</b> table:
  * If the status contains a graphics card manufacturer, model and driver (eg: "NVIDIA Corporation -- NVIDIA GeForce GT 650M OpenGL Engine"), then WebGL is enabled.
  * If the status is something like "Blocked for your graphics card because of unresolved driver issues" or "Blocked for your graphics driver version", then your graphics card/driver is blacklisted.

If your graphics card/drivers are blacklisted, you can override the blacklist. Warning: this is not recommended! (see blacklists note below). To override the blacklist:

* Go to `about:config`
* Search for `webgl.force-enabled`
* Set it to `true`

(Like Chrome, Firefox has a <b>Use hardware acceleration when available</b> checkbox, in <b>Preferences > Advanced > General > Browsing</b>. However, unlike Chrome, Firefox does not require this checkbox to be checked for WebGL to work.)

cheton, 2016-11-03:
This issue can be closed now since WebGL is no longer required by disabling 3D view in version 1.7.0.
Related issues: #76, #86 

[image]

commit a0321873 (2016-05-26): Adding a fallback to Canvas renderer when WebGL is unavailable (fixes #37)
closed 2016-11-03 by cheton