cncjs/cncjs #38: Small bug on Axes Widget

smikey, 2016-06-08:
Hi,

at first: Thanks for this great piece of software! I'm totally impressed.

I'm using it on a rPi 1B and GRBL0.9j and it works pretty fine. The only thing that doesn't work are the buttons in the Axes Widget to set the axes position to zero (eg G92 X0 Y0 Z0). If I press the button the string "[Object object]" appears in the GRBK-console widget. So I think there's some bug with the command to be sent. I was looking through the code and it seems good, but it doesn't work. Feels like a JavaScript-Object problem.

Greets
smikey

cheton, 2016-06-08:
Ooops! It failed due to an incompatible API change with react-boostrap. I will fix it soon.

cheton, 2016-06-08:
https://github.com/react-bootstrap/react-bootstrap/blob/master/CHANGELOG.md#v0290

Breaking: Change onSelect signature to be (eventKey: any, event: SyntheticEvent) => any on all React-Bootstrap components, instead of the old inconsistent mishmash (#1604, #1677, #1756)

cheton, 2016-06-08:
Fixed in v1.2.1

smikey, 2016-06-08:
Wow that's incredibly fast. Thanks! I'll test it as soon as it is in the npm repository.

smikey, 2016-06-08:
Works fine! Thanks!

closed 2016-06-08 by smikey