cncjs/cncjs #39: Syntax error on raspberry pi

belese, 2016-06-12:
Hi, 

i was using v1.0 without problem on a raspberry pi B,
today, i want to upgrade, both raspbian jessie and cncjs are now up to date, but i've got a problem when i want to restart cnc 

```
/home/pi/.npm/lib/node_modules/cncjs/node_modules/webappengine/index.js:10
module.exports = (options) => {
                           ^^
SyntaxError: Unexpected token =>
    at exports.runInThisContext (vm.js:73:16)
    at Module._compile (module.js:443:25)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:3313:19)
    at __webpack_require__ (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:21:30)
    at Object.module.exports.Object.defineProperty.value (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:2619:22)
```

i've tried to uninstall, reinstall fro git, same problem occurs.

my node version
node -v
v0.12.6

any idea?

Thanks

Belese

cheton, 2016-06-12:
I will

cheton, 2016-06-12:
Thanks for reporting this issue. There's a bug within the webappengine module. I will fix it soon.

belese, 2016-06-12:
Thanks, i've tried your fix, it was working now.

btw, very great app!!

cheton, 2016-06-12:
It's been fixed in [v1.2.3](https://github.com/cheton/cnc/releases/tag/v1.2.3), you can run `npm install -g cncjs@1.2.3` to install the latest version. Sorry for the inconvenience!

closed 2016-06-12 by cheton