cncjs/cncjs #40: Language translation

dronecz, 2016-06-13:
Hi, 

first of all, many thanks for very nice SW. 

I would like to translate UI to my language, but I can not find right file where I can make some changes. 

Could you tell me where I should look? 

Many thanks.

cheton, 2016-06-13:
Thanks for asking! Current [i18n resources files](https://github.com/cheton/cnc/blob/master/src/web/i18n/en/resource.json) were using pre-generated hash keys, I felt it's not easy for human to compare keys between two languages without using a translation tool. I will make some changes to i18n resources files to make it easy to translate.

cheton, 2016-06-13:
Hi @dronecz,

I have updated the resource file for all supported languages, it should be more easily for translations. For examples:
- English: https://github.com/cheton/cnc/blob/master/src/web/i18n/en/resource.json#L150
- French: https://github.com/cheton/cnc/blob/master/src/web/i18n/fr/resource.json#L150
- German: https://github.com/cheton/cnc/blob/master/src/web/i18n/de/resource.json#L150

First, check out [contributions guide](https://github.com/cheton/cnc/blob/master/CONTRIBUTING.md#localization-l10n) and install [v1.2.4](https://github.com/cheton/cnc/releases/tag/v1.2.4). You can do [runtime translations](https://github.com/cheton/cnc/blob/master/CONTRIBUTING.md#runtime-translations) with Google Chrome browser. Once you have finished your translations, copy translated `resource.json` file to `src/web/i18n/{{lng}}/resource.json` in [your forked project](https://github.com/cheton/cnc/blob/master/CONTRIBUTING.md#code-contributions), and create a pull request to submit changes. Thanks.

dronecz, 2016-06-15:
Hi @cheton 

I did upgrade from 1.2.1 to 1.2.4 on my virtual Ubuntu and tryed translate part of english file to my language and refresher my webpage but nothing changed. I also restarted server but still no change on webpage. I´m probably doing something wrong, but i have no idea what. 

For test I made changes in installed version of cnc: ~/.npm/lib/node_modules/cncjs/src/web/i18n/en .

[image]

Many thanks for help.

cheton, 2016-06-15:
Try making changes to the `~/.npm/lib/node_modules/cncjs/dist/web/i18n/{{lng}}/resource.json` file, and make sure the `[v] Disable cache (while DevTools is open)` option is checked with respect to your Developer Toools settings.

dronecz, 2016-06-15:
I´m editing file directly so I´m not using browser. When I edit resource.json in this location, it works. 

Many thanks for now.

cheton, 2016-06-15:
Great! Thank you for your hard work.

cheton, 2016-06-17:
I may change a little bit by adjusting table and button styles for some widgets (e.g. Axes and G-code widgets) to make the UI layout consistent while translating to other languages.

dronecz, 2016-06-17:
Hi @cheton 

I made neceserry changes to resource.json and I followed instructions in contribution guide, but I´m not able to push my changes to my branch. I´m newbe with github so please be patient with me.

I got this error when I try to push to git: 
[image]

Many thanks for adwise.

cheton, 2016-06-17:
You can send the file to cheton@gmail.com. This is a straightforward approach. :smiley:

dronecz, 2016-06-17:
Ok, here it is ;-).

cheton, 2016-06-17:
I didn't receive your mail from my mailbox. Can you try to send it again? Thanks.

dronecz, 2016-06-18:
Second try ;-)

2016-06-17 15:26 GMT+02:00 Miroslav Zuzelka <mzuzelka@gmail.com>:

cheton, 2016-06-18:
Sorry for bothering again. There might be some problems that I still cannot find it from my mailbox (cheton@gmail.com). Can you copy/paste your resource.json file at here, I will help move your file to a proper location.

dronecz, 2016-06-18:
Hi @cheton 

my email probably falling in to spam :-(. Here is zip file ;-).

[cs.zip](https://github.com/cheton/cnc/files/321675/cs.zip)

Many thanks

cheton, 2016-06-18:
Got it! I will save these files to the `src/web/i18n/cs/` directory. Thanks!

cheton, 2016-06-18:
There is another one that needs your help. How do you say `Loading...` in Czech?

[image]

The source file is located in https://github.com/cheton/cnc/blob/master/src/app/i18n/cs/resource.json.

dronecz, 2016-06-18:
Hi,

I feel realy happy that I can contribute to this project.

We say : Nahrávám..

I was thinking hoe to change it but after that I give up bacause word
Loading is very often seen in games, DVDs, TV and really everythink which
needs to load something ;-)

Anyway, thank you very much for your great work.

Miroslav

P.S. I got few ideas about UI of your project, should I fill those in Issue
section of github or can I write to email and send it to you?

cheton, 2016-06-18:
Thanks a lot! You can post issues on GitHub that will be much easier to track. :)

cheton, 2016-06-22:
I will close the issue now. You can update here any time if you need to correct translations.

closed 2016-06-22 by cheton