cncjs/cncjs #42: Idea: Remote pendant

dronecz, 2016-06-15:
Hi @cheton 

I have (maybe) stupid idea, but I want to know your opinion. 

I want make remote pendant based on Atmega 32u4 or maybe ESP8266 which will work as HID device. I guess, if this device would be connected through USB will work like any other keyboard, but would be possible to make changes to your cnc server to listen on some port (like 8888) and if there are some  data, it will accept it as keyboard action. This comunnication will be through LAN/WIFI. I´m not programmer but I think I could make some Arduino sketch with my friends which will send this sort of data. 

Is this possible or not from your side? 

Thanks for answer.

cheton, 2016-06-15:
The web interface communicates with cnc server via the socket.io v1.x client library, I think it is doable using the socket.io library for Arduino. There are several useful resources on GitHub:
- https://github.com/billroy/socket.io-arduino-client
- https://github.com/chadstachowicz/socket_io_arduino_cc3000
- https://github.com/washo4evr/Socket.io-v1.x-Library

The last link contains an [example](https://github.com/washo4evr/Socket.io-v1.x-Library/tree/master/examples) that demostrates how to send messages between Arduino and Node.js server. You can also refer to the [TinyWeb example](https://github.com/cheton/cnc/blob/master/examples/tinyweb/controller.js) to learn how to send or receive messages for communicating with cnc server.

However, I don't have ESP8266 on hand and never programmed Arduino codes, it would be great if we can get someones's help to go for it.

dronecz, 2016-06-19:
Hi @cheton 

I´m looking in to it, but just to be sure I want to verify that I understand it write and CNC server will listen for all communication on port 8000, is that right? I found some examples for Arduino + Node.js and I will make few tests, but I want be sure that I´m using right port for communication. 

Thanks

cheton, 2016-06-19:
The cnc server uses a path "<b>/socket.io/</b>" for Socket.IO communication under the same HTTP server port (e.g. 8080). Looking forward to seeing your work!

dronecz, 2016-06-21:
Hi @cheton 

I´m very happy to report small step forward: 
[image]

It´s just simple connection but thats just for start, I hope.

dronecz, 2016-06-22:
Hi @cheton 

can I somehow start server to see full communication with client? I´m now using cnc -vv, but there is limited info what is hapennig.

Many thanks for answer.

cheton, 2016-06-22:
`cnc -vv` is the full verbose mode in current version. May I know what information you would like to see (e.g. full response from Grbl)?

dronecz, 2016-06-22:
I was just thinkink if I can see whole communication between server&client. Something likeI click on X move in browser and on server side I will see :

connect from .. on IP... 
received G91 X100 F500
accepted & executed

I know, I want probably to much, but I just need to know what server receiving when I´m trying to send data to it. So far I can connect to server, but I´m learning what I should send to the server. I looked through tinyweb examples and I bit know what I should send, but I need to know what get through and what not.

dronecz, 2016-06-22:
I take it back.. :-p I was testing server at work where a had no connection to GRBL and when I test it now at home with GRBL I see that verbose mode does just what I describe in my previous message. Sorry for my stupidy ;-) :-p

cheton, 2016-06-22:
I will make a patch release (v1.3.2) later that can output all Grbl responses with `cnc -vvv` for debugging purpose. Note that this will generate a lot of Grbl status logs and not recommended for production, but you can pipe to `grep -v` to filter out unnecessary logs.

dronecz, 2016-06-23:
Hi @cheton 

thanks for patch, I hope someone else than just me will use it ;-).

I would like you to ask what is right syntax of sending commands to server? 

I found out that, I need to send `client.send("write","",""); / client.send("command","","");` but I´m unable to find right syntax for the rest of the line. EverytimeI send something to server, I get this response.
[image]

Many thanks for your patient and for your help.

cheton, 2016-06-23:
I'm not sure if your client interface on Arduino is the same as client library in JavaScript. You can try out the below commands:

**Emit events from Arduno to cnc server:**
```js
// List ports
client.send('list');

// Open port
client.send('open', '/dev/cu.xxxxxxxx', 115200);

// Close port
client.send('close', '/dev/cu.xxxxxxxx');

// Write
client.send('write', '/dev/cu.xxxxxxxx', 'G0 X0 Y0 Z0\n'); // should contain a newline character
client.send('write', '/dev/cu.xxxxxxxx', '?'); // No newline for Grbl realtime commands

// Command
client.send('command', '/dev/cu.xxxxxxxx', 'feedhold');
client.send('command', '/dev/cu.xxxxxxxx', 'cyclestart');
client.send('command', '/dev/cu.xxxxxxxx', 'reset');
```

**Available listener events:**
```js
serialport:list
serialport:open
serialport:close
serialport:error
serialport:read
serialport:write
grbl:status
grbl:parserstate
gcode:statuschange
```

AustinSaintAubin, 2016-12-04:
@cheton can you point me in the right direction.
I am trying to send serial command to grbl from a python script.

I am attempting to get a PS3 Controller to work with CNC.
I have almost everything figured out expect for how to send the jog commands from the python script to CNC or GRBL.

Currently Trying this with no success, does not run (no idea what I am doing).
```
#client test
import socket
client_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
client_socket.connect(('localhost', 8000))

client_socket.send("list")
client_socket.send("'open', '/dev/ttyUSB0', 115200")
client_socket.send("'close', '/dev/ttyUSB0'")

client_socket.send('write', '/dev/ttyUSB0', '$H\n')
```

cheton, 2016-12-04:
You have to import socketIO-client instead of importing websocket directly. I will take a look at it tomorrow.

tritao, 2016-12-04:
I am also attempting to interface with the API remotely (from C# though).

Is there any specific feature the code needs from Socket.io or could this be changed to use just standard WebSocket instead?

Because I am having trouble finding good libraries for Socket.IO for C# for instance.

I can find lots of libraries for WebSocket though, and most languages include one in the standard libraries these days, so it could make it easier to use the API.

AustinSaintAubin, 2016-12-04:
I have had hour of !fun trying to get anything to work with socket.io on python, bash, other tools.
I ended up looking into using Node.js to work with HID & Gamepads, see [Idea: Remote Pendant (Playstation 3 Dualshock Controller / SIXAXIS Controller) #103](https://github.com/cheton/cnc/issues/103).

tritao, 2016-12-04:
Same here, I tried connecting with https://github.com/Quobject/SocketIoClientDotNet.

After some logging I've found the problem, some kind of authorization header is needed.

```
2016-12-04 23:55:43 275 [10] SocketIoClientDotNet-IO.cs:Socket#39 - ignoring socket cache for http://localhost:8000/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9
2016-12-04 23:55:43 281 [10] SocketIoClientDotNet-Manager.cs:Open#192 - readyState CLOSED
2016-12-04 23:55:43 281 [10] SocketIoClientDotNet-Manager.cs:Open#192 - opening http://localhost:8000/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9
2016-12-04 23:55:43 299 [10] SocketIoClientDotNet-Manager.cs:Open#192 - connection attempt will timeout after 20000
2016-12-04 23:55:45 760 [16] SocketIoClientDotNet-Manager.cs:Open#192 - readyState OPENING
2016-12-04 23:55:45 761 [16] SocketIoClientDotNet-Manager.cs:Open#192 - opening http://localhost:8000/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9
2016-12-04 23:55:45 761 [16] SocketIoClientDotNet-Manager.cs:Open#192 - connection attempt will timeout after 20000
2016-12-04 23:55:46 876 [18] SocketIoClientDotNet-Manager.cs:Open#192 - connect_error
2016-12-04 23:55:46 889 [18] SocketIoClientDotNet-Manager.cs:Reconnect#440 - will wait 1000ms before reconnect attempt
2016-12-04 23:55:47 903 [16] SocketIoClientDotNet-Manager.cs:Reconnect#464 - EasyTimer Reconnect start
2016-12-04 23:55:47 903 [16] SocketIoClientDotNet-Manager.cs:Reconnect#464 - attempting reconnect
2016-12-04 23:55:47 903 [16] SocketIoClientDotNet-Manager.cs:Open#192 - readyState CLOSED
2016-12-04 23:55:47 903 [16] SocketIoClientDotNet-Manager.cs:Open#192 - opening http://localhost:8000/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9
2016-12-04 23:55:47 903 [16] SocketIoClientDotNet-Manager.cs:Open#192 - connection attempt will timeout after 20000
2016-12-04 23:55:47 903 [16] SocketIoClientDotNet-Manager.cs:Reconnect#464 - EasyTimer Reconnect finish
2016-12-04 23:55:47 905 [20] SocketIoClientDotNet-Manager.cs:OnOpen#263 - open
2016-12-04 23:55:47 916 [20] SocketIoClientDotNet-Manager.cs:Packet#354 - writing packet Type:0 Id:-1 Nsp:/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9 Data:{} Attachments:0
2016-12-04 23:55:47 916 [20] SocketIoClientDotNet-Parser.cs:Encode#54 - encoding packet Type:0 Id:-1 Nsp:/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9 Data:{} Attachments:0
2016-12-04 23:55:47 917 [20] SocketIoClientDotNet-Parser.cs:EncodeAsString#103 - encoded Type:0 Id:-1 Nsp:/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9 Data:{} Attachments:0 as 0/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9,{}
2016-12-04 23:55:47 927 [20] SocketIoClientDotNet-Manager.cs:Reconnect#440 - reconnect success
2016-12-04 23:55:47 932 [20] SocketIoClientDotNet-Parser.cs:decodeString#271 - decoded 4{"message":"No Authorization header was found","code":"credentials_required","type":"UnauthorizedError"} as Type:4 Id:-1 Nsp:/ Data:{"message":"No Authorization header was found","code":"credentials_required","type":"UnauthorizedError"} Attachments:0
2016-12-04 23:55:47 933 [20] SocketIoClientDotNet-Parser.cs:decodeString#271 - decoded 4/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9,"Invalid namespace" as Type:4 Id:-1 Nsp:/socket.io&token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9 Data:"Invalid namespace" Attachments:0
```

cheton, 2016-12-05:
Hi @tritao,

Are you running cnc with a pre-built executable (i.e. Electron app) on Windows? If yes, the port will be a random port number above 1024, not 8000.

tritao, 2016-12-05:
I compiled it from Git and am launching it with `NODE_ENV=development bin/cnc -vv -p 8000`.

In the above trace the token was not correct, but even after fixing that I cannot seem to get past the authorization check. I've been checking `socketio-jwt` sources to check for any clues but no luck so far.

Any chance the communication channel could be switched to straight WebSockets to make communication with non-JS environments easier?

cheton, 2016-12-06:
You can refer to the client sample code in JavaScript:
https://github.com/cheton/cnc/blob/master/examples/tinyweb/controller.js

An authorized token is stored in localStorage, this demostrates how to retrieve the token stored in local storage. The "error" event indicates that the token is no longer valid, so it redirects user to the Sign In page. Once the user is successfully authenticated, you can retrieve the token from localStorage.

```js
var root = window;
var token = '';

// Retrieve session token
try {
    var cnc = JSON.parse(localStorage.getItem('cnc') || {});
    cnc.state = cnc.state || {};
    cnc.state.session = cnc.state.session || {};
    token = cnc.state.session.token || '';
} catch (err) {
    // Ignore error
}

var socket = root.io.connect('', {
    query: 'token=' + token
});

socket.on('connect', function() {
    // Add your code here
});

// Redirect user to the Sign In page
socket.on('error', function() {
    socket.destroy();
    window.location = '/';
});
```

cheton, 2016-12-06:
Hi @tritao,

`socketio-jwt` is only necessary on the server-side, you don't need to include it for your client-side implementation.

AustinSaintAubin, 2016-12-09:
Take a look at #103 if having issues connecting to socket.io
I got Node.js client working.

cheton, 2017-02-06:
Close this issue. You can get started with [cncjs-pendant-ps3](https://github.com/cncjs/cncjs-pendant-ps3/blob/master/bin/cncjs-pendant-ps3) to learn how to create a pendant controller.

closed 2017-02-06 by cheton

yasamoka, 2018-02-03:
Finally succeeded in writing a cnc.js pendant in Python!

https://gist.github.com/yasamoka/fc45ca11742b88c92cb5df316c5ccf7d

I would appreciate any feedback, especially if this is essentially helpful and stands to be improved.