cncjs/cncjs #43: UI not loading in cnc-1.2.5-win-x64.exe and further issues...

dl9sec, 2016-06-17:
Starting cnc 1.2.5 tries to load the UI, but the area leaves blank. After "View" -> "Reload" the UI is drawn well. Same effect after managing widgets and clicking "Save".

Moving the Webcam widget from lower left to upper right seems not to be persistent. After restarting cnc the Webcam widget is again in the lower left position. Moving the other left sided widgets to the right works persistent.

Regards, Thorsten

cheton, 2016-06-17:
The second issue is probably a bug fixing side-effect with this commit (https://github.com/cheton/cnc/commit/be4eb893f6084b8c91ec078c87b2da3bd6709376), I will fix it soon.

cheton, 2016-06-17:
It should be fixed in this build: https://github.com/cheton/cnc/releases/tag/v1.2.6. The Windows x86/x64  binaries will be available later.

cheton, 2016-06-17:
The Windows x86 build is available at https://github.com/cheton/cnc/releases/download/v1.2.6/cnc-1.2.6-win-ia32.exe. Please let me know if you've encountered any issues. Thanks.

closed 2016-06-19 by cheton