cncjs/cncjs #45: Helvetica Neue ♦♦♦♦ css declared in 404

minithc, 2016-06-18:
It seems like Helvetica Neue font, declared in the 404.hogan, provides a limited char set. I finally disabled the css with DevTools, and it is ok now, and I can't repeat the issue. But at one moment, the screen ♦♦♦♦ ♦♦♦♦, ♦♦ ♦ have to disable css fonts to see the text. Used 1251 and UTF-8 code pages in locale.json, both the same result. I will try to repeat the issue anyway.

cheton, 2016-06-18:
You don't need to change `locale.json` since the file is not currently used in the code. Here is the screenshot in Russian language running with my local development server (in English).

[image]

minithc, 2016-06-18:
Thanks! Everything seems to be fine. It was probably my mistake or something with code pages. Thanks again for your time.

cheton, 2016-06-18:
Thank you for your great efforts. I will publish a release containing newly translated languages (e.g. German, Russian, andTraditional Chinese) in the next few days.

closed 2016-06-18 by cheton