cncjs/cncjs #46: RTSP support for webcam video stream?

dl9sec, 2016-06-18:
Is there support for rtsp wecam livestreams in the webcam widget?

Regards, Thorsten

cheton, 2016-06-18:
By now the webcam widget can only load a URL within iframe, I made a quick search and found this article in StackOverflow: http://stackoverflow.com/questions/2245040/how-can-i-display-an-rtsp-video-stream-in-a-web-page

I can try embedding the VLC ActiveX control to HTML like this:

```html
<OBJECT classid="clsid:9BE31822-FDAD-461B-AD51-BE1D1C159921"
     codebase="http://downloads.videolan.org/pub/videolan/vlc/latest/win32/axvlc.cab"
     width="640" height="480" id="vlc" events="True">
   <param name="Src" value="rtsp://cameraipaddress" />
   <param name="ShowDisplay" value="True" />
   <param name="AutoLoop" value="False" />
   <param name="AutoPlay" value="True" />
   <embed id="vlcEmb"  type="application/x-google-vlc-plugin" version="VideoLAN.VLCPlugin.2" autoplay="yes" loop="no" width="640" height="480"
     target="rtsp://cameraipaddress" ></embed>
</OBJECT>
```

Note this method can work on Firefox and not being supported by Chrome or some other browsers anymore. But even more important here, I didn't have a chance to test RTSP because I haven't getting an IP camera. 

Before I can make a working solution, could you try [IPCamLive](https://ipcamlive.com/) to generate a URL that can be embedded within iframe, as suggested below from the same article in StackOverflow:

---

[image]

cheton, 2016-06-18:
Found an alternative solution:
https://github.com/agsh/rtsp-ffmpeg

It runs [ffmpeg](http://www.ffmpeg.org/) to stream RTSP to MotionJPEG and update webcam image (i.e. JPEG) automatically, but the performance may not as good as viewing directly from RTSP.

cheton, 2016-06-19:
Stream any RTSP stream and output to websocket for consumption by [jsmpeg](https://github.com/phoboslab/jsmpeg)
https://github.com/kyriesent/node-rtsp-stream

cheton, 2016-06-19:
There is a workaround by [restraming RTSP to M-JPEG](https://github.com/cheton/cnc/wiki/FAQ#restream-rtsp-to-m-jpeg), you can check it out in the [FAQ](https://github.com/cheton/cnc/wiki/FAQ).

Basically, you can use <i>ffmpeg</i> to convert RTSP to M-JPEG stream, but it still needs a streaming server like <i>ffserver</i> to play M-JPEG streams. Unfortunately, <i>ffserver</i> is only available on Mac and Linux, but [not exists on Windows](http://stackoverflow.com/questions/2679569/how-to-do-simple-video-streaming-with-ffserver-on-windows). I believe there should have some Windows-based solutions besides <i>ffserver</i>, but currently I have no idea about it.

Integrating <i>ffmpeg</i> with a Node.js server might be a solution as a replacement to <i>ffserver</i> from my side, but it may take me a long time to get it done. I would suggest you setup a Linux VM to run <i>ffmpeg</i> and <i>ffserver</i> if you have urgent needs.

dl9sec, 2016-06-20:
Thanks for the efforts, saw the FAQ entry yesterday :-)
I have a Debian Linux server in my home Network, so runing the ffserver/ffmpeg should be no problem. I already tried this with quick installation and test, but it didn't work so far. I think it's the ACL setup. Will go deeper in it...

cheton, 2016-06-20:
Now I'm able to create a RTSP streaming server using <i>ffserver</i> and <i>ffmpeg</i> on my Mac:

**ffserver.conf**
```
HTTPPort 8000
RTSPPort 8001
HTTPBindAddress 127.0.0.1
RTSPBindAddress 127.0.0.1
MaxClients 100
MaxBandwidth 10000
NoDefaults

<Feed webcam.ffm>
   File /tmp/webcam.ffm
   FileMaxSize 20M
</Feed>

<Stream webcam.mp4>
   Feed webcam.ffm
   Format rtp
   VideoSize 640x480
   NoAudio
</Stream>
```

Launch <i>ffserver</i>:
```bash
$ ffserver -d -f ffserver.conf
```

Then I use <i>ffmpeg</i> to open the built in webcam on my Mac and send a live feed to <i>ffserver</i>:
```bash
$ ffmpeg -f avfoundation -i "0" http://127.0.0.1:8000/webcam.ffm
```

Playing the video streaming via RTSP protocol using the VLC player:
```
rtsp://127.0.0.1:8001/webcam.mp4
```

Later on I will start to do RTSP to M-JPEG streaming and let you know my test result.

cheton, 2016-06-20:
I made a slight modification to the [FAQ](https://github.com/cheton/cnc/wiki/FAQ#restream-rtsp-to-m-jpeg), now my webcam widget can play M-JPEG by restreaming RTSP to M-JPEG. 

The `ffserver.conf` file:
```
HTTPPort 8090

<Feed webcam.ffm>
  File /tmp/webcam.ffm
  FileMaxSize 50M
</Feed>

<Stream webcam.mjpg>
  Feed webcam.ffm
  Format mpjpeg
  VideoBufferSize 8000
  VideoCodec mjpeg
  VideoFrameRate 24
  VideoSize 640x480
  NoAudio
</Stream>
```

Launch <i>ffserver</i>:
```bash
ffserver -d -f ffserver.conf
```

Run <i>ffmpeg</i> to restream RTSP to M-JPEG:
``` bash
ffmpeg -i "rtsp://<ip-camera>/" http://localhost:8090/webcam.ffm
```

cheton, 2017-02-06:
Close this issue because I don't have any plans to add built-in support for RTSP streaming. You can use ffserver and ffmpeg instead.

closed 2017-02-06 by cheton