cncjs/cncjs #47: Enhancement: Login to server

dronecz, 2016-06-19:
Hi @cheton 

I think would be good have some user management to connect and control the server. I´m using Octoprint and there I have to login to control my 3D printer. I think this is good feature because, nobody wil mess up with printer if they don´t know login name. 

Thanks

cheton, 2016-11-17:
The account management is almost finished and it will be available in 1.8.0 release. Stay tuned for further updates.

### Sign in page
[image]

### Settings > Account
[image]

closed 2016-11-17 by cheton

dronecz, 2016-11-18:
Hi, 

many thanks for this feature. Can I ask if you plan any permissions for users? Like if someone is loged in, this user can control machine and if user is not loged in user can just see we isnterface. 

Many thanks for info and great work ;-).

cheton, 2016-11-19:
Yeah! There will be a separate UI or read-only mode for anonymous or guest users. Thank you for your suggestion.

commit 9f684fbd (2016-11-16): Secure the WebSocket connection (resolves #47)