cncjs/cncjs #48: Enhancement: Global setting

dronecz, 2016-06-19:
Hi @cheton 

would be possible make some page with global settings? I would like to have settings for webcam, visualisation setting, etc. 

This settings could be stored in some files and can be accesed for loged users only.

cheton, 2016-06-19:
Thank you for your suggestion. I use HTML5 local storage to persist config settings, you can access the <i>localStorage</i> object as below:

[image]

The reason why I chose to use <i>localStorage</i> instead of having a physical file is that localStorage is much easier to manage and migrate data among different releases especially breaking changes, storing to a config file is fine but I may have to carefully managing the content in case of data corruption or data loss before the development is going to be stable.

For the account based configuration like #47 , it makes sense to store data in a physical storage, I will try to find a better way to store account-based configuration while developing login page and user account manager for the web interface.

liqdfire, 2016-08-22:
@cheton have you thought about pouchdb?

Another option for the authentication would be some basic oauth integrations to google, github or something similar.

ralfkuh, 2016-11-23:
I would like to add one whish for the config:

Allow external connections on/off.

If i understand it correct, at the moment it is only possible to connect from private network. That is in the index.js file in the CNCServer start function. There is a static array ALLOWED_IP_RANGES. I can't find a way to allow connections from external IPs without modifying the source code.

cheton, 2016-11-24:
You're right. Allowed IP addresses and ranges should not be hard-coded in the source code. I will make it configurable in the <b>Settings > Global Settings</b> section.

cheton, 2016-11-25:
Hi @ralfkuh,

I add a command line option `--allow-remote-access` (commit https://github.com/cheton/cnc/commit/5fc8e1fe31e5bbf52770603c56292d86bf272171) that can be used to allow remote connections from Internet. You can turn it on like below:
```bash
$ cnc --allow-remote-access
```

This option will be available in 1.8.2.

ralfkuh, 2016-11-25:
Cheton you are awesome!
Many thanks.

closed 2017-03-25 by cheton