cncjs/cncjs #50: [Bug] Axes panel always zero

iceboundflame, 2016-06-22:
v1.3.0 on Mac (desktop version), GRBL 0.9 (Shapeoko 3)

When I jog, the coordinates don't change - they show 0 for machine and workspace coordinates.

cheton, 2016-06-22:
Could you type `?` in the Console widget to check Grbl status when it shows zeros on the Axes display panel?

```
> ?
<Idle,MPos:0.006,0.006,0.000,WPos:0.006,0.006,0.000>
```

cheton, 2016-06-22:
I found a critical problem in the OS X Electron build that it failed to correctly receive data from an established serial port. Instead of using desktop app, you can install Node.js and run  `npm install -g cncjs@1.3.1` on your Mac to install the web-based version.

iceboundflame, 2016-06-22:
Thanks for the quick reply - will give it a shot next time I'm at the workshop!

cheton, 2016-06-22:
Something weird. It just get back to work when I close the app and launch it again with the same installed version (v1.3.0). Please let me know if you can reproduce the case next time.

cheton, 2016-06-22:
Ooops! It only happened when I run both the web-based version and the OS X app at the same time since the port is being occupied by two cnc server instances.

You can try to check the Grbl status (?) next time you encountered the same situation.

cheton, 2016-06-22:
I saw you posted an issue at https://github.com/chilipeppr/widget-axes/issues/7.

Do you still have the same problem using chilipeppr?

iceboundflame, 2016-06-22:
I do indeed still see that other problem on Chilipeppr: the work coordinates are fine, but the machine coordinates are zero. But I have also been told I was using an outdated fork of Chilipeppr. Sadly the official Grbl version of Chilipeppr seems to not have work coordinate systems at all.

How does your interface determine the axes position to display? Maybe there's something different about the status format output by the Grbl version I'm using. I'll copy the Grbl status line for you when I next get access to the machine.

cheton, 2016-06-23:
The cnc server parses the Grbl status line using a regular expression as [below](https://github.com/cheton/cnc/blob/master/src/app/controllers/grbl/grbl.js#L21):
```jsx
// <Idle,MPos:5.529,0.560,7.000,WPos:1.529,-5.440,-0.000>
const matchGrblCurrentStatus = (msg) => {
    return msg.match(/<(\w+),\w+:([^,]+),([^,]+),([^,]+),\w+:([^,]+),([^,]+),([^,]+)>/);
};
```

It will not report current Grbl status to the web interface if it failed to parse the line.

iceboundflame, 2016-06-28:
I see, looks like my Grbl 0.9g (Shapeoko 3 XXL) has some additional statuses reported. I get this on terminal:

```
<Idle,MPos:-148.000,-191.000,-5.000,WPos:-30.000,-30.000,1.150,Buf:0,RX:0,Ln:0,F:0.>
```

I tried turning off all status report flags, but there's still other stuff in the status line that can't be disabled:

```
> $10=0
ok
> ?
<Idle, Ln:0,F:0.>
```

By the way, the in-app console truncates items and copying from it doesn't seem to put anything in my clipboard:
[image]

cheton, 2016-06-28:
You can click the "fullscreen"  icon to enlarge the console widget to see full message.

cheton, 2016-06-28:
I think you might have a customized version of Grbl 0.9g, the official Grbl image does not output `Ln:0,F:0.`.

[image]

iceboundflame, 2016-06-28:
It appears the firmware was built with USE_LINE_NUMBERS and REPORT_REALTIME_RATE:

https://github.com/grbl/grbl/blob/da5c65b54b10b3ec27ebaa466275c25954667729/grbl/report.c#L482

But it'd be great to handle and ignore extra data printed after the WPos.  Does your Axes panel still work for you when you set $10=31?

cheton, 2016-06-28:
I just tried for a while and found this regular expression pattern should work for most conditions:
```js
var msg = '';
var pattern = /[a-zA-Z]+(:([0-9\.\-]+,[0-9\.\-]+,[0-9\.\-]+(,[0-9\.\-]+)?)|:([0-9\.\-]+))?/g;

// $10=0
msg = '<Idle>';
msg.match(pattern);
// ["Idle"]

// 3-axes
msg = '<Idle,MPos:5.529,0.560,7.000,WPos:1.529,-5.440,-0.000>';
msg.match(pattern);
// ["Idle", "MPos:5.529,0.560,7.000", "WPos:1.529,-5.440,-0.000"]

// 4-axes
msg = '<Idle,MPos:5.529,0.560,7.000,0.000,WPos:1.529,-5.440,-0.000,0.000>';
msg.match(pattern);
// ["Idle", "MPos:5.529,0.560,7.000,0.000", "WPos:1.529,-5.440,-0.000,0.000"]

// $10=31
msg = '<Idle,MPos:-148.000,-191.000,-5.000,WPos:-30.000,-30.000,1.150,Buf:0,RX:0,Lim:000>';
msg.match(pattern);
// ["Idle", "MPos:-148.000,-191.000,-5.000", "WPos:-30.000,-30.000,1.150", "Buf:0", "RX:0", "Lim:000"]

// Custom output
msg = '<Idle,MPos:-148.000,-191.000,-5.000,WPos:-30.000,-30.000,1.150,Buf:0,RX:0,Ln:0,F:0.>';
msg.match(pattern);
// ["Idle", "MPos:-148.000,-191.000,-5.000", "WPos:-30.000,-30.000,1.150", "Buf:0", "RX:0", "Ln:0", "F:0."]
```

cheton, 2016-06-28:
Fixed in this commit 07a5acbb90748e355156e0d889994d08115e713f. Now it can support up to 6 axis and various kinds of output format.

A newer version v1.4.0 will be released in the next few days.

iceboundflame, 2016-07-01:
Tried out f68171e and it works great. Thanks!

closed 2016-07-01 by iceboundflame