cncjs/cncjs #51: [Enhancement] Keyboard jogging with arrows

iceboundflame, 2016-06-22:
Hi, what do you think of adding keyboard jogging with arrow keys (for XY) and pg-up/pg-down (for Z), and modifier keys for 10x/0.1x jog speed?

For users without a shuttle controller this can be much more intuitive; many other CNC control panels I've used (Mach3, bCNC, Chilipeppr) use this scheme.

cheton, 2016-06-22:
The link below contains a list of pre-defined keyboard shortcuts:
https://github.com/cheton/cnc/wiki/User-Guide#keyboard-shortcuts:

<kbd>!</kbd> - Feed Hold
<kbd>~</kbd> - Resume
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>h</kbd> - Homing
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>u</kbd> - Unlock
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>r</kbd> - Reset
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>x</kbd> - Select/Deselect X Axis
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>y</kbd> - Select/Deselect Y Axis
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>z</kbd> - Select/Deselect Z Axis
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>=</kbd> - Switch Jog Distance
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>[</kbd> - Jog Backward
<kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>command</kbd> + <kbd>]</kbd> - Jog Forward
   :   :   :

I would prefer not using arrow keys and page up/down since it will make trouble while performing some UI actions on the browser.

iceboundflame, 2016-06-22:
That's a reasonable concern, although in the context of a CNC controller, do users typically scroll around the web page using arrow keys? I think jogging is by far the more common action and should be easy (and possible with one hand); scrolling can still be accomplished by scrolling the mouse/touchpad or using the scroll bars.

I'm guessing you personally trigger the hotkeys using the shuttle wheel peripheral. The current choices of hotkeys are not ideal for keyboard users for several reasons:
- The hotkey is hard to press (too many modifiers required)
- x, y, and z are far from each other and require a lot of hand motion to switch between them
- switching jog distance button isn't idempotent, accidentally pressing it wrong number of times will result in wrong jog distance (bCNC for example has 1-4 for selecting 0.01, 0.1, 1, 10mm); that's why modifiers like shift/ctrl for fast/slow would be easier

BTW - great job with this project overall! I've been on the hunt for a good GRBL controller for a while and this looks like one of the most polished that I've seen so far, and built with modern web tools and development practices.

cheton, 2016-06-22:
Pressing arrow keys or pg-up/pg-down will also scroll the page horizontally and vertically, and it may trigger browser (or OS) default shortcuts when combining with modifier keys, which is not the desired behavior, and you may see the same problem with a web-based solution like Chilipeppr.

To not interfere with default browser action, I can try adding a keyboard jogging toggler on the Axes widget. When the jogging toggler is turned on, it will disable arrow key scrolling by preventing the default browser action, like so:
```js
window.addEventListener('keydown', function(e) { // arrow keys
    if([37, 38, 39, 40].indexOf(e.keyCode) > -1) {
        e.preventDefault();
    }
}, false);
```

Then you will be able to jog using arrow keys / pg-up / pg-down without causing scrolling issue.

iceboundflame, 2016-06-22:
That sounds great! Probably also need to add event listeners for Page Up, Page Down (33, 34).

cheton, 2016-06-25:
I added several keyboard shortcuts to the dev branch, and the rest will be completed in the next few days.

cheton, 2016-06-27:
Axes widget preview
[image]

iceboundflame, 2016-06-28:
Looking great! Can't wait to use this.

iceboundflame, 2016-07-01:
Tested it out, very nicely done. Thank you!

closed 2016-07-01 by iceboundflame