cncjs/cncjs #52: [Bug] on mac standalone version, dragging toolbar window leads to white screen

iceboundflame, 2016-06-28:
Attempting to drag the G-Code tool window out of the right sidebar leads to a white screen. Dev tools reports an error:
[image]

v1.3.2 on mac

cheton, 2016-06-28:
I fixed a lot of drag and drop issues recently. It will be fixed in the release.

iceboundflame, 2016-06-28:
Thanks! Good to hear.

iceboundflame, 2016-07-01:
FYI I built the latest build into a Mac OS X app and it still has this issue; I can move every panel except the Axes one. Also, if I collapse the Axes panel, then I can move it.

It just goes white screen and disconnects the dev tools without any error recorded. When I connected to the Mac app in Chrome (localhost:65008) it also crashed when dragging this panel - with an "Aw, snap" error (no JS error visible).

cheton, 2016-07-01:
It worked several days ago, but now I can reproduce this issue for both the web-based UI and Mac app.  The error seems very strange, I will try to figure out what's going wrong.

cheton, 2016-07-01:
This issue is related to https://github.com/wwayne/react-tooltip/issues/126 that I used [react-tooltip](https://github.com/wwayne/react-tooltip) to display the tooltip when hovering on the Keypad button.

[image]

However,  update react-tooltip to [v3.0.8](https://github.com/wwayne/react-tooltip/releases/tag/3.0.8) still cannot fix this problem when dragging around Axes widget. I will try to find an alternative solution or use a workaround way by removing the react-tooltip component when dragging the Axes widget.

cheton, 2016-07-02:
It's fixed in the master branch, but there is still a minor bug that the Axes widget will lose its original state due to re-rendering widget when dragging from the right pane to the left pane. I may fix it in the later build.

cheton, 2016-07-14:
Fixed in v1.4.0

closed 2016-07-14 by cheton