cncjs/cncjs #53: Probing without applying tool offset

iceboundflame, 2016-07-01:
Hi Cheton,

I use a Z probe is to set the Z zero of the WCS; the current implementation appears to:
- set G54 Z=0 upon contact
- apply a G43.1 tool length offset equal to the touch plate height.

Since G54 coordinates are persistent across power cycles and soft resets, while G43.1 settings are lost, do you think it would be better to just issue a single command `G10 L20 P1 Z<touch plate height>`? This would be especially handy because I like to soft reset before starting jobs. It also preserves coordinates if I have to E-stop, power cycling the controller; then I don't have to re-zero or re-apply the tool length offset.

Do you see any downsides to this approach?

Thanks!
David

cheton, 2016-07-01:
Hi @meikrantz,

May I know what do you think of turning
```bash
G49            # Cancel TLO
G91
G38.2 Z-10 F25  # Probe toward workpiece
G90
G10 L20 P1 Z0  # Zero Out Work Z axis for G54
G43.1 Z-8      # Set TLO to -8mm (touch plate height)
G91 G0 Z2      # Retract slightly from touch plate (2mm)
G90
```

into this for Z-probing (remove G49 and G43.1):
```bash
G91
G38.2 Z-10 F25  # Probe toward workpiece
G90
G10 L20 P1 Z8  # Set G54 Z=8mm (touch plate height)
G91 G0 Z2      # Retract slightly from touch plate (2mm)
G90
```

meikrantz, 2016-07-01:
I think that would be fine.  I have done it without TLO in the past,
because at the time I don't think grbl supported it.

cheton, 2016-07-02:
Thanks @iceboundflame and @meikrantz,

It's now available in the master branch.

iceboundflame, 2016-07-06:
Thanks @cheton! I tried to test out the master branch today but was running into an issue where no serial output was being registered on the UI - i.e. the console, Grbl status panel, Axes panel. Both 9e0aa26 and 58f8506 had this issue. f68171e is still working fine for me though.

cheton, 2016-07-06:
Hi @iceboundflame,

Could you try running `./bin/cnc -vvv` to check if there are any logs or errors outputted to the server console?
```bash
$ npm run prepublish  # Make production build
$ ./bin/cnc -vvv  # Maximum verbosity level
   :   :   :
Started the server at http://0.0.0.0:8000/.
```

iceboundflame, 2016-07-07:
Hmmm -- I tried the above and it seems to work now. Strange. Going to close this issue since the probing function now works great. Thanks for the change!

BTW - looking back, it appears I only ran the following commands the first time:
```
npm install --production
./bin/cnc
```
I didn't run `npm run prepublish`. Is that necessary? From the instructions at https://github.com/cheton/cnc/wiki/Installation#git-installation it seems like  `npm install --production` is all that is needed.

closed 2016-07-07 by iceboundflame

cheton, 2016-07-07:
Ooops! I forgot to update the Git installation flow. The `npm run prepublish` command is necessary for creating a production build, I will update the wiki later.