cncjs/cncjs #54: [enhancement] Axes panel - easily edit current position (G10 L20)

iceboundflame, 2016-07-01:
This is an idea I also suggested to Chilipeppr folks. I thought I'd mention it here since it seems you are actively developing cnc with great enthusiasm.

It's useful to be able to set the current tool position, i.e. if I've jogged the toolhead somewhere and would like to define the current position as (-50, -10, 10), I should be able to edit the coordinates and easily redefine the current coordinates. Right now I have to write `G10 L20 P1 X-50 Y-10 Z10` in the terminal to do so. This is handy in several use cases:
- sometimes you can't reach the position you would like to set as 0
- you want to position the work from some other reference point (i.e. the G-code zero is at the top but I want to align the bottom of the work to the current position).
- sometimes you want to edit the current position to run an "air pass" (run without actually touching the stock)

In some other systems I've used (Mach3, bCNC), clicking the axis readout turns the number into a text box, where you can "edit" the current position without moving the tool. e.g. clicking the X work position and setting it to 5 would issue `G10 L20 P1 X5`.

Separately, it might also be useful to have a button to move to a specified coordinate - although this is lower priority because G0 is pretty easy to type from the console.

cheton, 2016-07-01:
That sounds great! It should be easy to implement.

iceboundflame, 2016-07-07:
Awesome, thank you! Very nice to use.

closed 2016-07-07 by iceboundflame