cncjs/cncjs #57: Suggestion: Optical work zero adjustment

dl9sec, 2016-07-07:
This Feature (with much more functionality) is already used in [GRBLize](https://github.com/heise/GRBLize).
Prequisition is a working USB 2.0 endoscope camera vertically mounted at Z (near spindle/tool).

Workflow:
1. Calibration of the camera/tool offset
 - A Little hole is drilled/milled on a random position and this position is stored.
 - The camera is jogged to the Position of the  hole.
 - A crosshair with a circle in the middle (adjustable in diameter) is shown in the camera image.
 - Hole and crosshair/circle are jogged one above the other.
 - The X/Y-offset from the stored position is entered as offset from the work position

2. Usage
 - The crosshair is jogged to the workpiece position or wherever the work position should be and marked.
 - The camera/tool-offset from the calibration is used to calculate the work Zero position of the tool.

cheton, 2016-07-14:
It looks good to me. I can create a crosshair vector as an overlay on top of the camera image, it's helpful for locating a pre-defined position.

To change the work position from a reference point, you can refer to https://github.com/cheton/cnc/issues/54. This feature is now available in v1.4.0, you can edit the current position as below:

[image]

cheton, 2016-08-23:
bCNC - Probe Camera Alignment
https://github.com/vlachoudis/bCNC/wiki/Probe-Camera-Alignment

cheton, 2016-09-01:
This feature will be available soon in v1.6.0. I will keep you updated.

[image]

cheton, 2016-09-02:
I've just released [v1.6.0](https://github.com/cheton/cnc/releases/tag/v1.6.0). Move the mouse over the webcam widget, you will be able to adjust the image scale and toggle the crosshair for centering a workpiece.

dl9sec, 2016-09-04:
Hi Cheton,
great stuff.
Unfortunately i can not use it, because i have no webcam supporting the appropriate stream format and i coudn't get a streaming converter to work.
Would it be possible to support USB-cams (maybe through the direct media interface) too? It doesn't make sense for me to install a webcam, nearly as big as my spindle at the Z-axiy. There are cheap and well working endoscopic cameras (5 to 7 mm in diameter) at a****n an e***y very suitable to be mounted near the spindle....

Regards, Thorsten

cheton, 2016-09-04:
It is possible to use HTML5 &lt;video&gt; element to display video from a connected USB camera, which is supported by modern browsers like Edge, Firefox, and Chrome. If you have a USB camera, you can try the demo link (https://cezarywojtkowski.com/react-webcam/examples/) of the [react-webcam](https://github.com/cezary/react-webcam) project to see if that works for you.

Note that the USB camera must be connected to your host machine running the web interface (or app). If that's your case, I can add support for HTML5 video in a future release.

dl9sec, 2016-09-04:
Hi Cheton,
thanks a million for your efforts.
YES it works :-D
So HTML5 video support would be really cool...

cheton, 2016-09-16:
Added in v1.6.4

closed 2016-09-17 by cheton