cncjs/cncjs #58: Raspberry Pi "Error opening serial port"

soswow, 2016-07-17:
I am not sure what's wrong with my setup.
```
$ .npm/bin/cnc --host 10.0.0.23
Started the server at http://10.0.0.23:8000/.
2016-07-17T12:13:40.292Z - error: [cncserver] Error opening serial port "ttyAMA0": err={}     at /home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:1353:44
```
I can connect via **minicom** and **cutecom**, but not with cncjs =(
Not sure what to try next.

soswow, 2016-07-17:
```
pi@raspberrypi ~ $ node -v
v4.4.7
pi@raspberrypi ~ $ npm -v
2.15.8
```

cheton, 2016-07-17:
May I know what version of cnc you're running?
```bash
$ .npm/bin/cnc -V
1.4.0
```

soswow, 2016-07-17:
`1.4.0`
I've installed it today. via npm.

cheton, 2016-07-17:
Not sure if it's related to the serialport library, I just upgraded serialport from v3.1.2 to v4.0.1 in the latest version (i.e. cncjs@1.4.0). Could you try installing `cncjs@1.3.2` and check if it works?
```bash
$ npm uninstall -g cncjs
$ npm install -g cncjs@1.3.2
```

meikrantz, 2016-07-18:
I had the same issue when upgrading to 1.4.0 on RPi.  I did a full uninstall/install instead of upgrade and it worked fine after that.  The full install from scratch updated the serialport version, and then things started working again.

cheton, 2016-07-18:
Hi @soswow,

Did you have any previously version installed? If yes, try a simple uninstall and reinstall, as suggested by @meikrantz, to see if that fixes the problem you're having.

soswow, 2016-07-18:
I didn't have previously installed cncjs. I just followed instruction on
how to bump gcc etc and installed serialport. Now, when I thought about it
- before bumping gcc and following those instructions I tried to install it
without gcc 4.8, and it failed. So I installed again after I bumped gcc. I
will try to uninstall and install again.

cheton, 2016-07-18:
To compile addons (e.g. serialport) for Node.js v4 or higher, gcc/g++ 4.8 or higher is required. The raspbian-jessie image ships with gcc version 4.8, you can start with it to make things simple.

In addition, I also recommend you update NPM v2 to v3, which resolves several installation problems and it can install package dependencies in a flat way. You can upgrade npm using this command:
```bash
$ npm install -g npm
$ npm -v
3.10.5
```

After upgrading to NPM v3, you might have to re-install (i.e. uninstall and install) global packages to rebuild package dependencies.

soswow, 2016-07-19:
I've tried npm 3.
```
$ sudo .npm/bin/cnc --host 10.0.0.23
Started the server at http://10.0.0.23:8000/.
2016-07-19T11:24:54.321Z - error: [cncserver] Error opening serial port "ttyAMA0": err={}     at /home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:1353:44
```

```
npm install -g cncjs
/home/pi/.npm/bin/cnc -> /home/pi/.npm/lib/node_modules/cncjs/bin/cnc

> serialport@4.0.1 install /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@4.4.7 (node-v46 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build'
/home/pi/.npm/lib
```
```
pi@raspberrypi ~ $ npm -v
3.10.5
pi@raspberrypi ~ $ which npm
/home/pi/bin/npm
pi@raspberrypi ~ $ ls -la /home/pi/bin/npm
lrwxrwxrwx 1 pi pi 21 Jul 19 10:46 /home/pi/bin/npm -> /home/pi/.npm/bin/npm
pi@raspberrypi ~ $ which node
/usr/local/bin/node
pi@raspberrypi ~ $ node -v
v4.4.7
pi@raspberrypi ~ $ ls -la /usr/local/bin/node
-rwxr-xr-x 1 root staff 22270888 Jul 19 09:51 /usr/local/bin/node
```
I am using grbl shield for RPI. I will try their prebuild image. http://wiki.protoneer.co.nz/Raspberry_Pi_CNC_User_Interface_SD_Card_Image
Also I've downloaded clean jessi image. Just in case.

Also, I was thinking of trying write some simple nodejs app and test if I can connect to that port at all. For debugging purposes

cheton, 2016-07-20:
Hi @soswow,

I just using an USB cable to connect my RPi2 with an Arduino UNO R3 running Grbl v0.9j. The serial port I connected is `/dev/ttyUSB0`, not `ttyAMA0`. May I know if you're using the GPIO serial device on the Pi, but not an USB cable?

[image]

soswow, 2016-07-20:
As I said in my prev. comment -I am using this shield http://wiki.protoneer.co.nz/Raspberry_Pi_CNC done by @Protoneer #19

meikrantz, 2016-07-20:
I have the same Protoneer board, and it shows up as /dev/ttyAMA0.  I seem to remember having to uninstall serialport manually in the past, but memory is a bit fuzzy.

cheton, 2016-07-20:
Could you try using the serialport-terminal tool to check if it connect to the `/dev/ttyAMA0` device?
```bash
$ /home/pi/.npm/lib/node_modules/serialport/bin/serialport-terminal.js -p /dev/ttyAMA0 -b 115200
```

If it does not report any errors, add a `~/.cncrc` file under your user home (i.e. `/home/pi/.cncrc`) with the following configuration settings and re-launch `cnc` again:
```json
{
    "ports": [
        {
            "comName": "/dev/ttyAMA0",
            "manufacturer": ""
        }
    ]
}
```

It will add a `/dev/ttyAMA0` entry to the serialport list in addition to `ttyAMA0`.

cheton, 2016-07-20:
```bash
node_modules/serialport/bin/serialport-terminal.js -p /dev/ttyAMA0 -b 115200
Error [Error: Error: Permission denied, cannot open /dev/ttyAMA0]
```

If you got a permission denied error as above, you can check out this article for the fixes:
https://www.raspberrypi.org/forums/viewtopic.php?p=139945

cheton, 2016-07-20:
Note that you may still need to add a `~/.cncrc` file as I mentioned before in order to use `/dev/ttyAMA0` instead of `ttyAMA0`.

soswow, 2016-07-20:
# **I AM IN!**

I've reinstalled **serialport** manually with -g. 
And I've added .cncrc
Thought I was needed to run `--config ~/.cncrc` specifically to see this new items. I have a suspicion, that .cncrc was turning factor. I wonder if this is a genuine bug?

closed 2016-07-20 by soswow

soswow, 2016-07-20:
Item in the FAQ about it - was it there always, or added just now? =)

cheton, 2016-07-20:
In general, you don't need to manually specify the `--config` parameter unless the `.cncrc` file is not located at the login (home) directory.

I just updated the FAQ, it will help others if they encountered the similar problem.