cncjs/cncjs #59: Raspbian Jessy: Error on serial connection

thymjan, 2016-08-01:
I've tried to install cncjs two times now (as instructed). The server application is starting.
On calling the server adress on the client the frontend shows up.
But on starting the serial connection there are errors each time, then nothing happens:

`2016-08-01T18:10:49.956Z - error: uncaughtException: Symbol is not defined date=Mon Aug 01 2016 20:10:49 GMT+0200 (CEST), pid=19238, uid=1000, gid=1000, cwd=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nodejs, version=v0.10.29, argv=[node, /home/pi/.npm/bin/cnc], rss=51122176, heapTotal=43238372, heapUsed=25742296, loadavg=[0.6337890625, 0.79150390625, 0.7919921875], uptime=243399.775887884, trace=[column=47, file=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js, function=GrblLineParser.parse, line=1813, method=parse, native=false, column=39, file=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Grbl.parse, line=2326, method=parse, native=false, column=34, file=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js, function=, line=2752, method=null, native=false, column=17, file=events.js, function=SerialPort.emit, line=95, method=emit, native=false, column=17, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=null, line=24, method=null, native=false, column=null, file=null, function=Array.forEach, line=null, method=forEach, native=true, column=13, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=, line=23, method=null, native=false, column=18, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=SerialPort._emitData, line=310, method=_emitData, native=false, column=14, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=290, method=null, native=false, column=7, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=303, method=null, native=false, column=17, file=fs.js, function=Object.wrapper [as oncomplete], line=463, method=wrapper [as oncomplete], native=false], stack=[ReferenceError: Symbol is not defined,     at GrblLineParser.parse (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:1813:47),     at Grbl.parse (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:2326:39),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:2752:34),     at SerialPort.emit (events.js:95:17),     at /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:24:17,     at Array.forEach (native),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:23:13),     at SerialPort._emitData (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:310:18),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:290:14),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:303:7),     at Object.wrapper [as oncomplete] (fs.js:463:17)]
2016-08-01T18:10:50.029Z - error: uncaughtException: Symbol is not defined date=Mon Aug 01 2016 20:10:50 GMT+0200 (CEST), pid=19238, uid=1000, gid=1000, cwd=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nodejs, version=v0.10.29, argv=[node, /home/pi/.npm/bin/cnc], rss=51216384, heapTotal=43238372, heapUsed=26069596, loadavg=[0.6337890625, 0.79150390625, 0.7919921875], uptime=243399.931657921, trace=[column=47, file=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js, function=GrblLineParser.parse, line=1813, method=parse, native=false, column=39, file=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Grbl.parse, line=2326, method=parse, native=false, column=34, file=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js, function=, line=2752, method=null, native=false, column=17, file=events.js, function=SerialPort.emit, line=95, method=emit, native=false, column=17, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=null, line=24, method=null, native=false, column=null, file=null, function=Array.forEach, line=null, method=forEach, native=true, column=13, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=, line=23, method=null, native=false, column=18, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=SerialPort._emitData, line=310, method=_emitData, native=false, column=14, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=290, method=null, native=false, column=7, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=303, method=null, native=false, column=17, file=fs.js, function=Object.wrapper [as oncomplete], line=463, method=wrapper [as oncomplete], native=false], stack=[ReferenceError: Symbol is not defined,     at GrblLineParser.parse (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:1813:47),     at Grbl.parse (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:2326:39),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app/index.js:2752:34),     at SerialPort.emit (events.js:95:17),     at /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:24:17,     at Array.forEach (native),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:23:13),     at SerialPort._emitData (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:310:18),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:290:14),     at SerialPort.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:303:7),     at Object.wrapper [as oncomplete] (fs.js:463:17)]
`

I'm a newbie to node.js. Quite a bit too massive for me the message. But something went wrong with the serial port, does it?

This is my installation procedure:
[cnc_installation_procedure.txt](https://github.com/cheton/cnc/files/394482/cnc_installation_procedure.txt)

Can you give me a hint what went wrong?

cheton, 2016-08-02:
Hi @thymjan,

According to your installation logs, you have two Node.js running in your Raspbian Jessey: Node.js v4.4.7 (installed by nvm) and Node.js v0.10 (system default).

and I found this error in your logs:
```bash
nvm is not compatible with the npm config "prefix" option: currently set to "/home/pi/.npm"
Run `npm config delete prefix` or `nvm use --delete-prefix v4.4.7` to unset it.
```

The npm that you've updated is running with Node.js v0.10, not Node.js v4.4.7. To resolve this issue, you can remove the `~/.npmrc` file or just run `npm config delete prefix` to remove the prefix setting. Also remember to add the following lines to `~/.bashrc` or `~/.bash_profile` to have `nvm` automatically sourced upon login:
```bash
export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm
```

Then, make sure you're running Node.js v4 with the following command:
```bash
$ source ~/.bashrc
$ node -v
v4.4.7
```

After that, you can run below steps to install cncjs:
```bash
$ npm install npm@latest -g
$ npm -v
3.10.5
$ npm install -g cncjs
```

thymjan, 2016-08-02:
Dear @cheton ,

thank you for your fast answer with detailed an clear instructions! I followed them step by step.
"node -v" was still "v0.10.29" so I tried "nvm install 4" once more. Answer was:
   v4.4.7 is already installed.
   Now using node v4.4.7 (npm v2.15.8)
   Creating default alias: default -> 4 (-> v4.4.7)

And then... success! The installation of cnc worked without troubles, the serial connection either.

Thank you very much! This is a nice and useful frontend.

closed 2016-08-05 by cheton