cncjs/cncjs #61: Suggestion: macro support - 3 axis touchplate

meikrantz, 2016-08-18:
I have been using a 3 axis auto zero touch plate from  Triquetra CNC.  http://triquetra-cnc.com
It basically generates some gcode based on bit diameter, etc. that you can then cut and paste into a console, or upload as a simple program.  It would be nice to have some macro buttons that would let you store the gcode for various bit diameters to simplify cut and paste or loading multiple programs.  I'm sure there are other uses for stored macros.

If you really want to go overboard, you could build the 3 axis zero as feature in the program, and have the user input bit diameter in addition to z height and automate further.

There are similar features in bCNC and common in Mach 3.

With other touch plate shapes you could also do ring centering, inside/outside edge, etc., see http://www.themakersguide.com/home/products/triple-edge-finder-2

-- John

cheton, 2016-08-20:
I've seen the demo video at http://triquetra-cnc.com/triquetra-demo/ and it looks really awesome.

It may take me some time to think of a better way to manage macros from the web interface, but I can firstly add 3 axis probe support to the probe widget in addition to current Z probe.

From the [G-code Generator](https://triquetra-cnc.com/product/g-code-generator-without-touch-plate/) page, some parameters like "bit diameter" and "touch X/Y axis width" are necessary for running 3 axis probe:

[image]

May I know if you have some generated g-code samples for reference? Thanks.

meikrantz, 2016-08-23:
The numbers that are input for Touch X/Y/Z axis are based on the touch plate size.  For the Triquetra plate, the edges of X and Y are 2 inches from the edge of the touch plate, and the touch plate height above the measured surface is .8 inches.   Here is some example g-code with a 1/4" bit:

G92 X0
G92 Y0
G92 Z0
G20
G38.2 X-0.5F3.0
G92 X2.1250
G91 G0 X0.5
G91 G0 Y1
G91 G0 X-1
G38.2 Y-1 F3.0
G92 Y2.1250
G91 G0 Y0.125
G91 G0 Z0.900
G91 G0 Y- 0.625
G38.2 Z-1 F3.0
G92 Z0.8000
G91 G0 Z0.125
G91 G0 Y1 X1 

Let me know if this makes sense.

cheton, 2016-08-23:
May I know if the following parameters (e.g. X0.5, Y1, Z0.900) are fixed default values defined by the Triquetra generator?
```
G91 G0 X0.5
G91 G0 Y1
G91 G0 X-1
   :   :
G91 G0 Z0.900
G38.2 Z-1 F3.0
```

cheton, 2016-08-23:
I just purchased a touch plate and got the following result with the generator:

```
# Desire Feed: 76 mm/min
# Bit Diameter: 3.1750 mm
# Touch Z Axis Thickness: 20.3200 mm
# Touch X Axis Width: 50.8000 mm
# Touch Y Axis Width: 50.8000 mm

G92 X0
G92 Y0
G92 Z0
G21
G38.2 X-12.7F76
G92 X52.3875
G91 G0 X12.7
G91 G0 Y25.4
G91 G0 X-25.4
G38.2 Y-25.4 F76
G92 Y52.3875
G91 G0 Y03.175
G91 G0 Z24.07
G91 G0 Y- 15.875
G38.2 Z-25.4 F76
G92 Z20.32
G91 G0 Z3.175
G91 G0 Y25.4 X25.4
```

The result makes sense to me, but I may consider dividing "one-time 3 axis zero" into two steps with X-Y probing (e.g. http://www.mycncuk.com/attachment.php?attachmentid=8990&d=1370155187) and Z-probing to make it more generic. I'm also planning to add Macro to the web interface, it will be able to cover all different types of use.

cmsteinBR, 2016-08-23:
Dear,
I'll try to help, using your data.

# Desire Feed: 76 mm/min
# Bit Diameter: 3.1750 mm
# Touch Z Axis Thickness: 20.3200 mm
# Touch X Axis Width: 50.8000 mm
# Touch Y Axis Width: 50.8000 mm

G92 X0                     # Y=0
G92 Y0                     # Y=0
G92 Z0                     # Y=0
G21                          # millimeters (G20 = inches)
G38.2 X-12.7F76      # search for X position of touch with feed of 76 (initial X must be <= 12.7mm)
G92 X52.3875          # when find X, X0=52.3875 (Touch X Axis Width + Bit Diameter/2: 
                                # X0 = 50.8 + 3.175/2 = 52.3875)
G91 G0 X12.7          # returns to disengage the Touch (X=12.7)
G91 G0 Y25.4          # returns Y to stay in position (behind the piece) (initial Y = 25.4mm)
G91 G0 X-25.4        # forward X to stay in position (behind the piece): X = 12.7 - 25.4 = -12,7mm
G38.2 Y-25.4 F76    # search for Y position of touch with feed of 76 (initial Y must be <= 25.4mm)
G92 Y52.3875         # when find Y, Y0=52.3875 (Touch Y Axis Width + Bit Diameter/2: 
                                # Y0 = 50.8 + 3.175/2 = 52.3875)
G91 G0 Y03.175      # returns to disengage the Touch (Y=3.175)
G91 G0 Z24.07        # returns Z to stay in position (over the piece) (initial Z = 24.07mm)
G91 G0 Y- 15.875    # forward Y to stay in position (over the piece): Y = 3.175 - 15.875 = -12,7mm
G38.2 Z-25.4 F76    # search for Z position of touch with feed of 76 (initial Z must be <= 25.4mm)
G92 Z20.32             # when find Z, Z0=20.32 (Touch Z Axis Thickness): 
G91 G0 Z3.175       # goes to the Z end position 3.175mm over the piece
G91 G0 Y25.4 X25.4       # goes to the X and Y end positions: -12.7 + 25.4 = 12.7mm away piece

Zero is X0, Y0, Z0.
I do not use the program yet (discovered its existence yesterday), but I think the best would be to create this procedure as XYZ probing, and leave the current procedure as Z probing. This procedure can be used when a drill is changed.

meikrantz, 2016-08-24:
I would leave Z probing as a separate function.  It is used differently, i.e. to set bit height between changes.  But do the X,Y,Z with touchplate as a single function also.

cheton, 2016-08-26:
Sure! In addition to XYZ probing, I will also add a macro widget to automate frequently used workflows. Please stay tuned for further updates.

cheton, 2016-09-01:
The macro support will soon be available in v1.6.0. I will keep you updated.
[image]

cheton, 2016-09-02:
I have just released [v1.6.0](https://github.com/cheton/cnc/releases/tag/v1.6.0). You can use macro as a temporary workaround before XYZ probing is supported.

meikrantz, 2016-09-05:
I was able to upgrade to 1.6.0.  The macro function does not work for me.  When I create a macro, it never shows up in the list of macros after saving.

cheton, 2016-09-06:
Hi @meikrantz,

A newly created macro will be added to the `~/.cncrc` file. May I know if you can find the file in your environment?

[image]

You can also append `/api/macro` to the url to check if it returns all available macros from the `.cncrc` file:
[image]

meikrantz, 2016-09-06:
File exists, but only contains {}.  I am running on RPi2, accessing from remote browser on MacBook.  When I go to hostname:8000/api/macro web page just shows [ ]

cheton, 2016-09-06:
Hi @meikrantz,

I can reproduce this issue after a fresh install, a missing "macro" entry will fail the API call.
Could you try to paste the following content to your `~/.cncrc` file and check if that works? I will fix this issue in next patch release.

```json
{
    "macros": []
}
```

meikrantz, 2016-09-06:
That fixed it for now.

cheton, 2016-09-06:
Fixed in [v1.6.1](https://github.com/cheton/cnc/releases/tag/v1.6.1)

meikrantz, 2016-09-06:
Side comment - most of the time when I update via npm update cncjs, it doesn't make any changes.  I have to uninstall the previous version and then install the latest.

cheton, 2016-09-07:
Could you find the version of your installed npm using `npm -v`? If it's below v3.x, you can upgrade npm to the latest version via `npm install -g npm`.

meikrantz, 2016-09-07:
npm version is 3.3.12.  I ran npm install -g npm, and it looked like it was installing new, but version still shows 3.3.12.  Same result with npm update -g npm.

cheton, 2016-09-08:
It looks weird. Since npm v2 and v3 have different directory structures (nested vs. flat) for installed packages, try cleaning the `~/.npm` directory completely and use [nvm](https://github.com/cheton/cnc#nodejs-installation) to install and manage multiple Node.js versions, I think it should resolve most installation issues.

meikrantz, 2016-09-10:
I have had a chance to play with the macro feature and put together a few macros for different bit diameters to use with 3 axis touch plate.  It works great!  

[image]

cheton, 2016-09-19:
A new probing UI will be put in 2.0 milestone. I'll create a separate workspace for probing.

closed 2016-10-26 by cheton

nickm324, 2018-08-16:
has support for the TRIQUETRA ever been added?

nickm324, 2018-08-17:
@neilferreri  Ok, thought I saw somewhere in this thread where 3 Axis was going to get added to the ZPROBE widget.

neilferreri, 2018-08-17:
@nickm324 I think you're right, but there's no reason you couldn't use a regular macro in the meantime.  Doesn't he give you the gcode for the 3-axis probe?
What machine are you running?

nickm324, 2018-08-17:
@neilferreri Yea I am going to use the Macro feature when my Triquetra arrives, just thought I may have been overlooking it in the interface for the current version and this thread originated back in 2016 so I  figured it had made it into the current version by now.

matthoskins1980, 2020-01-19:
A quick bump to this - would love to see the feature added.

vfrman1, 2020-02-01:
I too would like to see the 3 axis feature.  Prior to switching to CNCjs, I was using Chilipeppr and there was a 3 axis super touch probe widget on there.  My only complaint was I had to re-enter all parameters every time I booted up to system.

imwhocodes, 2020-02-02:
@vfrman1 the feature is already implemented in the master branch (which is ahead of the release package)