cncjs/cncjs #64: Autostart

slimline33, 2016-09-05:
Hi there,

I know thats not the real point here to ask a question, because I can't get the ~/.npm/bin/cnc to autostart. I use rc.local and cncjs won't start.

If I start the rc.local manually cncjs starts perfekt!

Can anybody give a hint?

[code]
#!/bin/sh -e
/home/.npm/bin/cnc &
exit 0
[/code]

Regards Christoph

cheton, 2016-09-06:
The first line of the `cnc` script is `#!/usr/bin/env node`, which is used to locate where Node.js is installed. I guess the <b>PATH</b> variable may not include the search path (e.g. `/usr/local/bin` or others) for system startup scripts. So you need to make sure the PATH contains that directory before running `/home/.npm/bin/cnc`. You can find the path as below:

```bash
$ echo $PATH
/home/cheton/.nvm/versions/node/v4.4.7/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
```

Assume that `/home/cheton/.nvm/versions/node/v4.4.7/bin` is the search path, then set the PATH in the startup script:

```bash
#!/bin/sh -e

PATH=$PATH:/home/cheton/.nvm/versions/node/v4.4.7/bin
/home/.npm/bin/cnc &
```

Hope it works for you.

slimline33, 2016-09-07:
Hi cheton,

thanks for your replay.

Yes. The problem was the PATH.

Now I use the crontab -e for starting something at reboot with following variable:

`PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin:/home/pi/.npm/bin/:/usr/local/bin:
@reboot /home/pi/cnc >> /home/pi/cnc.log 2>&1
`

And it works :)

this says the cnc.log:
Started the server at http://0.0.0.0:8000/.

Thanks for helping so quick :)

Regards Christoph

closed 2016-09-08 by cheton

AustinSaintAubin, 2016-09-10:
@slimline33 what commands did you use to install on your pi? I cannot get my'n to autostart without error. See [Raspberry Pi Install Documentation #65](https://github.com/cheton/cnc/issues/65)

tupinamba77, 2024-08-10:
Hello,
Sorry to post in this thread after 8 years... 
I am a Linux noob and I can't manage to solve this PATH issue on my own. cncjs installation on Ubuntu 20.04LTS went smoothly but noting happens when I double-clic on the program's icon.
Step-by-step command line instructions would be much appreciated... (as I don't know how to start the rc.local manually or what crontab -e is... :-/ )
All the best.

tupinamba77, 2024-08-10:
I tried to add the following:
```
#!/bin/sh-e
PATH=$PATH:/home/alienized/.nvm/versions/node/v4.9.1/bin
/home/.npm/bin/cnc &
```
at the beginning of cncjs in /usr/bin.

But it doesn't change the situation : I cannot start the app...
I think there is a good chance I did not gedit the right file... :-)

tupinamba77, 2024-08-18:
Hello,
Solved. 
I modified the Exec in the .desktop file so it pointed to the actual cncjs-app location and now it launches perfectly.