cncjs/cncjs #65: Raspberry Pi Install Documentation

AustinSaintAubin, 2016-09-07:
Please update documentation to have a simple and straight forward install for the latest version of RASPBIAN. A prebuilt image would also be nice.

cheton, 2016-09-07:
The Node.js installation (https://github.com/cheton/cnc/wiki/Installation#nodejs-installation) method using Node Version Manager (NVM) is recommended and it can work with Raspbian Jessie. May I know if you encountered any installation problems?

AustinSaintAubin, 2016-09-08:
I am encountering problems...
Install is working. But keeping the node server running is the current problem.

Install Commands
```
# Updates & Upgrades
sudo apt-get update
sudo apt-get upgrade

# Install GIT ( already install on Raspbian Jessie )
sudo apt-get install git

# Install GCC & G++ ( already install on Raspbian Jessie )
sudo apt-get install gcc g++

# Install Node.js
git clone https://github.com/creationix/nvm.git ~/.nvm
cd ~/.nvm
git checkout `git describe --abbrev=0 --tags`
cd ..
. ~/.nvm/nvm.sh
nvm install 4

# Update Profile
cat >> ~/.profile <<EOF

export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm
EOF

# Run Profile to update current session
. ~/.profile

# Setup Node.js v3 support
#If you're using Node.js v4 or earlier versions, it's recommended that you use npm v3 to install packages.
# To upgrade, run:
npm install npm@latest -g

# Update Prefix
echo "prefix=~/.npm" >> ~/.npmrc
cat ~/.npmrc
#Output: prefix=~/.npm

# Install CNC.JS
npm install -g cncjs
```

Install Log:
```
pi@raspberrypi:~ $ sudo apt-get update
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
[... 566 lines cut ...]
  │ │   └── base64-url@1.2.1
  │ ├── i18next@3.1.0
  │ ├── multihost@0.1.1
  │ └─┬ serve-static@1.10.3
  │   └── send@0.13.2
  └─┬ winston@2.2.0
    ├── async@1.0.0
    ├── colors@1.0.3
    ├── cycle@1.0.3
    ├── eyes@0.1.8
    ├── isstream@0.1.2
    ├── pkginfo@0.3.1
    └── stack-trace@0.0.9

pi@raspberrypi:~ $ cnc
-bash: cnc: command not found
pi@raspberrypi:~ $ ~/.npm/bin/cnc
Started the server at http://0.0.0.0:8000/.
```

After First Reboot
```
pi@raspberrypi:~ $ ~/.npm/bin/cnc
2016-09-08T04:44:15.847Z - error: uncaughtException: /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register date=Thu Sep 08 2016 04:44:15 GMT+0000 (UTC), pid=1563, uid=1000, gid=1000, cwd=/home/pi/.npm/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nodejs, version=v0.10.29, argv=[node, /home/pi/.npm/bin/cnc], rss=30507008, heapTotal=28957028, heapUsed=12732680, loadavg=[0.7509765625, 0.67138671875, 0.29443359375], uptime=203.971941349, trace=[column=32, file=module.js, function=Module.load, line=356, method=load, native=false, column=12, file=module.js, function=Function.Module._load, line=312, method=Module._load, native=false, column=17, file=module.js, function=Module.require, line=364, method=require, native=false, column=17, file=module.js, function=require, line=380, method=null, native=false, column=44, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/bindings/bindings.js, function=bindings, line=76, method=null, native=false, column=35, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/bindings.js, function=, line=3, method=null, native=false, column=26, file=module.js, function=Module._compile, line=456, method=_compile, native=false, column=10, file=module.js, function=Object.Module._extensions..js, line=474, method=Module._extensions..js, native=false, column=32, file=module.js, function=Module.load, line=356, method=load, native=false, column=12, file=module.js, function=Function.Module._load, line=312, method=Module._load, native=false, column=17, file=module.js, function=Module.require, line=364, method=require, native=false, column=17, file=module.js, function=require, line=380, method=null, native=false, column=25, file=/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=12, method=null, native=false, column=26, file=module.js, function=Module._compile, line=456, method=_compile, native=false, column=10, file=module.js, function=Object.Module._extensions..js, line=474, method=Module._extensions..js, native=false, column=32, file=module.js, function=Module.load, line=356, method=load, native=false], stack=[Error: /home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register,     at Module.load (module.js:356:32),     at Function.Module._load (module.js:312:12),     at Module.require (module.js:364:17),     at require (module.js:380:17),     at bindings (/home/pi/.npm/lib/node_modules/cncjs/node_modules/bindings/bindi [... cut ...] e (module.js:456:26),     at Object.Module._extensions..js (module.js:474:10),     at Module.load (module.js:356:32),     at Function.Module._load (module.js:312:12),     at Module.require (module.js:364:17),     at require (module.js:380:17),     at Object.<anonymous> (/home/pi/.npm/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:12:25),     at Module._compile (module.js:456:26),     at Object.Module._extensions..js (module.js:474:10),     at Module.load (module.js:356:32)]
```

cheton, 2016-09-08:
I guess you may have two versions of Node.js installed in your system, one is Node.js v4.5.0 (which is installed by nvm), one is the Node.js v0.10 or v0.12 (prebuilt with system default). After reboot, you may run into the problem of using old Node.js (0.10 or 0.12) to run Node.js v4.5.0 packages, so a native module "serialport" reported an error of "undefined symbol".

It seems that I should remove the following paragraph from README and wiki, or it may cause confusions and result in installation problems:
```
If you're not using Node Version Manager, make sure you have 
prefix=~/.npm in ~/.npmrc to install global packages without sudo.
For example:

$ echo "prefix=~/.npm" >> ~/.npmrc
$ cat ~/.npmrc
prefix=~/.npm
```

To fix your problem, you have to delete the `~/.npmrc` file or remove the <b>prefix</b> line when using  Node Version Manager.

Just follow below steps to fix the problem:
```bash
pi@raspberrypi:~ $ cd ~/
pi@raspberrypi:~ $ rm .npmrc
pi@raspberrypi:~ $ rm -rf .npm   # Clean all files in .npm
pi@raspberrypi:~ $ nvm ls   # List all available Node.js versions
->       v4.5.0
pi@raspberrypi:~ $ nvm use 4  # Make sure you're using Node.js v4
Now using node v4.5.0 (npm v3.10.6)
pi@raspberrypi:~ $ npm install -g cncjs  # Re-install the package
pi@raspberrypi:~ $ which cnc  # The path should start with the "~/.nvm/" prefix
~/.nvm/versions/node/v4.5.0/bin/cnc
pi@raspberrypi:~ $ cnc -vv
Started the server at http://0.0.0.0:8000/.
```

#### Important note
If your `.profile` was not executed when started a bash terminal, try moving the nvm startup script to `.bash_profile` (or `.bashrc`), or you will fallback to system default's Node.js (v0.10 or v0.12).

Run `which node` or `node -v` can determine which Node.js version you're running:
```bash
$ which node
~/.nvm/versions/node/v4.5.0/bin/node
```

dronecz, 2016-09-08:
Hi, 

I wrote my own tutorial for my project, so check it out: 

[http://hackerspacepardubice.cz/cnc/wiki/doku.php?id=sw:instalation](url)

AustinSaintAubin, 2016-09-08:
@dronecz how are you accomplishing auto start on your Raspberry Pi.

dronecz, 2016-09-08:
Hi @AustinSaintAubin, 

I did not tested yet with cheton´s CNC program, but I think I did used it with other software and it worked. Follow instructions at this link: https://www.raspberrypi.org/documentation/linux/usage/rc-local.md

cheton, 2016-09-09:
You can also check it out at https://github.com/cheton/cnc/issues/64

AustinSaintAubin, 2016-09-09:
I think I finally got it figured out... bellow is my proccess... need to do a little more testing with it.. but so far it is solid... need sleep.

## Raspberry Pi Install
### Install NOOBS & RASPBIAN on your Raspberry Pi
https://www.raspberrypi.org/downloads/noobs/

### Configure Raspberry Pi
```sudo raspi-config```

1. Change Timezone
2. Change Hostname

### Updates & Upgrades
```
sudo apt-get update
sudo apt-get upgrade -y
sudo reboot
```

### Install GIT
```sudo apt-get install git -y```

### Install GCC & G++
```sudo apt-get install gcc g++ -y```

### Install Node.js
Installing an ARM-version of Node has become very easy:

```
sudo apt-get update
sudo apt-get install nodejs npm -y

-- OR --

wget https://nodejs.org/dist/v4.5.0/node-v4.5.0-linux-armv6l.tar.xz
tar -xvf node-v4.5.0-linux-armv6l.tar.xz 
cd node-v4.5.0-linux-armv6l
sudo cp -R * /usr/local/
```

### Install Node Version Manager
https://github.com/creationix/nvm

```
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.31.7/install.sh | bash
source ~/.profile  # Rerun profile after installing nvm

-- OR --

# Install Node Version Manager
git clone https://github.com/creationix/nvm.git ~/.nvm
cd ~/.nvm
git checkout `git describe --abbrev=0 --tags`
cd ..
. ~/.nvm/nvm.sh
nvm install 4

## Update Profile for Node Version Manager
# bashrc
cat >> ~/.bashrc <<EOF

export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm
EOF

# profile
cat >> ~/.profile <<EOF

export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm
EOF

# Run Profile to update current session
source ~/.profile

# Show Node Version Manager Information
nvm ls
nvm --version
```
To make sure it ran correctly, run ``node -v``, ``npm -v``, & ``nvm --version``. It should return the current versions.

### Setup Node.js Update
If you're using Node.js v4 or earlier versions, it's recommended that you use npm v3 to install packages.

```
# To upgrade, run:
sudo npm install npm@latest -g

# Fixes (If needed)
#npm install -g minimatch@3.0.2  # Fixes: npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
```

### Install Socket.IO (OPTIONAL)
```
npm install -g socket.io
npm install -g socket.io-parser
```

### Install Node.JS Serial Port (OPTIONAL)
```npm install serialport```

### Install CNC.JS
```
# Open Permisions to /usr/lib/node_modules
sudo chmod 777 /usr/lib/node_modules  # npm WARN checkPermissions Missing write access to /usr/lib/node_modules

# Install CNC.JS
npm install -g cncjs

# Close Permisions to /usr/lib/node_modules
sudo chmod 755 /usr/lib/node_modules  # npm WARN checkPermissions Missing write access to /usr/lib/node_modules
```

### Upgrade CNC.JS
```npm update -g cncjs```

## Auto Start Options

### Auto Start on Boot with cron
```
# Open crontab
crontab -u pi -e

# Paste the following into Cron Tab [ NOTE: which cnc  # used to find location of application ]
PATH=$PATH:/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin:/home/pi/.npm/bin/:/usr/local/bin
HOME=/home/pi
@reboot /usr/bin/cnc >> $HOME/cnc.log 2>&1
```

#### Auto Start on Boot with Forever
```
# Install Forever
sudo npm install -g forever
sudo npm install -g forever-service

# Create Service
sudo forever-service install conchs --start --script cnc /usr/bin/cnc
```

AustinSaintAubin, 2016-09-10:
Everything works except for auto start.
Get error message:
```
2016-09-10T22:05:49.286Z - error: uncaughtException: /home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register date=Sat Sep 10 2016 22:05:49 GMT+0000 (UTC), pid=457, uid=1000, gid=1000, cwd=/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nodejs, version=v0.10.29, argv=[node, /home/pi/.nvm/versions/node/v4.5.0/bin/cnc], rss=31567872, heapTotal=31025380, heapUsed=14977220, loadavg=[0.99951171875, 0.2275390625, 0.0751953125], uptime=20.274028898, trace=[column=32, file=module.js, function=Module.load, line=356, method=load, native=false, column=12, file=module.js, function=Function.Module._load, line=312, method=Module._load, native=false, column=17, file=module.js, function=Module.require, line=364, method=require, native=false, column=17, file=module.js, function=require, line=380, method=null, native=false, column=44, file=/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/bindings/bindings.js, function=bindings, line=76, method=null, native=false, column=35, file=/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/serialport/lib/bindings.js, function=, line=3, method=null, native=false, column=26, file=module.js, function=Module._compile, line=456, method=_compile, native=false, column=10, file=module.js, function=Object.Module._extensions..js, line=474, method=Module._extensions..js, native=false, column=32, file=module.js, function=Module.load, line=356, method=load, native=false, column=12, file=module.js, function=Function.Module._load, line=312, method=Module._load, native=false, column=17, file=module.js, function=Module.require, line=364, method=require, native=false, column=17, file=module.js, function=require, line=380, method=null, native=false, column=25, file=/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=12, method=null, native=false, column=26, file=module.js, function=Module._compile, line=456, method=_compile, native=false, column=10, file=module.js, function=Object.Module._extensions..js, line=474, method=Module._extensions..js, native=false, column=32, file=module.js, function=Module.load, line=356, method=load, native=false], stack=[Error: /home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/serialport/build/Release/serialport.node: undefined symbol: node_module_register,     at Module.load (module.js:356:32),     at Function.Module._load (module.js:312:12),     at Module.require (module.js:364:17),     at require (module.js:380:17),     at bindings (/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/bindings/bindings.js:76:44),     at Object.<anonymous> (/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/serialport/lib/bindings.js:3:35),     at Module._compile (module.js:456:26),     at Object.Module._extensions..js (module.js:474:10),     at Module.load (module.js:356:32),     at Function.Module._load (module.js:312:12),     at Module.require (module.js:364:17),     at require (module.js:380:17),     at Object.<anonymous> (/home/pi/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:12:25),     at Module._compile (module.js:456:26),     at Object.Module._extensions..js (module.js:474:10),     at Module.load (module.js:356:32)]
```

AustinSaintAubin, 2016-09-11:
Got it working

## Raspberry Pi Install: System Preparation
### Install NOOBS & RASPBIAN on your Raspberry Pi
https://www.raspberrypi.org/downloads/noobs/

### Configure Raspberry Pi
https://www.raspberrypi.org/documentation/linux/usage/users.md

```
# Change User Passwords
sudo passwd pi
sudo passwd root

sudo raspi-config
# Change Timezone
# Change Hostname
# Change Boot Option: Boot to CLI (No GUI)
```

### Updates & Upgrades
```
# Update System
sudo apt-get update
sudo apt-get upgrade -y
sudo reboot

# Install Useful Tools
sudo apt-get install htop iotop nmon lsof screen git build-essential -y
```

~~### Uninstall NodeJS & NPM installed with Apt-Get
```sudo apt-get autoremove node nodejs npm -y```~~

## Install NodeJS, Node Version Manager (NVM), Node Package Manager (NPM)

### [Install Node Version Manager (NVM)](https://github.com/creationix/nvm)
https://github.com/creationix/nvm

```
# Install Node Version Manager (NVM)
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.31.7/install.sh | bash

# Rerun Profile script to start NVM
source ~/.bashrc  # Rerun profile after installing nvm
```

### [Install Node.js](https://nodejs.org)
Installing an ARM-version of Node has become very easy:

```
# Install Node.js using Node Version Manager
nvm install 4  # Installs Node v4, (nvm install stable) installs Latest version of node
nvm use 4  # Sets Node to use v4
```

### [Update Node Package Manager (NPM)](https://docs.npmjs.com/getting-started/installing-node)
```npm install npm@latest -g```

To make sure it ran correctly, run ``npm -v``, ``nvm --version``, & ``node -v``. It should return the current versions.

```
# Output Node Related Version Info
echo "[NPM] ============"; which npm; npm -v;
echo "[NVM] ============"; nvm --version; nvm ls
echo "[NODE] ============"; which node; node -v
```

## Install CNC.js

~~### Install Node.JS Serial Port application first (OPTIONAL)
```npm install serialport```~~

### Install CNC.js
```npm install -g cncjs```

### Upgrade CNC.js
```npm update -g cncjs```

## Auto Start Options

### https://github.com/chovy/node-startup
```
# Install [Production Process Manager [PM2]](http://pm2.io)
npm install pm2 -g

# Start CNC.js with PM2
pm2 start /home/pi/.nvm/versions/node/v4.5.0/bin/cnc

# Setup PM2 Startup Script
pm2 startup debian
#[PM2] You have to run this command as root. Execute the following command:
sudo su -c "env PATH=$PATH:/home/pi/.nvm/versions/node/v4.5.0/bin pm2 startup debian -u pi --hp /home/pi"

# Set current running apps to startup
pm2 save

# Get list of PM2 processes
pm2 list
```

### Auto Start on Boot with cron (Alternative)
```
# Open crontab
crontab -u pi -e

# Paste the following into Cron Tab [ NOTE: which cnc  # used to find location of application ]
@reboot env PATH=$PATH:/home/pi/.nvm/versions/node/v4.5.0/bin /home/pi/.nvm/versions/node/v4.5.0/bin/cnc >> $HOME/cncjs.log 2>&1
```

# The Big Easy (just cut and paste)
### Configure Raspberry Pi
https://www.raspberrypi.org/documentation/linux/usage/users.md

```
# Change User Passwords
sudo passwd pi
sudo passwd root

sudo raspi-config
# Change Timezone
# Change Hostname
# Change Boot Option: Boot to CLI (No GUI)
```

### Install Everything
```
# Update System
sudo apt-get update
sudo apt-get upgrade -y

# Install Node Version Manager (NVM)
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.31.7/install.sh | bash

# Rerun Profile script to start NVM
source ~/.bashrc  # Rerun profile after installing nvm

# Install Node
nvm install 4  # Installs Node v4, (nvm install stable) installs Latest version of node
nvm use 4  # Sets Node to use v4

# Update Node Package Manager (NPM)
npm install npm@latest -g

# Get Version info
echo "[NPM] ============"; which npm; npm -v;
echo "[NVM] ============"; nvm --version; nvm ls
echo "[NODE] ============"; which node; node -v

# Install CNC.js
npm install -g cncjs

# Install [Production Process Manager [PM2]](http://pm2.io)
npm install pm2 -g

# Start CNC.js with PM2
pm2 start /home/pi/.nvm/versions/node/v4.5.0/bin/cnc

# Setup PM2 Startup Script
pm2 startup debian
#[PM2] You have to run this command as root. Execute the following command:
sudo su -c "env PATH=$PATH:/home/pi/.nvm/versions/node/v4.5.0/bin pm2 startup debian -u pi --hp /home/pi"

# Set current running apps to startup
pm2 save

# Get list of PM2 processes
pm2 list

# Reboot to test
sudo reboot
```

AustinSaintAubin, 2016-09-11:
Let me know what you think. Would like to get input and put this in the Wiki.

cheton, 2016-09-12:
Your help is greatly appreciated. I just created a wiki page and you can modify the content in below link:
https://github.com/cheton/cnc/wiki/Raspberry-Pi-Setup-Guide

Thanks!

AustinSaintAubin, 2016-09-12:
Updated documentation in the wiki.

closed 2016-09-12 by AustinSaintAubin

slimline33, 2016-09-16:
easy to use install! tested yesterday and all words fine on a fresh installed deb jessy and rpi 2. Thanks!