cncjs/cncjs #69: Drag & Drop upload does not seem to work

AustinSaintAubin, 2016-09-08:
Drag & Drop upload does NOT seem to work.
Clicking the "Upload G-code" button does work.
Tested on Win 7 x64 & Mac OS X 10.11 with released version apps v1.6.2

### Console Log
```
POST http://127.0.0.1:61099/api/gcode/upload 404 (Not Found)
2016-09-08T13:42:00-05:00 E Failed to upload file Error: Not Found(…) Response {req: Request, xhr: XMLHttpRequest, text: "<!--[if IE]><![endif]-->↵<!DOCTYPE html>↵<!--[if I…-date link</li>↵    </ul>↵</div>↵</body>↵</html>↵", statusText: "Not Found", statusCode: 404…}accepted: falsebadRequest: falsebody: nullcharset: "utf-8"clientError: trueerror: Error: cannot POST /api/gcode/upload (404)
    at Response.toError (http://127.0.0.1:61099/c117423d/vendor.js:43:31894)
    at Response._setStatusProperties (http://127.0.0.1:61099/c117423d/vendor.js:43:31549)
    at new Response (http://127.0.0.1:61099/c117423d/vendor.js:43:28355)
    at Request.<anonymous> (http://127.0.0.1:61099/c117423d/vendor.js:43:28857)
    at Request.Emitter.emit (http://127.0.0.1:61099/c117423d/vendor.js:138:24759)
    at XMLHttpRequest.xhr.onreadystatechange (http://127.0.0.1:61099/c117423d/vendor.js:44:2077)forbidden: falseheader: Objectheaders: Objectinfo: falsenoContent: falsenotAcceptable: falsenotFound: trueok: falsereq: RequestserverError: falsestatus: 404statusCode: 404statusText: "Not Found"statusType: 4text: "<!--[if IE]><![endif]-->↵<!DOCTYPE html>↵<!--[if IE 9]>         <html dir="" lang="" class="no-js ie9"> <![endif]-->↵<!--[if gt IE 9]><!--> <html dir="" lang="" class="no-js"> <!--<![endif]-->↵<head>↵<meta charset="utf-8">↵<meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1">↵<title>Page Not Found :(</title>↵<meta name="description" content="">↵<meta name="viewport" content="width=device-width, initial-scale=1">↵<style>↵::-moz-selection {↵    background: #b3d4fc;↵    text-shadow: none;↵}↵↵::selection {↵    background: #b3d4fc;↵    text-shadow: none;↵}↵↵html {↵    padding: 30px 10px;↵    font-size: 20px;↵    line-height: 1.4;↵    color: #737373;↵    background: #f0f0f0;↵    -webkit-text-size-adjust: 100%;↵    -ms-text-size-adjust: 100%;↵}↵↵html,↵input {↵    font-family: Helvetica, Arial, sans-serif;↵}↵↵body {↵    max-width: 500px;↵    _width: 500px;↵    padding: 30px 20px 50px;↵    border: 1px solid #b3b3b3;↵    border-radius: 4px;↵    margin: 0 auto;↵    box-shadow: 0 1px 10px #a7a7a7, inset 0 1px 0 #fff;↵    background: #fcfcfc;↵}↵↵h1 {↵    margin: 0 10px;↵    font-size: 50px;↵    text-align: center;↵}↵↵h1 span {↵    color: #bbb;↵}↵↵h3 {↵    margin: 1.5em 0 0.5em;↵}↵↵p {↵    margin: 1em 0;↵}↵↵ul {↵    padding: 0 0 0 40px;↵    margin: 1em 0;↵}↵↵.container {↵    max-width: 380px;↵    _width: 380px;↵    margin: 0 auto;↵}↵↵/* google search */↵↵#goog-fixurl ul {↵    list-style: none;↵    padding: 0;↵    margin: 0;↵}↵↵#goog-fixurl form {↵    margin: 0;↵}↵↵#goog-wm-qt,↵#goog-wm-sb {↵    border: 1px solid #bbb;↵    font-size: 16px;↵    line-height: normal;↵    vertical-align: top;↵    color: #444;↵    border-radius: 2px;↵}↵↵#goog-wm-qt {↵    width: 220px;↵    height: 20px;↵    padding: 5px;↵    margin: 5px 10px 0 0;↵    box-shadow: inset 0 1px 1px #ccc;↵}↵↵#goog-wm-sb {↵    display: inline-block;↵    height: 32px;↵    padding: 0 10px;↵    margin: 5px 0 0;↵    white-space: nowrap;↵    cursor: pointer;↵    background-color: #f5f5f5;↵    background-image: -webkit-linear-gradient(rgba(255,255,255,0), #f1f1f1);↵    background-image: -moz-linear-gradient(rgba(255,255,255,0), #f1f1f1);↵    background-image: -ms-linear-gradient(rgba(255,255,255,0), #f1f1f1);↵    background-image: -o-linear-gradient(rgba(255,255,255,0), #f1f1f1);↵    -webkit-appearance: none;↵    -moz-appearance: none;↵    appearance: none;↵    *overflow: visible;↵    *display: inline;↵    *zoom: 1;↵}↵↵#goog-wm-sb:hover,↵#goog-wm-sb:focus {↵    border-color: #aaa;↵    box-shadow: 0 1px 1px rgba(0, 0, 0, 0.1);↵    background-color: #f8f8f8;↵}↵↵#goog-wm-qt:hover,↵#goog-wm-qt:focus {↵    border-color: #105cb6;↵    outline: 0;↵    color: #222;↵}↵↵input::-moz-focus-inner {↵    padding: 0;↵    border: 0;↵}↵</style>↵</head>↵<body>↵<div class="container">↵    <h1>Not found <span>:(</span></h1>↵    <p>Sorry, but the page you were trying to view does not exist.</p>↵    <p>It looks like this was the result of either:</p>↵    <ul>↵        <li>a mistyped address</li>↵        <li>an out-of-date link</li>↵    </ul>↵</div>↵</body>↵</html>↵"type: "text/html"unauthorized: falsexhr: XMLHttpRequest__proto__: Object_parseBody: (str)_setHeaderProperties: (header)_setStatusProperties: (status)constructor: Response(req,options)get: (field)toError: ()__proto__: Object     at Object.error (http://127.0.0.1:61099/c117423d/app.js:1:23197)
```

Would also like to be able to upload with connecting over serial first.
Would also like to be able to specify if the gcode is metric or imperial.

cheton, 2016-09-09:
Ooops. It's likely to be some code change side effects since last week. I will fix it soon.

cheton, 2016-09-09:
Fixed in v1.6.3.

I just submitted another issue with respect to your suggestions. Thanks.

closed 2016-09-09 by cheton