cncjs/cncjs #70: Prompt user to choose the units option if it's not available in the G-code file

cheton, 2016-09-09:
If it cannot determine whether a G-code file should run in G20 (inches) or G21 (mm), prompt user to choose the units option to make sure the job will cut at the proper size.

This feature is already available in ChiliPeppr, like below:
[image]

AustinSaintAubin, 2016-09-10:
I recommend doing It differently than chili pepper. 
Popups are annoying.
I recommend having an option in settings or on the center preview widgit to change matric or imperial would be nice.

cheton, 2017-08-24:
This is not a necessary feature because user can change units on the UI anytime. I will close this issue.

closed 2017-08-24 by cheton