cncjs/cncjs #75: Proposals for technical extension of cnc.js

slimline33, 2016-09-12:
Hi cheton,

I'm not using cnc.js V. 1.3 so long like mach3 because I have some suggestions for new versions of your cnc.js.

- Live folder watch for uploading gcode
- Macros which can start sample/safed gcode
- Informing when a updated version is available and select update from the menu
- button for reboot (sudo reboot)
- shutdown (sudo halt)
- a menu point link to a own bash scripts..

This is would I suggest and that would be really nice add ons or widgets.

Is there something similar integrated shame on my. I haven't found that allready :D

Christoph

cheton, 2016-09-12:
> Live folder watch for uploading gcode

A watch directory will help you to import new files faster. I may add this feature in a future release.

> Macros which can start sample/safed gcode

Macros are supported in v1.6.

> Informing when a updated version is available and select update from the menu

Yes! I'm planning on adding update notifications to inform new updates.

> button for reboot (sudo reboot)
> shutdown (sudo halt)
> a menu point link to a own bash scripts..

I think the above three can be combined into one by selecting a menu item to trigger custom scripts from UI.

slimline33, 2016-09-12:
Yes that sounds really nice!

I use a precomposed raspberry image with installed cnc.js. I think i made a own image. Than it's easyer to update. 

Custom menu is a good idea. I think much people using only cnc.js on the raspberry for the own cnc machine. 

I think I add a kiosk mode which will only show a browser with cnc.js and that runs on the raspberry with a small touch lcd.

AustinSaintAubin, 2016-10-12:
Could really use the shutdown and reboot option for my raspberry pi though the GUI.

cheton, 2016-11-03:
With the 1.7.0 release, it will notify users when a new version (e.g. 1.7.x, 1.8.x) is available to download.

slimline33, 2016-11-27:
Now there is a "update found" message. Whats about with an "update" button? Something like this:

[image]

cheton, 2016-11-27:
By now the Electron app (i.e. executable installer .exe, .dmg) should be able to perform an update internally, but I have not yet implemented it to the app. It might be added in a future version.

For Node.js installation, I can try to force the update by running `npm install -g cncjs@latest` on the server side, but this may encounter failed update for certain reasons, maybe the software was installed using a different account, or a different installation path.

Do you have any recommendations for this? Trigger a custom script to update package might be a solution to it.

slimline33, 2016-11-27:
mhmm... sorry I have no solution for this. I understand the problems with the installation path because a update script could help I think.

This is the update-script from octoprint: https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/softwareupdate/scripts/update-octoprint.py but its python and no node. It can not help. 

First of all the software have to check the path where it is installed. On a raspberry I got with npm install -g cncjs@latest always the error with the serialmodule... I have to perform a  sudo npm install -g cncjs --unsafe-perm.
 
I hope anybody of other users have a solution? Should we open a new enhancement issue?

cheton, 2016-11-28:
OctoPrint's update script use git to update its source code, but that's not the case with the cnc software.

How about passing a command-line option to specify a path to the update script? For example:
```
$ cnc --update-script="/path/to/update-script.sh"
```

Then the web interface will be able to perform an update when the update script is available to use.

slimline33, 2016-11-28:
After a fresh installation the path is "/usr/lib/node_modules/cncjs/bin" - the script can check the path for the executable and if not found the user can input the real path. 

Or in the update page on the site the user can change the path.

cheton, 2016-12-02:
The watch directory support is added in the 1.8.5 release. You can check out the usage below:

### Usage

```
 $ cnc --watch-directory /home/pi/watch
```

Instead of passing command line options for `--watch-directory` and `--allow-remote-access`, you can create a `~/.cncrc` file that contains the following configuration in JSON format:

```json
{
  "watchDirectory": "/path/to/dir",
  "allowRemoteAccess": false
}
```

Then you can browse the watch directory to select a G-code file to load:

[image]

cheton, 2016-12-29:
The support for running shell commands is added in 95698ebceed2257edcd4900a247ed66ef0b6ed24, and later it will be published in 1.8.10 release. An example configuration (i.e. `~/.cncrc` file) will look like below:
```json
{
    "state": {
        "checkForUpdates": true
    },
    "commands": [
        {
            "text": "Reboot",
            "command": "/sbin/reboot"
        },
        {
            "text": "Shutdown",
            "command": "/sbin/shutdown"
        }
    ],
    "macros": [],
    "users": []
}
```

cheton, 2017-01-12:
@slimline33,

To perform an upgrade, you can run `pm2 start cnc` to monitor cnc process, and add "Update" command in `~/.cncrc` to trigger an update from UI. Once an update has been completed, it will kill the cnc process, then pm2 will automatically start the process again:
```json
{
  "commands": [
    {
      "text": "Update (root user)",
      "command": "sudo npm install -g cncjs@latest --unsafe-perm; pkill -a -f cnc"
    },
    {
      "text": "Update (non-root user)",
      "command": "npm install -g cncjs@latest; pkill -a -f cnc"
    }
  ]
}
```

closed 2017-01-12 by cheton

Thrillz, 2018-05-20:
@cheton hi, i have installed cncjs using npm and would like to add a custom widget to it. Going by the tutorial i have to create a directory under src/widgets but i can't find the "src" directory.

Thrillz, 2018-05-20:
Hi @cheton how do i add a custom widget to an npm installation of cncjs because i can't find the "src" directory even in node_modules

cheton, 2018-05-22:
The "src/widgets" folder you mentioned is located in cncjs-widget-boilerplate:
https://github.com/cncjs/cncjs-widget-boilerplate

You can follow the [development guide](https://github.com/cncjs/cncjs-widget-boilerplate#development
) to mount the default widget for testing

---

https://github.com/cncjs/cncjs-widget-boilerplate#development

### Development
Run npm run dev to start a local development server for development, then connect to http://localhost:5000 and wait until bundle finished.

You can specify a mount path to test your widgets with CNCjs:

cnc -vv --mount /widget:/path/to/cncjs-widget-boilerplate/dist

cheton, 2018-05-22:
Mounting from a remote server is also supported:
```bash
$ cncjs --mount /widget:https://cncjs.github.io/cncjs-widget-boilerplate/v1/
```

Just walk through the instruction to learn how to load a custom widget:
https://github.com/cncjs/cncjs-widget-boilerplate#configure-cncjs