cncjs/cncjs #76: SLOWNESS with vendor.js

AustinSaintAubin, 2016-09-12:
I have noticed a major performance issue with my lower power workstations and mobile devices. 
Tested with Firefox and Chrome on Xubuntu 16.04 LTS, both had major preformance issues with CNC webpage.
Tested: http://threejs.org/examples/#webgl_multiple_renderers
- Firstfox showed some lag with webgl_multiple_renderers
- Chrome showed NO lag with webgl_multiple_renderers

Firefox Developer Tools Console:
```
mutating the [[Prototype]] of an object will cause your code to run very slowly; instead create the object with the correct initial [[Prototype]] value using Object.create vendor.js:41:126
THREE.WebGLRenderer 78 vendor.js:26:25127
```
[preformace-profile.json.txt](https://github.com/cheton/cnc/files/468451/preformace-profile.json.txt)

------- 

Possible Solution Explanations / Related Issues:
* http://stackoverflow.com/questions/23807805/why-is-mutating-the-prototype-of-an-object-bad-for-performance
* https://github.com/mrdoob/three.js/issues/9364

AustinSaintAubin, 2016-09-15:
I think most of the performance issue is related to the preview rendering.
Anyway the preview widget could have the option to be disabled?
I would like to have the ability on tablets to disable the proview for performance reasons.

[image]

AustinSaintAubin, 2016-09-17:
Performance seems better as of v1.6.5

However, multiple devices do not sync current view.
Last device to connect get updates as CNC runs though gcode.

cheton, 2016-09-19:
Related Issue: https://github.com/cheton/cnc/issues/33

cheton, 2016-11-03:
This issue can be closed now since WebGL is no longer required by disabling 3D view in version 1.7.0.
Related issues: #33, #37, #86 

[image]

closed 2016-11-03 by cheton