cncjs/cncjs #77: UI scale on high dpi devices

liqdfire, 2016-09-14:
I am running cnc.js on a 2046x1440 12 inch screen, and having the ability to control the UI scaling would be nice.

I am not familiar enough with Electron to just add it, but it looks like it supports controlling the scaling through the webview component.

cheton, 2016-09-15:
The webContent object has methods for controlling zoom level: 
https://github.com/electron/electron/blob/master/docs/api/web-contents.md#contentssetzoomlevellevel

I will try to make the test and get back to you later.

cheton, 2016-09-15:
Added in this commit d6a157e59c2f7e784026c73a0946c831ca24eed6

[image]

I will publish a new version (v1.6.4) in the next few days that will contain the enhancement.

liqdfire, 2016-09-15:
Outstanding, thank you!

cheton, 2016-09-16:
Fixed v1.6.4

closed 2016-09-16 by cheton