cncjs/cncjs #78: Can't update my cnc-version with "npm update -g cncjs"

thymjan, 2016-09-16:
Hi Cheton,

I tried to update my cnc-version (1.5.0) with "npm update -g cncjs".
But afterwards "cnc -V" tells me still "1.5.0".

I also tried to update the packet manager with "npm install npm@latest -g" but had no luck.

Am I stupid? Can you give me a hint?

Thanks,
Stefan

cheton, 2016-09-16:
Hi @thymjan,

Can you try uninstalling it first and then install it again?
```bash
$ npm uninstall -g cncjs
$ npm install -g cncjs@latest
```

cheton, 2016-09-17:
You can also check out a wiki page for installing a specific cncjs version when having troubles on update:
https://github.com/cheton/cnc/wiki/Installation#installing-a-specific-version-of-cncjs-release

```bash
# Removed Current Version of CNC.js
sudo npm uninstall -g cncjs

# Install Specific Version of CNC.js
sudo npm install -g cncjs@1.6.3  --unsafe-perm
```

thymjan, 2016-09-17:
Hi @cheton,

thanks for your fast help again. "npm install -g cncjs@latest" without uninstall did the job.
So what's the difference between the update-procedure and the install-procedure?

Stefan

AustinSaintAubin, 2016-09-18:
The only big difference is that an already installed module with fuzzy versioning ...
gets ignored by npm install
gets updated by npm update

http://stackoverflow.com/questions/12478679/npm-install-vs-update-whats-the-difference

cheton, 2016-09-18:
Thanks for finding the answer. It seems that running `npm install -g cncjs@latest` is a better way for installing the latest version. I will update README and wiki as well.

loop @meikrantz

closed 2016-09-18 by cheton