cncjs/cncjs #80: Layout Suggestion for Macro Widget

AustinSaintAubin, 2016-09-17:
Could the Macro widget be edited to save a little space.
[image]

I sujest moving the Add Macro to a function as a Cog or "+" on the top of the widget bar like the 
[image]

cheton, 2016-09-17:
How about keeping the <kbd>Create Macro</kbd> button and the total records at the same line?

[image]

cheton, 2016-09-17:
This commit 77b563a7cb23b99ecef7354e07d06872524f9e63 adds support for remembering widget minimized state. Now you can minimize rarely used widgets to save more space on UI.

[image]

cheton, 2016-09-17:
Fixed in v1.6.5

closed 2016-09-17 by cheton

AustinSaintAubin, 2016-09-17:
That works, is cleaner now.