cncjs/cncjs #83: I can't run cncjs

Pinaute, 2016-09-19:
When I run the program
`$ /usr/bin/cnc -vv`
it says errors in my console logs as below :
```
2016-09-19T20:09:01.733Z - debug: app.settings: {"x-powered-by":false,"etag":"weak","env":"production","query parser":"extended","subdomain offset":2,"trust proxy":true,"views":"/usr/lib/node_modules/cncjs/dist/cnc/app/views","jsonp callback name":"callback","view cache":true,"verbose errors":false,"case sensitive routing":true,"strict routing":false,"view engine":"html"}     at appMain (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:2579:24)
Error: EEXIST: file already exists, mkdir './sessions'
    at Error (native)
    at Object.fs.mkdirSync (fs.js:922:18)
    at appMain (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:2599:23)
    at server (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:5950:43)
    at /usr/lib/node_modules/cncjs/node_modules/webappengine/dist/app/app.multihost.js:131:25
    at Array.forEach (native)
    at Object.appMain (/usr/lib/node_modules/cncjs/node_modules/webappengine/dist/app/app.multihost.js:96:31)
    at createServer (/usr/lib/node_modules/cncjs/node_modules/webappengine/dist/app/server.js:64:49)
    at serverMain (/usr/lib/node_modules/cncjs/node_modules/webappengine/dist/app/server.js:116:9)
    at createServer (/usr/lib/node_modules/cncjs/node_modules/webappengine/dist/app/index.js:57:33)
    at module.exports (/usr/lib/node_modules/cncjs/node_modules/webappengine/index.js:13:12)
    at Object.createServer (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:5988:33)
    at cnc (/usr/lib/node_modules/cncjs/dist/cnc/cnc.js:70:22)
    at Object.<anonymous> (/usr/lib/node_modules/cncjs/bin/cnc:11:1)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
2016-09-19T20:09:01.754Z - error: The multi-host server does not exist: {
    "type": "server",
    "route": "/"
}     at /usr/lib/node_modules/cncjs/node_modules/webappengine/dist/app/app.multihost.js:137:27 id=0, pid=3752
Started the server at http://0.0.0.0:8000/.
```
Is there any other system configuration was needed?
I'm seeing this in FireFox 45 for Debian jessie x64

Antoine

cheton, 2016-09-20:
There might be a permission problem to the `sessions` directory. Check out the permission of the path `/usr/lib/node_modules/cncjs/dist/cnc/app/sessions`.

If cncjs was installed using the root account, and you run cncjs with another login account, you may have to grant access to the `/usr/lib/node_modules/cncjs/dist/cnc/app/` folder.

Pinaute, 2016-09-20:
> If cncjs was installed using the root account, and you run cncjs with another login account

Yes it is.

Solve the issue with :
`chmod -R  go+w /usr/lib/node_modules/cncjs/dist/cnc/app/`
Now I run cncjs.

closed 2016-09-21 by Pinaute