cncjs/cncjs #84: support: system commands

MoonshineSG, 2016-10-02:
Is there any way to add custom command that can be somehow called via the web UI ? 
I have a RPi GPIO that turns on/off power for the cnc. I would like to have a way to switch that via the (so awesome)  cnc ui

cheton, 2016-10-02:
I'm planning on adding an user-defined command that can trigger custom scripts from UI. This feature is also mentioned in #75. Please stay tuned for further updates.

MoonshineSG, 2016-10-03:
good to hear that it's in the pipeline....

cheton, 2016-12-29:
This feature is added in 95698ebceed2257edcd4900a247ed66ef0b6ed24, and later it will be published in 1.8.10 release. An example configuration will look like below:
```json
{
    "state": {
        "checkForUpdates": true
    },
    "commands": [
        {
            "text": "Reboot",
            "command": "/sbin/reboot"
        },
        {
            "text": "Shutdown",
            "command": "/sbin/shutdown"
        }
    ],
    "macros": [],
    "users": []
}
```

MoonshineSG, 2016-12-30:
great. is there a way to update to non release versions (live on the edge) ?

cheton, 2016-12-30:
If you're willing to use the latest changes in the master branch, you can git clone this project on RPi, and run `npm install && npm run prepublish` to create a new build. You can follow all of the steps listed on this page: https://github.com/cheton/cnc/wiki/Installation#git-installation.

I will publish a new release within two days. Please stay tuned for further updates.

MoonshineSG, 2016-12-30:
I did that, but got a bunch of errors... I'll just be patient and wait until you do a release :D
Thanks!!!

cheton, 2016-12-30:
Just to let you know for security considerations, I didn't provide a page to add a shell command directly from the web interface, or hackers might be able to execute arbitratry commands or scripts on the server.

You have to manually update <b>~/.cncrc</b> file to add / remove commands for the time being.

MoonshineSG, 2016-12-30:
that's perfectly fine.

AustinSaintAubin, 2016-12-31:
@cheton It would be nice to have feedback / console output in CNC gui.
Might need a new system terminal widget or push notification of some type...

Also, here are my commands (includes CNC & System update)
```
  "commands": [
    {
      "text": "Check network connection",
      "command": "ping -c 10 127.0.0.1"
    },
    {
      "text": "Update System",
      "command": "sudo apt-get update; sudo apt-get upgrade -y; sudo apt-get dist-upgrade -y"
    },
    {
      "text": "Update CNCjs",
      "command": "sudo npm install npm@latest -g; sudo npm install -g cncjs --unsafe-perm; pm2 restart cnc; pm2 save; sudo npm install pm2 -g; pm2 update"
    },
    {
      "text": "Reboot",
      "command": "sudo /sbin/reboot"
    },
    {
      "text": "Shutdown",
      "command": "sudo /sbin/shutdown"
    }
  ],
```

cheton, 2017-01-01:
I'm also thinking about opening a new browser window (or notifications) to receive stdout output, might try to add it in a later release.

In addition to WebSockets, [SSE (Server-Sent Events)](http://www.w3schools.com/html/html5_serversentevents.asp) might be an alternative solution to it, browsers can receive server-sent event notifications automatically.

Some discussions on stackoverflow: 
http://stackoverflow.com/questions/21934831/nodejs-express-stream-stdout-instantly-to-the-client
http://stackoverflow.com/questions/5195452/websockets-vs-server-sent-events-eventsource

jeffeb3, 2017-01-19:
SSE or websockets would be nice if you want to see it while it's happening, but just hosting the log as a text file when it ends might be very vey useful, with little effort. Also, you might not want to see the output by default, just success/failure, with an option to check the output.

I feel like this is related, but maybe it's not, what about starting stopping commands based on the states of the CNC machine? I'd like to make a timelapse, while cutting, and I could do that if I could call a script when the gcode starts, finishes, or errors. I hope I'm not hyjacking this issue, or if I just haven't read enough docs to find the way you already provide to do this. I have used something similar with octoprint, they have a very good set of events that I use to turn lights on and off, or send a notification, or whatever I want.

cheton, 2017-02-07:
Added support for push notifications in 1.9.0-alpha. Now you can turn on notifications by clicking on the bell icon aside the "Command" header if not enabled.

[image]

Each executed command will follow by an animation icon indicating the work in progress.
[image]

On completion, you will receive a push notification showing command succeeded or command failed.

#### Command succeeded
[image]

#### Command failed
[image]

cheton, 2017-02-07:
**OctoPrint Events**
https://github.com/foosel/OctoPrint/wiki/Available-Events

cheton, 2017-02-07:
@jeffeb3 

I just made a reference to OctoPrint's events. How about adding some events to the <b> ~/.cncrc</b> file which might look like the following? You can send certain gcode lines or run a script once an event triggered.

```js
{
    "events": [
        { // Load
            "event": "command:load",
            "trigger": "gcode",
            "command": "G21 ; use millimeters\n$H",
        },
        { // Unload
            "event": "command:unload",
            "trigger": "gcode",
            "command": [ // array
                "M5" // spindle stop
            ]
        },
        { // Start
            "event": "command:start",
            "trigger": "system",
            "command": "/path/to/script"
        },
        { // Stop
            "event": "command:stop",
            "trigger": "system",
            "command": "/path/to/script"
        },
        { // Pause
            "event": "command:pause",
            "trigger": "system",
            "command": "/path/to/script"
        },
        { // Resume
            "event": "command:resume",
            "trigger": "system",
            "command": "/path/to/script"
        }
    ]
}
```

jeffeb3, 2017-02-07:
That looks very useful. 

Some thoughts:
How would you know which are system calls and which are gcode? You could have separate variable names. system_command and gcode_command.
Would the command have to be a script? Would it work to just call something in your path? Script only seems fine, but I could see that confusing users. 
The commands and scripts could be lists, right? They would get executed in serial, I assume? Including sleeps?
For the gcode, is there a use case for someone loading a file's with of commands? Like spindle off, move z up, go to machine home, z down to park position.

cheton, 2017-02-08:
Thank you for your input. From my original thoughts, events management should be done on the UI, not from a text editor. This can help avoid confusion.

The trigger type (i.e. "system" or "gcode") can be used to differentiate between gcode and system calls, both of them can be a multi-line text input control separated by a line break.  For examples:

1. The gcode trigger is smilar to macro:
   ```
   G21
   G90
   ```

2. The system trigger will be a script executed line by line:
   ```
   /path/to/your/script
   sleep 5
   shutdown 
   ```

The gcode trigger can also be used to support M6 tool change command (#118) that's similar to the use case you mentioned above.

cheton, 2017-02-16:
@jeffeb3 

In the next few days, I'm going to add preliminary support for event trigger, which includes the following events:

---

gcode:load
gcode:unload
gcode:start
gcode:pause
gcode:resume
gcode:stop
feedhold
cyclestart
homing
sleep  # <kbd>Sleep</kbd> is currently supported by Grbl 1.1
loadmacro
loadfile

---

Did you find anything useful but missing in above list?

jeffeb3, 2017-02-20:
Sorry I didn't respond sooner, I've had a lot on my plate.

These look like they are enough for my needs. I'm planning on starting a timelapse when gcode:start and then stopping it on gcode:stop. That happens when the gcode finishes not just when I stop it, right? I might also add a pushbullet or slack notification on gcode:stop as well.

If I stretch my head, I could see an event when the cnc.js starts as being useful. An event for specific gcode, like M6 might be good too. I could have it send me a notification when it needs a tool change. I haven't done a tool change yet, but that seems useful.

I'll see if I can pull these changes onto my CNC today or tomorrow. Thanks.

cheton, 2017-02-20:
This feature is available in 1.9.0-alpha.3, but the management UI is not implemented yet. I will add this part in 1.9.0-beta or the final 1.9.0 release.

cheton, 2017-03-09:
Commands and events configuration UI are now available in 1.9.0-alpha.5.

closed 2017-03-09 by cheton