cncjs/cncjs #86: Response Time to Jog Commands

GeerGuy, 2016-10-07:
When using an RPi 2 or 3 with the newest full Raspbian Jessie with Pixel image I can "over-load" my Grbl controller (Shapeoko 3) if I hold down a jog key.  It only occurs when I use a real keyboard and the Custom move amount is large, like 5 or more.  I have to reset the controller to continue.

Should the RPi settings be optimized to help with this or would the Raspbian Jessie Lite or would the NOOBS or NOOBS Lite be any better?  I will try these to compare.

I will also test if the issue is from the keyboard repeat rate.

GeerGuy, 2016-10-13:
I think that the issue is with the speed of the device running the browser.  If it is too slow or doesn't have enough RAM the jog commands are not "smooth".

The RPi3 works fine when no Gcode is loaded or only small files, but bogs down with larger files.  With a more powerful computer everything works fine.

cheton, 2016-10-25:
@GeerGuy

I recenly had a similar issue with my MacBook Air (Intel HD3000 Graphics Card, 2012), Google Chrome browser reported that WebGL is disabled due to a blacklisted graphics card. Eventually I resolved this issue by navigating to `chrome://flags/` and enable the option <b>Override software renderning list</b>. Everything works fine after restarting Chrome.

[image]

You can go to http://webglreport.com/ to check if your browser support WebGL. If WebGL is disabled for certain reason, check out http://superuser.com/questions/836832/how-can-i-enable-webgl-in-my-browser to enable WebGL in your browser.

GeerGuy, 2016-10-26:
It looks like the RPi3 does not support WebGL right now.  I will try it again when it is supported.

cheton, 2016-10-26:
The RPi3 is not necessary to support WebGL, but a device (desktop / laptop) running the web browser should support WebGL. Did you use a RPi3 touchscreen display to open a browser?

GeerGuy, 2016-10-26:
I would like to use an RPi3 running Raspbian Jessie with Pixel as the remote device.  I use an RPi2 to run the CNC server, it works great.

cheton, 2016-10-26:
You can refer to these articles to get your system and browser ready for WebGL:
https://www.scirra.com/blog/ashley/24/a-better-way-to-get-webgl-on-the-raspberry-pi-2
http://raspberrypi.stackexchange.com/questions/49690/chromium-50-webgl-support-on-jessie

It looks like Chromium is the only browser which works well for WebGL:
>
When you've finished installing, Chromium should appear under "Internet" in the main menu. Start it up, and you'll see it looks and feels a lot like Chrome. WebGL support might work out of the box: visit get.webgl.org to check. Otherwise, type chrome://flags in the address bar, under "Override software rendering list" click "Enable", relaunch the browser when it prompts you to, and try again. You should see a spinning cube at the get.webgl.org page! You can also visit chrome://gpu to check the hardware acceleration status - it should say "WebGL: Hardware accelerated" in green.

GeerGuy, 2016-10-26:
I tried all of that, but it is still experimental and has issues with stability.  At this point it is not reliable enough to be useful.  I'm going to wait for it to get worked out.  Thanks

cheton, 2016-10-26:
I may provide an option that can disable 3D visualizer in a future release, that would be suitable for devices without GPU hardware acceleration and WebGL support.

cheton, 2016-11-03:
From version 1.7.0, the 3D view will be automatically turned off if the machine or browser does not have GPU hardware acceleration and WebGL support.

You can also disable 3D view at anytime, it will provide better performance when sending G-code to  CNC controllers.

**3D view enabled**
[image]

**3D view disabled**
[image]

closed 2016-11-03 by cheton