cncjs/cncjs #87: Page not found - Linux installer not working

CadBrad, 2016-10-14:
I'm using Linux Ubuntu 14.04 and this program isn't installing correctly for me.  I'm getting a "page not found :( " error with the deb file, and I can't extract the tar file. I've tried 1.6.6 and 1.6.5 in case its a version compatibility issue, but this didn't solve the issue. 

Any suggestions on how to easily run this? Source code was my next step.

CadBrad, 2016-10-14:
I can get it to work from the source code, but only by calling "sudo cnc" if I try "cnc" I get the same error as I did with the deb installer.  This leads me to believe that the deb installer probably did work but didn't run it as an admin, which means its my error and not an issue of the program.

closed 2016-10-14 by CadBrad

cheton, 2016-10-14:
There is also a similar permission issue that the program was running with a different account. You can check it out at https://github.com/cheton/cnc/issues/83.